coolsnowwolf/lede #8896: luci-app-cpufreq这个插件哪去了?

ghaner, 2022-02-17:
(好多插件怎么不在https://github.com/coolsnowwolf/lede/tree/master/package/lean这个目录了.

xmapst, 2022-02-18:
> 看下提交记录不就知道了

[cc9ce1dbb53b58c66fd5e72e6ec7a5e7428d91d2](https://github.com/coolsnowwolf/lede/commit/cc9ce1dbb53b58c66fd5e72e6ec7a5e7428d91d2)全删了?

deplives, 2022-02-18:
> > 看下提交记录不就知道了
> 
> [cc9ce1dbb53b58c66fd5e72e6ec7a5e7428d91d2](https://github.com/coolsnowwolf/lede/commit/cc9ce1dbb53b58c66fd5e72e6ec7a5e7428d91d2)全删了?

[image]
看见这句话了么

qqsir-dev, 2022-02-18:
看见了，但是怎么找不到这个目录啊？