coolsnowwolf/lede #8899: Newifimini隔一段时间会断网，日志显示ethernet eth0: transmit timed out

wrnxr233, 2022-02-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
https://github.com/coolsnowwolf/lede/issues/7960
https://github.com/coolsnowwolf/lede/issues/7665
https://github.com/coolsnowwolf/lede/issues/8425
https://github.com/coolsnowwolf/lede/issues/8407
https://github.com/coolsnowwolf/lede/issues/8432
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Newifimini隔一段时间会断网，日志显示ethernet eth0: transmit timed out

### (2) 路由器型号和固件版本

A：路由器型号Newifimini，CPUMT7620A，固件版本OpenWrt R22.2.2 / LuCI Master (git-22.046.49039-d9baa31)，内核版本5.10.100
### (3) 详细日志

A：Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.587406] ------------[ cut here ]------------
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.592172] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:468 dev_watchdog+0x26c/0x274
Fri Feb 18 12:17:35 2022 kern.info kernel: [ 2864.600633] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.608015] Modules linked in: xt_FULLCONENAT rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 lzo ipt_REJECT cfg80211 xt_time xt_tcpudp xt_socket xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_comment xt_TPROXY xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan lzo_decompress lzo_compress iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.608458]  asn1_decoder cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg hmac ecb cmac authenc arc4 crypto_acompress leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.764920] CPU: 0 PID: 0 Comm: swapper Not tainted 5.10.100 #0
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.770969] Stack : 806d0000 805d5bb8 00000000 00000000 80c09e0c 80800000 80646200 80645dc7
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.779546]         805ad820 00000000 808033c4 ffffffff 806c0000 00000001 80c09dc0 105cb797
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.788131]         00000000 00000000 805ad820 80c09c60 ffffefff 806b101c 00000000 00000000
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.796689]         00000000 0000008e 00000000 000babf8 00000000 80c09ea0 00000009 803e1328
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.805257]         00000009 ffffffff 806c0000 00000000 00000018 8030a56c 00000000 80800000
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.813841]         ...
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.816347] Call Trace:
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.818877] [<80008474>] show_stack+0x30/0x100
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.823432] [<80026354>] __warn+0xc0/0xe8
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.827549] [<80026408>] warn_slowpath_fmt+0x8c/0xac
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.832628] [<803e1328>] dev_watchdog+0x26c/0x274
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.837462] [<8006ee58>] call_timer_fn.isra.33+0x20/0x90
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.842892] [<8006f194>] run_timer_softirq+0x2cc/0x338
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.848160] [<8051c398>] __do_softirq+0x110/0x298
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.852968] [<80002fb0>] except_vec_vi_end+0xb8/0xc4
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.858074] [<8051c13c>] r4k_wait_irqoff+0x18/0x24
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.862985] [<8004e8fc>] do_idle+0xd0/0x104
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.867258] [<8004eb3c>] cpu_startup_entry+0x18/0x28
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.872357] [<806d6d30>] start_kernel+0x584/0x5ac
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.877159]
Fri Feb 18 12:17:35 2022 kern.warn kernel: [ 2864.878691] ---[ end trace 53931f5fae596159 ]---
Fri Feb 18 12:17:35 2022 kern.err kernel: [ 2864.883413] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Feb 18 12:17:35 2022 kern.info kernel: [ 2864.890005] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Feb 18 12:17:35 2022 kern.info kernel: [ 2864.896411] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=021c0000, max=1024, ctx=125, dtx=125, fdx=113, next=125
Fri Feb 18 12:17:35 2022 kern.info kernel: [ 2864.907707] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=021a8000, max=1024, calc=153, drx=155
Fri Feb 18 12:17:41 2022 kern.err kernel: [ 2867.337430] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Feb 18 12:17:41 2022 kern.info kernel: [ 2867.344049] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Feb 18 12:17:41 2022 kern.info kernel: [ 2867.350487] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=021c0000, max=1024, ctx=472, dtx=472, fdx=464, next=472
Fri Feb 18 12:17:41 2022 kern.info kernel: [ 2867.361786] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0221c000, max=1024, calc=356, drx=359
Fri Feb 18 12:17:47 2022 kern.err kernel: [ 2869.941643] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Feb 18 12:17:47 2022 kern.info kernel: [ 2869.948248] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Feb 18 12:17:47 2022 kern.info kernel: [ 2869.954693] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=021c0000, max=1024, ctx=720, dtx=720, fdx=434, next=720
Fri Feb 18 12:17:47 2022 kern.info kernel: [ 2869.965999] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=021a8000, max=1024, calc=505, drx=507
Fri Feb 18 12:17:52 2022 kern.err kernel: [ 2872.446049] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Feb 18 12:17:52 2022 kern.info kernel: [ 2872.452655] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Feb 18 12:17:52 2022 kern.info kernel: [ 2872.459088] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=021c0000, max=1024, ctx=603, dtx=603, fdx=176, next=603
Fri Feb 18 12:17:52 2022 kern.info kernel: [ 2872.470398] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=0221c000, max=1024, calc=437, drx=439
Fri Feb 18 12:18:02 2022 kern.err kernel: [ 2876.041552] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Fri Feb 18 12:18:02 2022 kern.info kernel: [ 2876.048172] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Fri Feb 18 12:18:02 2022 kern.info kernel: [ 2876.054619] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=021c0000, max=1024, ctx=142, dtx=142, fdx=16, next=142
Fri Feb 18 12:18:02 2022 kern.info kernel: [ 2876.065824] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=021a8000, max=1024, calc=229, drx=230
Fri Feb 18 13:05:17 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 3a:47:d2:77:8b:6e
Fri Feb 18 13:05:17 2022 daemon.info hostapd: wlan0: STA 3a:47:d2:77:8b:6e IEEE 802.11: disassociated due to inactivity
Fri Feb 18 13:05:18 2022 daemon.info hostapd: wlan0: STA 3a:47:d2:77:8b:6e IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

xiaopang0v0, 2022-03-01:
其实我的也有，现在换着x-wrt,算是稳定一点了，但是软件包还要自己编译慢慢调https://github.com/coolsnowwolf/lede/issues/8046

wrnxr233, 2022-03-02:
> 其实我的也有，现在换着x-wrt,算是稳定一点了，但是软件包还要自己编译慢慢调#8046

那x-wrt有这个问题吗

xiaopang0v0, 2022-03-09:
> > 其实我的也有，现在换着x-wrt,算是稳定一点了，但是软件包还要自己编译慢慢调#8046
> 
> 那x-wrt有这个问题吗

x-wrt没有这个问题，但是x-wrt界面差异较大，而且要调的比较多，很多lede能用的主题用不了，不考虑这些的话，x-wrt还行

xiaopang0v0, 2022-03-09:
> > 其实我的也有，现在换着x-wrt,算是稳定一点了，但是软件包还要自己编译慢慢调#8046
> 
> 那x-wrt有这个问题吗
x-wrt的内核很新，没有这个问题，内核也很好记，5.10.100

coolsnowwolf, 2026-09-05:
Newifi Mini 的 `ralink_soc_eth` watchdog 是旧 MT7620/5.10 组合问题，已有重复 issue，缺少现行版本复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf