coolsnowwolf/lede #8910: 5.15内核 ntfs3模块 crash：空指针引用

WWWWindGG, 2022-02-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

- [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：挂载ntfs磁盘，在挂载目录下执行`ls`，内核crash，设备重启

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
> [   84.787202] BUG: kernel NULL pointer dereference, address: 0000000000000008
[   84.794407] #PF: supervisor read access in kernel mode
[   84.799767] #PF: error_code(0x0000) - not-present page
[   84.805143] PGD 800000012540e067 P4D 800000012540e067 PUD 11fffb067 PMD 0 
[   84.812282] Oops: 0000 [#1] SMP PTI
[   84.815948] CPU: 3 PID: 8591 Comm: ls Not tainted 5.15.23 #0
[   84.821835] Hardware name:  YL-SKUL6/YL-SKUL6, BIOS 5.12 05/28/2018
[   84.828361] RIP: 0010:d_flags_for_inode+0x60/0x80
[   84.833284] Code: 0d 08 0f 45 c2 c3 31 c0 c3 b8 00 00 20 00 f6 c2 02 75 e3 48 8b 4f 20 48 83 39 00 74 24 83 ca 02 66 89 57 02 eb d0 48 8b 4f 20 <48> 83 79 08 00 75 09 83 ca 04 66 89 57 02 eb ab b8 00 00 60 00 eb
[   84.852837] RSP: 0018:ffffc9000765bb80 EFLAGS: 00010246
[   84.858302] RAX: 0000000000000000 RBX: ffff88811dedc178 RCX: 0000000000000000
[   84.865746] RDX: 0000000000000008 RSI: 0000000000000286 RDI: ffff88811dedc178
[   84.873217] RBP: ffffc9000765bbb0 R08: 0000000000000000 R09: ffffc9000765baf8
[   84.880673] R10: 0000000000000001 R11: 0000000000000000 R12: ffff888117215540
[   84.888107] R13: 0000000000000000 R14: ffff888103d5e838 R15: ffff888117215598
[   84.895536] FS:  00007f919f552b48(0000) GS:ffff888362d80000(0000) knlGS:0000000000000000
[   84.903987] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   84.909964] CR2: 0000000000000008 CR3: 000000011fefe003 CR4: 00000000003706e0
[   84.917322] Call Trace:
[   84.919812]  <TASK>
[   84.921976]  ? __d_add+0x41/0x170
[   84.925359]  d_splice_alias+0x87/0x280
[   84.929208]  decompress_lznt+0x58d/0x9c0 [ntfs3]
[   84.933965]  __lookup_slow+0x70/0x110
[   84.937708]  walk_component+0x107/0x170
[   84.941652]  path_lookupat+0x68/0x1b0
[   84.945404]  filename_lookup+0xae/0x160
[   84.949345]  ? strncpy_from_user+0x53/0x150
[   84.953635]  user_path_at_empty+0x3a/0x60
[   84.957763]  vfs_statx+0x66/0xf0
[   84.961079]  ? atime_needs_update+0x3c/0xe0
[   84.965370]  __do_sys_newlstat+0x2f/0x50
[   84.969394]  ? iterate_dir+0xfa/0x190
[   84.973154]  ? switch_fpu_return+0x48/0x80
[   84.977349]  ? exit_to_user_mode_prepare+0x63/0x130
[   84.982347]  __x64_sys_newlstat+0x11/0x20
[   84.986458]  do_syscall_64+0x43/0x90
[   84.990147]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[   84.995315] RIP: 0033:0x7f919f526aae
[   84.998982] Code: 41 83 f8 9c 74 06 41 80 39 2f 75 20 b8 06 00 00 00 81 f9 00 01 00 00 74 09 85 c9 75 0f b8 04 00 00 00 4c 89 cf 48 89 ee 0f 05 <eb> 13 49 63 f8 4c 63 d1 b8 06 01 00 00 4c 89 ce 48 89 ea 0f 05 41
[   85.018188] RSP: 002b:00007ffdb16f6950 EFLAGS: 00000246 ORIG_RAX: 0000000000000006
[   85.025941] RAX: ffffffffffffffda RBX: 00007ffdb16f6a30 RCX: 00007f919f526aae
[   85.033230] RDX: 00007ffdb16f6a30 RSI: 00007ffdb16f6958 RDI: 00007f919f553d30
[   85.040556] RBP: 00007ffdb16f6958 R08: 00000000ffffff9c R09: 00007f919f553d30
[   85.047879] R10: 0000000000000000 R11: 0000000000000246 R12: 00007f919f553d30
[   85.055208] R13: 00007f919f553d32 R14: 00007f919f553d30 R15: 000000000045b6e8
[   85.062532]  </TASK>
[   85.064770] Modules linked in: xt_FULLCONENAT pppoe ppp_async wireguard snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec pppox ppp_mppe ppp_generic libchacha20poly1305 libblake2s ipt_REJECT curve25519_x86_64 chacha_x86_64 blake2s_x86_64 ax88179_178a zstd xt_time xt_tcpudp xt_state xt_socket xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_esp xt_conntrack xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD xt_CT via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm tcp_bbr snd_usbmidi_lib snd_intel_dspcfg snd_hda_core slhc rtl8150 regmap_core r8168 r8152 r8125 poly1305_x86_64 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323
[   85.064814]  nf_nat_ftp nf_nat_amanda nf_log_syslog nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mlx5_core mlx4_en mlx4_core macvlan lzo libcurve25519_generic libchacha libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ip_tables igc iavf i40e forcedeth e1000e crc_ccitt br_netfilter bnx2x bnx2 alx 8139too 8139cp configs evdev i2c_dev ledtrig_audio cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables
[   85.154219]  nf_reject_ipv6 bonding snd_via82xx snd_intel8x0 snd_mpu401_uart ixgbe igbvf e1000 amd_xgbe mdio_devres snd_ac97_codec ac97_bus dummy sit mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore xfrm_user xfrm_ipcomp af_key xfrm_algo ntfs3 dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 vxlan udp_tunnel ip6_udp_tunnel ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 hmac echainiv des_generic libdes deflate authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm igb exfat btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug ptp pps_core mii libphy cbc encrypted_keys trusted tpm oid_registry asn1_encoder asn1_decoder
[   85.329508] CR2: 0000000000000008
[   85.332900] ---[ end trace 589f2e4fb9e6ba6c ]---
[   85.337628] RIP: 0010:d_flags_for_inode+0x60/0x80
[   85.342467] Code: 0d 08 0f 45 c2 c3 31 c0 c3 b8 00 00 20 00 f6 c2 02 75 e3 48 8b 4f 20 48 83 39 00 74 24 83 ca 02 66 89 57 02 eb d0 48 8b 4f 20 <48> 83 79 08 00 75 09 83 ca 04 66 89 57 02 eb ab b8 00 00 60 00 eb
[   85.361723] RSP: 0018:ffffc9000765bb80 EFLAGS: 00010246
[   85.367086] RAX: 0000000000000000 RBX: ffff88811dedc178 RCX: 0000000000000000
[   85.374413] RDX: 0000000000000008 RSI: 0000000000000286 RDI: ffff88811dedc178
[   85.381751] RBP: ffffc9000765bbb0 R08: 0000000000000000 R09: ffffc9000765baf8
[   85.389065] R10: 0000000000000001 R11: 0000000000000000 R12: ffff888117215540
[   85.396374] R13: 0000000000000000 R14: ffff888103d5e838 R15: ffff888117215598
[   85.403663] FS:  00007f919f552b48(0000) GS:ffff888362d80000(0000) knlGS:0000000000000000
[   85.411965] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   85.417858] CR2: 0000000000000008 CR3: 000000011fefe003 CR4: 00000000003706e0
[   85.425197] Kernel panic - not syncing: Fatal exception
[   85.430599] Kernel Offset: disabled
[   85.434239] Rebooting in 3 seconds..

WYC-2020, 2022-03-01:
我这边一切正常呢，x86-64  ntfs分区，上传、新建和ls 都没问题

coolsnowwolf, 2026-09-05:
该 NULL dereference 来自早期 Linux 5.15 ntfs3 实现；当前内核已包含后续 ntfs3 修复，报告对应的旧内核提交不再适用，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf