coolsnowwolf/lede #8911: Zerotier 无法正常启动 `host/network `' not found`

CuminLo, 2022-02-19:
###  详细日志

```
root@OpenWrt:/etc# ./zerotier.start
iptables v1.8.7 (legacy): host/network `' not found
Try `iptables -h' or 'iptables --help' for more information
```

第一次使用大雕的源码，zerotier配置是迁移过来的，一开始不行，以为是迁移兼容性问题，后来删除配置重新新建，也不能启动。

最后找到启动命令 `/etc/zerotier.start`，报 `host/network ' not found`

搜索一番，说是 `iptables` 相关的，对这里不了解，请求各位帮助。

lxl6125, 2022-02-20:
启动命令 /etc/zerotier.start ？？？
[image]

WYC-2020, 2022-02-22:
我这边一切正常呢，x86 5.15 内核
[image]
[image]
@CuminLo

CuminLo, 2022-02-22:
> x86 5.15

@WYC-2020 大佬，可能是什么问题导致的呢？ 我的是 x86_64 5.10

coolsnowwolf, 2026-09-06:
旧版 ZeroTier 启动脚本的 iptables NAT 参数为空，当前脚本已不再使用该旧 NAT 命令路径；没有现行复现，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf