coolsnowwolf/lede #8913: 5.15.23 内核不显示图表了。要咋搞？

flyhong666, 2022-02-20:
5.15.23 内核不显示图表了。要咋搞？之前5.15.15 都正常的。是不是将luci 相关程序大规模挪动后有影响？
[image]

WYC-2020, 2022-02-20:
不可能，移动库和这个没关系

flyhong666, 2022-02-20:
> 不可能，移动库和这个没关系

那时是情况造成？

WYC-2020, 2022-02-21:
luci 位置变动只会导致 引用的位置变化，不会导致这个问题，除非删除的版本 和那边新加的版本不一致，或者是你源有多个重复的，此时-f 命令不会覆盖不是在lede 目录下的，会根据你源的顺序来使用第一次遇到的哪个包，这个没时间去看，你这个是什么包嘛，我编译看下

flyhong666, 2022-02-21:
> luci 位置变动只会导致 引用的位置变化，不会导致这个问题，除非删除的版本 和那边新加的版本不一致，或者是你源有多个重复的，此时-f 命令不会覆盖不是在lede 目录下的，会根据你源的顺序来使用第一次遇到的哪个包，这个没时间去看，你这个是什么包嘛，我编译看下

luci-app-vnstat

WYC-2020, 2022-02-22:
一切正常呢 
[image]
[image]
@flyhong666

flyhong666, 2022-02-22:
make clean 后也正常了。。。

closed 2022-02-22 by flyhong666