coolsnowwolf/lede #8914: kernel 5.15：编译的时候提示mac80211编译失败

ricksuzade-maker, 2022-02-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
编译的时候提示mac80211编译失败

### (2) 路由器型号和固件版本

A：ESXi 7 x86-64, 5.15.23 kernel

### (3) 详细日志

A：

cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-install/lib/libboost_system*.so* /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system/usr/lib/
find /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/patchelf" /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/rstrip.sh /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system
rstrip.sh: /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system/usr/lib/libboost_system.so.1.76.0: shared object
(cd /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages
/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/fakeroot /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/ipkg-build -m "" /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages
Packaged contents of /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost-system into /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages/boost-system_1.76.0-1_x86_64.ipk
rm -rf /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system.installed /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system
install -d -m0755 /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system/usr/lib
cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-install/lib/libboost_system*.so* /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system/usr/lib/
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system.installed
removed '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages/boost_1.76.0-1_x86_64.ipk'
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/targets/x86/64/packages /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost/CONTROL /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo
true
find /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/patchelf" /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/rstrip.sh /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost
(cd /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages
/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/host/bin/fakeroot /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/scripts/ipkg-build -m "" /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages
Packaged contents of /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/ipkg-x86_64/boost into /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/bin/packages/x86_64/packages/boost_1.76.0-1_x86_64.ipk
rm -rf /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost.installed /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost
true
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost.installed
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-date_time/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.boost-date_time_installed
echo "boost-date_time" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/boost.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-program_options/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.boost-program_options_installed
echo "boost-program_options" >> /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/pkginfo/boost.default.install
mkdir -p /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-x86_64_musl/boost_1_76_0/.pkgdir/boost-system/. /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-x86_64_musl/root-x86/'
[... cut ...]
time: package/kernel/mac80211/regular/compile#1.44#0.55#2.31
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: *** [package/kernel/mac80211/compile] Error 1
package/Makefile:114: recipe for target 'package/kernel/mac80211/compile' failed
make[1]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt'
/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
make: *** [package/compile] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-20:
这个我修复过没人留意

WYC-2020, 2022-02-22:
@ricksuzade-maker https://github.com/WYC-2020/lede/commit/1055d93e5f4761ed317a38b9c8f3a0e2a8131065

closed 2022-02-28 by AmadeusGhost

WhaleFell, 2024-12-05:
same problem