coolsnowwolf/lede #8917: 用几天，系统和内核日志一直错误提示，网不能用

tianiue, 2022-02-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用几天后突然出现，可以登录web页，网络不能正常使用

### (2) 路由器型号和固件版本

A：x86，5.10和5.15都出现

### (3) 详细日志

A：
[image]

[image]

pangfeng161, 2022-02-21:
使用昨天编译固件也是这个问题,不过不用重启等一分钟又恢复

tianiue, 2022-02-22:
> 使用昨天编译固件也是这个问题,不过不用重启等一分钟又恢复,反复出现

找不到具体触发的原因，更新插件数据有时会出现，我用两天突然会出现

pangfeng161, 2022-02-22:
> > 使用编译不过作为这个问题，不用重启等一分钟又重复一遍
> 
> 找不到具体触发的原因，更新插件有时会出现，我用突然会出现
关闭FullCone-NAT解决,但是不是这问题还需要时间验证

tianiue, 2022-03-01:
> > > 使用编译不过作为这个问题，不用重启等一分钟又重复一遍
> > 
> > 
> > 找不到具体触发的原因，更新插件有时会出现，我用突然会出现
> > 关闭FullCone-NAT解决,但是不是这问题还需要时间验证

18号编译的5.10.100，没有出现这个问题

coolsnowwolf, 2026-09-05:
x86 网络数日后失效只有截图，没有 conntrack、接口状态、内核和服务日志，无法定位，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf