coolsnowwolf/lede #8918: WAN口(pppoe)手动重新连接，带动其他接口一起重新连接

cokyrain, 2022-02-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：WAN口(pppoe)手动重新连接，会带动其他接口一起重新连接。有一定概率使LAN口重启失败，导致网络瘫痪，必须重启。包括以下几种方式：/cgi-bin/luci/admin/network/network点连接，命令ifup wan，ifconfig eth1 down + ifconfig eth1 up，尝试都是如此。

另外，测试手动重新连接其他接口都正常，没有带动其他接口。

### (2) 路由器型号和固件版本

A：路由器型号：PVE下x86，版本[Powered by LuCI Master (git-22.046.49039-d9baa31)](https://github.com/coolsnowwolf/luci) / OpenWrt R22.2.2

### (3) 详细日志

A：
Sun Feb 20 22:16:31 2022 daemon.notice netifd: Interface 'wan_6' is now down
Sun Feb 20 22:16:31 2022 daemon.notice netifd: Network alias '' link is down
Sun Feb 20 22:16:31 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Sun Feb 20 22:16:31 2022 daemon.notice netifd: Interface 'wan_6' is disabled
Sun Feb 20 22:16:31 2022 kern.info kernel: [ 1151.805865] netifd[30248]: segfault at 7fb140741078 ip 000000000041bbd6 sp 00007ffd642c9960 error 4 in netifd[404000+1b000]
Sun Feb 20 22:16:31 2022 kern.info kernel: [ 1151.807971] Code: 01 75 31 48 8b 53 58 be fe 3b 42 00 bf 02 00 00 00 31 c0 e8 01 e6 ff ff 48 89 df e8 23 f0 ff ff c7 83 90 00 00 00 00 00 00 00 <48> 8b 7b 68 5b e9 f1 cf ff ff 5b c3 53 48 8d 9f 48 ff ff ff 83 fa
Sun Feb 20 22:16:31 2022 user.notice ddns-scripts[32523]: ipv4: PID '32523' terminated by 'SIGTERM' at 2022-02-20 22:16
Sun Feb 20 22:16:31 2022 user.notice ddns-scripts[32524]: ipv6: PID '32524' terminated by 'SIGTERM' at 2022-02-20 22:16
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: 8021ad
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: 8021q
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: macvlan
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: veth
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: bridge
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: Network device
Sun Feb 20 22:16:36 2022 user.notice : Added device handler type: tunnel
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.901799] br-lan: port 4(eth5) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.903353] br-lan: port 3(eth4) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.904811] br-lan: port 2(eth3) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.906168] br-lan: port 1(eth0) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.909359] device eth5 left promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.910718] br-lan: port 4(eth5) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.945015] device eth4 left promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.946449] br-lan: port 3(eth4) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.985049] device eth3 left promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1156.986373] br-lan: port 2(eth3) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.025018] device eth0 left promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.026432] br-lan: port 1(eth0) entered disabled state
Sun Feb 20 22:16:36 2022 daemon.notice pppd[30399]: Modem hangup
Sun Feb 20 22:16:36 2022 daemon.err miniupnpd[31626]: Failed to get IP for interface eth2
Sun Feb 20 22:16:36 2022 daemon.warn miniupnpd[31626]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sun Feb 20 22:16:36 2022 daemon.info pppd[30399]: Connect time 3.0 minutes.
Sun Feb 20 22:16:36 2022 daemon.info pppd[30399]: Sent 1248587 bytes, received 5006517 bytes.
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.161838] 8021q: adding VLAN 0 to HW filter on device eth0
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.164374] br-lan: port 1(eth0) entered blocking state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.165769] br-lan: port 1(eth0) entered disabled state
Sun Feb 20 22:16:36 2022 daemon.notice pppd[30399]: Connection terminated.
Sun Feb 20 22:16:36 2022 daemon.info pppd[30399]: Connect time 3.0 minutes.
Sun Feb 20 22:16:36 2022 daemon.info pppd[30399]: Sent 1248587 bytes, received 5006517 bytes.
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.168036] device eth0 entered promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.172939] br-lan: port 1(eth0) entered blocking state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.174345] br-lan: port 1(eth0) entered forwarding state
Sun Feb 20 22:16:36 2022 daemon.notice netifd: Interface 'lan' is enabled
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.181419] 8021q: adding VLAN 0 to HW filter on device eth3
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.184827] br-lan: port 2(eth3) entered blocking state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.186146] br-lan: port 2(eth3) entered disabled state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.190305] device eth3 entered promiscuous mode
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.191738] br-lan: port 2(eth3) entered blocking state
Sun Feb 20 22:16:36 2022 kern.info kernel: [ 1157.193106] br-lan: port 2(eth3) entered forwarding state
[... 79 lines cut ...]
Sun Feb 20 22:16:44 2022 daemon.notice netifd: Interface 'wan_6' is now up
Sun Feb 20 22:16:46 2022 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Feb 20 22:16:47 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of lan (br-lan)
Sun Feb 20 22:16:47 2022 daemon.err nlbwmon[8179]: The netlink receive buffer size of 1048576 bytes will be capped to 212992 bytes
Sun Feb 20 22:16:47 2022 daemon.err nlbwmon[8179]: by the kernel. The net.core.rmem_max sysctl limit needs to be raised to
Sun Feb 20 22:16:47 2022 daemon.err nlbwmon[8179]: at least 1048576 in order to sucessfully set the desired receive buffer size!
Sun Feb 20 22:16:47 2022 daemon.notice miniupnpd[31626]: shutting down MiniUPnPd
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: sendto(udp_shutdown=6): No such device
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[31626]: Failed to broadcast good-bye notifications
Sun Feb 20 22:16:47 2022 daemon.err miniupnpd[8220]: could not open lease file: /var/upnp.leases
Sun Feb 20 22:16:47 2022 daemon.notice miniupnpd[8220]: HTTP listening on port 5000
Sun Feb 20 22:16:47 2022 daemon.notice miniupnpd[8220]: Listening for NAT-PMP/PCP traffic on port 5351
Sun Feb 20 22:16:57 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Sun Feb 20 22:17:07 2022 user.notice firewall: Reloading firewall due to ifup of IPTV (eth2)
Sun Feb 20 22:17:08 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of IPTV (eth2)

cokyrain, 2022-02-21:
附上编译配置：

CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_GRUB_IMAGES=y
CONFIG_PACKAGE_6in4=y
CONFIG_PACKAGE_ip6tables=y
# CONFIG_PACKAGE_iptables-mod-ipsec is not set
CONFIG_PACKAGE_ipv6helper=y
# CONFIG_PACKAGE_kmod-crypto-cbc is not set
# CONFIG_PACKAGE_kmod-crypto-deflate is not set
# CONFIG_PACKAGE_kmod-crypto-des is not set
# CONFIG_PACKAGE_kmod-crypto-echainiv is not set
# CONFIG_PACKAGE_kmod-crypto-hmac is not set
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
# CONFIG_PACKAGE_kmod-crypto-md5 is not set
# CONFIG_PACKAGE_kmod-ipsec is not set
# CONFIG_PACKAGE_kmod-ipt-ipsec is not set
CONFIG_PACKAGE_kmod-ipt-nat6=y
# CONFIG_PACKAGE_kmod-ipt-offload is not set
CONFIG_PACKAGE_kmod-iptunnel=y
# CONFIG_PACKAGE_kmod-iptunnel6 is not set
# CONFIG_PACKAGE_kmod-nf-flow is not set
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-shortcut-fe=y
CONFIG_PACKAGE_kmod-shortcut-fe-cm=y
CONFIG_PACKAGE_kmod-sit=y
# CONFIG_PACKAGE_kmod-tun is not set
# CONFIG_PACKAGE_kmod-wireguard is not set
# CONFIG_PACKAGE_libgmp is not set
# CONFIG_PACKAGE_libminiupnpc is not set
# CONFIG_PACKAGE_libnatpmp is not set
# CONFIG_PACKAGE_libstdcpp is not set
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_btrfs_progs is not set
# CONFIG_PACKAGE_luci-app-diskman_INCLUDE_lsblk is not set
# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
CONFIG_PACKAGE_luci-app-netdata=y
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_SHORTCUT_FE=y
CONFIG_PACKAGE_luci-app-udpxy=y
# CONFIG_PACKAGE_luci-app-vlmcsd is not set
# CONFIG_PACKAGE_luci-app-wireguard is not set
# CONFIG_PACKAGE_luci-app-zerotier is not set
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-proto-ipv6=y
# CONFIG_PACKAGE_luci-proto-wireguard is not set
CONFIG_PACKAGE_netdata=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
# CONFIG_PACKAGE_strongswan is not set
CONFIG_PACKAGE_udpxy=y
# CONFIG_PACKAGE_vlmcsd is not set
# CONFIG_PACKAGE_wireguard-tools is not set
# CONFIG_PACKAGE_zerotier is not set
CONFIG_TARGET_IMAGES_GZIP=y

Tszlam-Cheung, 2022-02-21:
问题相同 Debug过程如下
+ 重新编译 5.15 和5.10 问题重现
+ 重刷旧版 -发现已知20210124-20220222的版本都有这个问题
+ 关闭WAN IPV6 发现问题消失
+ 关闭LAN IPV6 发现问题消失

已知这个问题和IPV6有关 目前采取办法
+ watchcat设定1h
+ Lan侧不下发IPV6

nb12nb34, 2022-02-22:
问题同上，此问题还造成了docker0网关丢失，暂时关闭ipv6

poromony, 2022-02-23:
我试着打开lan接口的桥接，问题似乎解决，还在观察

cokyrain, 2022-02-23:
> 

多lan口，本身就是桥接。你试试连续重连wan几次看看，长时间连接后重连wan有概率正常一两次，有玄学，原因不明。

cokyrain, 2022-02-23:
多lan口，本身就是桥接。你连续重连wan几次试试看。
长时间连接后重连wan有概率正常一两次，有玄学，原因不明。大多时候还是带动其他口重连

Tszlam-Cheung, 2022-02-23:
> 我试着打开lan接口的桥接，问题似乎解决，还在观察
我已经试过了之前 别这么做 我得尝试方法是
bridge multiple interface 取消
单口eth0绑定lan
结果出现得问题是
一定概率ifup wan的 时候 你lan直接挂掉了 然后没有就死在了那里 DHCP下发什么都挂掉 只能static ip访问

moonerick, 2022-04-07:
问题解决了么，我从5.15的某个版本开始遇到类似问题
双wan，由于家里挂了网心云，流量过大，运营商每过不到24H会强制将我的wan重连，重连后大概率出现udpxy，upnp故障，我分析问题在于wan重连后相应接口名称有改动，日志如下：
Tue Mar 29 08:55:47 2022 local0.err udpxy[17439]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17440]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17452]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17453]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17458]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17459]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17460]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:47 2022 local0.err udpxy[17461]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:48 2022 local0.err udpxy[17462]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:48 2022 local0.err udpxy[17463]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:48 2022 local0.err udpxy[17464]: set_multicast: setsockopt MCAST option: ADD: No such device
Tue Mar 29 08:55:48 2022 local0.err udpxy[17465]: set_multicast: setsockopt MCAST option: ADD: No such device
Sun Apr  3 11:24:35 2022 daemon.err miniupnpd[18540]: sendto(udp_notify=6, 192.168.2.1): No such device
Sun Apr  3 11:24:35 2022 daemon.err miniupnpd[18540]: sendto(udp_notify=6, 192.168.2.1): No such device
Sun Apr  3 11:24:35 2022 daemon.err miniupnpd[18540]: sendto(udp_notify=6, 192.168.2.1): No such device

cokyrain, 2022-04-11:
> 问题解决了么，我从5.15的某个版本开始遇到类似问题 双wan，由于家里挂了网心云，流量过大，运营商每过不到24H会强制将我的wan重连，重连后大概率出现udpxy，upnp故障，我分析问题在于wan重连后相应接口名称有改动，日志如下： Tue Mar 29 08:55:47 2022 local0.err udpxy[17439]: set_multicast: setsockopt [...]

没有解决，只能关闭ipv6

violarulan, 2022-05-16:
R4S 5.15同问题

ifup wan 有几率搞崩网络，出问题时会把 br-lan 一并重启，表现为 pppoe 可以正常获取 ipv4，但是 lan 口无法获取 v6（我是删除wan6接口，然后指定lan的v6长度使用的），路由器自身及下边设备均无网络

imkenf, 2023-01-08:
长期以来一直有这个问题，我现在是3个wan口，只要是其中一个断线重连也必定所有wan都会重连，最近更新的源码还会导致mwan异常，必须要重启mwan才能正常nat，估计是多wan需求不多，没人关注这个问题。

coolsnowwolf, 2026-09-05:
当前 netifd 已更新到 2026-07-08 上游版本，远超 issue 使用的 5.10.100/R22.2.2；旧版 netifd 重启时 segfault 不再对应当前代码，关闭为 completed。

closed 2026-09-05 by coolsnowwolf