coolsnowwolf/lede #8920: rt4230w rac2v1k 性能问题

rshi0212, 2022-02-21:
5.4.132 CPUmark 13000多
5.4.175 最新的CPUmark只有8000多了
启动内核日志要10分钟。。。。。。是什么问题啊？

rshi0212, 2022-02-21:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.175 (rshi0212@GPDWIN3-RSHI) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4208-a45b0fa8a)) #0 SMP Sun Feb 20 04:00:11 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line (ignored): 
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996168K/1015808K available (6265K kernel code, 198K rwdata, 1568K rodata, 1024K init, 232K bss, 19640K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000006] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000263] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000283] pid_max: default: 32768 minimum: 301
[    0.000490] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000513] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001586] CPU: Testing write buffer coherency: ok
[    0.002746] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002927] rcu: Hierarchical SRCU implementation.
[    0.003429] smp: Bringing up secondary CPUs ...
[    0.005384] smp: Brought up 1 node, 2 CPUs
[    0.005400] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005411] CPU: All CPU(s) started in SVC mode.
[    0.017479] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.017634] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017662] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.017803] pinctrl core: initialized pinctrl subsystem
[    0.018978] NET: Registered protocol family 16
[    0.019299] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020806] cpuidle: using governor ladder
[    0.020874] cpuidle: using governor menu
[    0.057071] usbcore: registered new interface driver usbfs
[    0.057155] usbcore: registered new interface driver hub
[    0.057230] usbcore: registered new device driver usb
[    0.057291] pps_core: LinuxPPS API ver. 1 registered
[    0.057304] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.057340] PTP clock support registered
[    0.058185] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.059372] clocksource: Switched to clocksource dg_timer
[    0.060259] thermal_sys: Registered thermal governor 'step_wise'
[    0.062940] NET: Registered protocol family 2
[    0.063125] IP idents hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.064269] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.064313] TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.064375] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.064469] TCP: Hash tables configured (established 8192 bind 8192)
[    0.064585] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064624] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064861] NET: Registered protocol family 1
[    0.064912] PCI: CLS 0 bytes, default 64
[    0.066281] workingset: timestamp_bits=14 max_order=18 bucket_order=4
[    0.073379] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 1244 lines cut ...]
[  299.675990] ath10k_pci 0001:01:00.0: rts threshold -1
[  299.677563] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  299.685894] debugfs: File 'virt_if' in directory 'stats' already present!
[  299.702768] wlan0: Created a NSS virtual interface.
[  299.702802] wlan0: RX-Q and tasklet initialized.
[  299.718993] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[... cut ...]
[  306.410708] br-lan: port 3(wlan0) entered disabled state
[  306.415537] device wlan0 entered promiscuous mode
[  306.849466] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  306.849836] br-lan: port 2(wlan1) entered blocking state
[  306.854897] br-lan: port 2(wlan1) entered forwarding state
[  307.657080] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  307.657352] br-lan: port 3(wlan0) entered blocking state
[  307.662622] br-lan: port 3(wlan0) entered forwarding state

rshi0212, 2022-02-21:
Mon Feb 21 12:54:38 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:54:43 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:55:04 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:55:29 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:29 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: disassociated
Mon Feb 21 12:55:30 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb 21 12:55:31 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: authenticated
Mon Feb 21 12:55:31 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: associated (aid 1)
Mon Feb 21 12:55:31 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:31 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d RADIUS: starting accounting session FCCA5A2F81AD4015
Mon Feb 21 12:55:31 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d WPA: pairwise key handshake completed (RSN)
Mon Feb 21 12:55:31 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:32 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:55:36 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:55:41 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:55:46 2022 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 8c:fa:ba:0f:43:2d 39 ack=1
Mon Feb 21 12:55:47 2022 daemon.notice hostapd: wlan0: BEACON-RESP-RX 8c:fa:ba:0f:43:2d 39 00 81263e27e6000000000064
Mon Feb 21 12:55:47 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 00:9e:c8:59:93:e3 68 ack=1
Mon Feb 21 12:55:47 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 00:9e:c8:59:93:e3 0 00 5101570159103f000000640000d400a0648fde48cb0024f607000108d2d686f008000000
Mon Feb 21 12:55:55 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:55 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: disassociated
Mon Feb 21 12:55:56 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb 21 12:55:57 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: authenticated
Mon Feb 21 12:55:57 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: associated (aid 1)
Mon Feb 21 12:55:57 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:57 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d RADIUS: starting accounting session B0F7AEF9116DB44F
Mon Feb 21 12:55:57 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d WPA: pairwise key handshake completed (RSN)
Mon Feb 21 12:55:57 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:55:58 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:56:02 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:56:07 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:56:08 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:56:08 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: disassociated
Mon Feb 21 12:56:09 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb 21 12:56:10 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: authenticated
Mon Feb 21 12:56:10 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: associated (aid 1)
Mon Feb 21 12:56:10 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:56:10 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d RADIUS: starting accounting session CC09CC8CE1397180
Mon Feb 21 12:56:10 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d WPA: pairwise key handshake completed (RSN)
Mon Feb 21 12:56:10 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:56:10 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:56:15 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 12:56:16 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 12:56:16 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: disassociated
Mon Feb 21 12:56:17 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
[... cut ...]
Mon Feb 21 13:29:56 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Mon Feb 21 13:30:12 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 8c:fa:ba:0f:43:2d
Mon Feb 21 13:30:12 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: disassociated
Mon Feb 21 13:30:13 2022 daemon.info hostapd: wlan0: STA 8c:fa:ba:0f:43:2d IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb 21 13:32:23 2022 daemon.warn odhcpd[1753]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

rshi0212, 2022-02-21:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.175 (rshi0212@GPDWIN3-RSHI) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4208-a45b0fa8a)) #0 SMP Sun Feb 20 16:15:52 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line (ignored): 
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996168K/1015808K available (6273K kernel code, 198K rwdata, 1572K rodata, 1024K init, 232K bss, 19640K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000261] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000283] pid_max: default: 32768 minimum: 301
[    0.000495] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000518] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001565] CPU: Testing write buffer coherency: ok
[    0.002722] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002900] rcu: Hierarchical SRCU implementation.
[    0.003403] smp: Bringing up secondary CPUs ...
[    0.005338] smp: Brought up 1 node, 2 CPUs
[    0.005354] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005364] CPU: All CPU(s) started in SVC mode.
[    0.017442] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.017595] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017622] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.017764] pinctrl core: initialized pinctrl subsystem
[    0.018932] NET: Registered protocol family 16
[    0.019254] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020748] cpuidle: using governor ladder
[    0.020816] cpuidle: using governor menu
[    0.056463] usbcore: registered new interface driver usbfs
[    0.056543] usbcore: registered new interface driver hub
[    0.056614] usbcore: registered new device driver usb
[    0.056672] pps_core: LinuxPPS API ver. 1 registered
[    0.056686] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.056720] PTP clock support registered
[    0.057553] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.058722] clocksource: Switched to clocksource dg_timer
[    0.059600] thermal_sys: Registered thermal governor 'step_wise'
[    0.062250] NET: Registered protocol family 2
[    0.062434] IP idents hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.063574] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.063620] TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.063680] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.063774] TCP: Hash tables configured (established 8192 bind 8192)
[    0.063889] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.063930] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064160] NET: Registered protocol family 1
[    0.064215] PCI: CLS 0 bytes, default 64
[    0.065561] workingset: timestamp_bits=14 max_order=18 bucket_order=4
[    0.072608] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 1268 lines cut ...]
[ 1389.853448] Buffer I/O error on dev mtdblock16, logical block 0, async page read
[ 1389.863147] Buffer I/O error on dev mtdblock17, logical block 0, async page read
[ 1389.867751] Buffer I/O error on dev mtdblock17, logical block 0, async page read
[ 1638.273097] device wlan0 left promiscuous mode
[ 1638.273296] br-lan: port 3(wlan0) entered disabled state
[ 1638.321390] ath10k_pci 0000:01:00.0: mac flush null vif, drop 0 queues 0xffff
[... cut ...]
[ 1645.658906] ath10k_pci 0000:01:00.0: rts threshold -1
[ 1645.659250] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 1645.670523] br-lan: port 3(wlan0) entered blocking state
[ 1645.673944] br-lan: port 3(wlan0) entered disabled state
[ 1645.680086] device wlan0 entered promiscuous mode
[ 1646.632397] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1646.634936] br-lan: port 3(wlan0) entered blocking state
[ 1646.637845] br-lan: port 3(wlan0) entered forwarding state

rshi0212, 2022-02-21:
Mon Feb 21 12:14:02 2022 kern.warn kernel: [  253.305685] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Mon Feb 21 12:14:02 2022 kern.warn kernel: [  253.305717] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
Mon Feb 21 12:14:02 2022 kern.info kernel: [  253.390242] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Mon Feb 21 12:14:02 2022 kern.info kernel: [  253.391135] ath10k_pci 0001:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
Mon Feb 21 12:14:02 2022 kern.info kernel: [  253.970538] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Mon Feb 21 12:14:02 2022 kern.debug kernel: [  254.061639] ath: EEPROM regdomain sanitized
Mon Feb 21 12:14:02 2022 kern.debug kernel: [  254.061656] ath: EEPROM regdomain: 0x64
Mon Feb 21 12:14:02 2022 kern.debug kernel: [  254.061666] ath: EEPROM indicates we should expect a direct regpair map
Mon Feb 21 12:14:02 2022 kern.debug kernel: [  254.061689] ath: Country alpha2 being used: 00
Mon Feb 21 12:14:02 2022 kern.debug kernel: [  254.061700] ath: Regpair used: 0x64
Mon Feb 21 12:14:02 2022 kern.err kernel: [  254.064145] debugfs: File 'virt_if' in directory 'stats' already present!
Mon Feb 21 12:14:02 2022 kern.info kernel: [  254.064632] wlan1: Created a NSS virtual interface.
Mon Feb 21 12:14:02 2022 kern.info kernel: [  254.070181] wlan1: RX-Q and tasklet initialized.
Mon Feb 21 12:14:02 2022 kern.info kernel: [  254.087285] batman_adv: B.A.T.M.A.N. advanced 2022.0-openwrt-2 (compatibility version 15) loaded
Mon Feb 21 12:14:02 2022 user.info kernel: [  254.087845] kmodloader: done loading kernel modules from /etc/modules.d/*
Mon Feb 21 12:14:04 2022 daemon.notice procd: /etc/rc.d/S15cpu_freq: vm.min_free_kbytes = 65536
Mon Feb 21 12:14:04 2022 daemon.notice procd: /etc/rc.d/S15cpu_freq: net.netfilter.nf_conntrack_max = 65535
Mon Feb 21 12:14:05 2022 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Feb 21 12:14:05 2022 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Feb 21 12:14:09 2022 kern.info kernel: [  261.831709] ECM init
Mon Feb 21 12:14:09 2022 kern.info kernel: [  261.831767] ECM database jhash random seed: 0x881937da
Mon Feb 21 12:14:09 2022 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: net.bridge.bridge-nf-call-ip6tables = 1
Mon Feb 21 12:14:09 2022 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: net.bridge.bridge-nf-call-iptables = 1
Mon Feb 21 12:14:09 2022 kern.info kernel: [  261.909407] ECM init complete
Mon Feb 21 12:14:09 2022 daemon.notice procd: /etc/rc.d/S19qca-nss-ecm: dev.nss.general.redirect = 1
Mon Feb 21 12:14:10 2022 daemon.notice wpa_supplicant[1704]: Successfully initialized wpa_supplicant
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: 8021ad
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: 8021q
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: macvlan
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: veth
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: bridge
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: Network device
Mon Feb 21 12:14:11 2022 user.notice : Added device handler type: tunnel
Mon Feb 21 12:14:11 2022 kern.warn kernel: [  264.134967] print_req_error: 30 callbacks suppressed
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.134974] blk_update_request: I/O error, dev mtdblock12, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.140203] blk_update_request: I/O error, dev mtdblock12, sector 8 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.150592] blk_update_request: I/O error, dev mtdblock12, sector 16 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.161357] blk_update_request: I/O error, dev mtdblock12, sector 24 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.172480] blk_update_request: I/O error, dev mtdblock12, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Mon Feb 21 12:14:11 2022 kern.warn kernel: [  264.182343] buffer_io_error: 1 callbacks suppressed
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.182347] Buffer I/O error on dev mtdblock12, logical block 0, async page read
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.198609] blk_update_request: I/O error, dev mtdblock12, sector 32 op 0x0:(READ) flags 0x80700 phys_seg 7 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.206121] blk_update_request: I/O error, dev mtdblock12, sector 40 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.216889] blk_update_request: I/O error, dev mtdblock12, sector 48 op 0x0:(READ) flags 0x80700 phys_seg 5 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.227705] blk_update_request: I/O error, dev mtdblock12, sector 56 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.238669] blk_update_request: I/O error, dev mtdblock12, sector 64 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 0
Mon Feb 21 12:14:11 2022 kern.err kernel: [  264.252472] Buffer I/O error on dev mtdblock12, logical block 1, async page read
[... cut ...]
Mon Feb 21 13:26:37 2022 daemon.notice hostapd: wlan0: BEACON-RESP-RX 2a:67:8e:65:04:c5 23 00 8124d9d1ebe5b231fb00640000de237829eda5a36b0000000000011add180050f2020101810003a4000027a4000042435e0062322f0002020101
Mon Feb 21 13:26:38 2022 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS e6:87:90:2b:75:75 25 ack=1
Mon Feb 21 13:26:39 2022 daemon.notice hostapd: wlan0: BEACON-RESP-RX e6:87:90:2b:75:75 25 00 812a2b
Mon Feb 21 13:30:30 2022 daemon.warn odhcpd[1814]: A default route is present but there is no public prefix on lan thus we don't announce a default route!

rshi0212, 2022-02-21:
这两台是512m rom的，还有一台256m rom的

rshi0212, 2022-02-21:
[image]

rshi0212, 2022-02-21:
[image]

rshi0212, 2022-02-21:
为什么没有选择UnblockNeteaseMusic，固件里还有这个UnblockNeteaseMusic？
[image]

Wilson-coder, 2022-02-21:
从系统日志看着似乎是存储区域有坏块了，其它问题不懂~~~

zheshifandian, 2022-02-22:
编译时把block-mount取消选择就好，另外那个包也是要手动取消选择

zheshifandian, 2022-02-22:
5.4性能拉了，那就试试5.10呗

rshi0212, 2022-02-22:
> 编译时把block-mount取消选择就好，另外那个包也是要手动取消选择

这样的话，usb口硬盘、u盘之类的还能用嘛？

rshi0212, 2022-02-22:
> 5.4性能拉了，那就试试5.10呗

怎么改5.10内核编译啊？谢谢指导~

zheshifandian, 2022-02-23:
> > 编译时把block-mount取消选择就好，另外那个包也是要手动取消选择
> 
> 这样的话，usb口硬盘、u盘之类的还能用嘛？

不知道，一般是用不了了

zheshifandian, 2022-02-23:
> > 5.4性能拉了，那就试试5.10呗
> 
> 怎么改5.10内核编译啊？谢谢指导~

target/linux/ipq806x/Makefile        5.4改5.10

rshi0212, 2022-02-23:
> > > 5.4性能拉了，那就试试5.10呗
> > 
> > 
> > 怎么改5.10内核编译啊？谢谢指导~
> 
> target/linux/ipq806x/Makefile 5.4改5.10

改了5.10，编译出错，是驱动问题吗？

arch/arm/boot/compressed/atags_to_fdt.c:63:17: warning: 'get_cell_size' defined but not used [-Wunused-function]
 static uint32_t get_cell_size(const void *fdt)
                 ^~~~~~~~~~~~~
arch/arm/boot/compressed/atags_to_fdt.c:43:12: warning: 'setprop_cell' defined but not used [-Wunused-function]
 static int setprop_cell(void *fdt, const char *node_path,
            ^~~~~~~~~~~~
arch/arm/boot/compressed/atags_to_fdt.c:25:12: warning: 'setprop' defined but not used [-Wunused-function]
 static int setprop(void *fdt, const char *node_path, const char *property,
            ^~~~~~~
  SHIPPED arch/arm/boot/compressed/lib1funcs.S
  AS      arch/arm/boot/compressed/lib1funcs.o
  SHIPPED arch/arm/boot/compressed/ashldi3.S
  AS      arch/arm/boot/compressed/ashldi3.o
  SHIPPED arch/arm/boot/compressed/bswapsdi2.S
  AS      arch/arm/boot/compressed/bswapsdi2.o
  LD      arch/arm/boot/compressed/vmlinux
  OBJCOPY arch/arm/boot/zImage
  Kernel: arch/arm/boot/zImage is ready
make[5]: Nothing to be done for 'Image'.
  DTC     arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb
Error: arch/arm/boot/dts/qcom-ipq8062.dtsi:97.1-6 Label or path nss0 not found
Error: arch/arm/boot/dts/qcom-ipq8062.dtsi:103.1-6 Label or path nss1 not found
FATAL ERROR: Syntax error parsing input tree
make[6]: *** [scripts/Makefile.lib:326: arch/arm/boot/dts/qcom-ipq8062-wg2600hp3.dtb] Error 1
make[5]: *** [Makefile:1359: dtbs] Error 2
make[5]: Leaving directory '/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.10.100'
make[4]: *** [Makefile:30: /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.10.100/.modules] Error 2
make[4]: Leaving directory '/home/rshi0212/rt4230w-lede/target/linux/ipq806x'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/rshi0212/rt4230w-lede/target/linux'
time: target/linux/compile#863.79#57.79#918.83
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/rshi0212/rt4230w-lede'
make[1]: *** [target/Makefile:18: /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/rshi0212/rt4230w-lede'
make: *** [/home/rshi0212/rt4230w-lede/include/toplevel.mk:230: world] Error 2

xfmly, 2022-11-17:
cpu 性能调节优化 Policy2 最大频率改1725000就正常了
[image]

closed 2026-09-05 by coolsnowwolf