coolsnowwolf/lede #8922: 编译错误，高手帮忙看看

keeno1982, 2022-02-22:
[1/266] Building C object library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o
FAILED: library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/bin/aarch64-openwrt-linux-musl-gcc  -I/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/include -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12=mbedtls-2.16.12 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -Wall -Wextra -W -Wdeclaration-after-statement -Wwrite-strings -Wlogical-op -Wshadow -Wmissing-declarations -Wmissing-prototypes -O2 -MD -MT library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -MF library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o.d -o library/CMakeFiles/mbedcrypto_static.dir/armv8ce_aes.c.o -c /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c: In function 'mbedtls_armv8ce_aes_crypt_ecb':
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12440:1: error: inlining failed in call to always_inline 'vaesmcq_u8': target specific option mismatch
 vaesmcq_u8 (uint8x16_t data)
 ^~~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:62:17: note: called from here
             x = vaesmcq_u8( x );
                 ^~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:61:17: note: called from here
             x = vaeseq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:66:13: note: called from here
         x = vaeseq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12426:1: error: inlining failed in call to always_inline 'vaeseq_u8': target specific option mismatch
 vaeseq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:66:13: note: called from here
         x = vaeseq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12447:1: error: inlining failed in call to always_inline 'vaesimcq_u8': target specific option mismatch
 vaesimcq_u8 (uint8x16_t data)
 ^~~~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:75:17: note: called from here
             x = vaesimcq_u8( x );
                 ^~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:74:17: note: called from here
             x = vaesdq_u8( x, k );
                 ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12447:1: error: inlining failed in call to always_inline 'vaesimcq_u8': target specific option mismatch
 vaesimcq_u8 (uint8x16_t data)
 ^~~~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:75:17: note: called from here
             x = vaesimcq_u8( x );
                 ^~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
[... 3 lines cut ...]
                 ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:79:13: note: called from here
         x = vaesdq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
In file included from /home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:30:
/home/dietpi/lede/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/include/arm_neon.h:12433:1: error: inlining failed in call to always_inline 'vaesdq_u8': target specific option mismatch
 vaesdq_u8 (uint8x16_t data, uint8x16_t key)
 ^~~~~~~~~
/home/dietpi/lede/build_dir/target-aarch64_cortex-a53_musl/mbedtls-2.16.12/library/armv8ce_aes.c:79:13: note: called from here
         x = vaesdq_u8( x, k );
             ^~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
[... cut ...]
make[3]: Leaving directory '/home/dietpi/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.23#0.08#0.30
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/dietpi/lede'
make[1]: *** [package/Makefile:110: /home/dietpi/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dietpi/lede'
make: *** [/home/dietpi/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-02-22:
是编译哪个ps3netsrv 这个吧，我这边测试一切正常，x86 5.15内核，mbedtls这个库顺利编译完毕

keeno1982, 2022-02-22:
> 是编译哪个ps3netsrv 这个吧，我这边测试一切正常，x86 5.15内核，mbedtls这个库顺利编译完毕

我先把它去掉看看

saoba1982, 2022-02-26:
我也是这样的错误。请教如何解决？

keeno1982, 2022-03-01:
把/package/libs/mbedtls/Makefile文件里面的
PKG_VERSION:=2.28.0改成2.16.12 就可以过

LGinC, 2022-06-04:
@keeno1982 
```bash
Hash of the downloaded file does not match (file: 294871ab1864a65d0b74325e9219d5bcd6e91c34a3c59270c357bb9ae4d5c393, requested: 6519579b836ed78cc549375c7c18b111df5717e86ca0eeff4cb64b2674f424cc) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/mbedtls-2.16.12.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:177: /home/we/lede/dl/mbedtls-2.16.12.tar.gz] Error 2
make[3]: Leaving directory '/home/we/lede/package/libs/mbedtls'
time: package/libs/mbedtls/compile#0.17#0.09#10.93
    ERROR: package/libs/mbedtls failed to build.
make[2]: *** [package/Makefile:116: package/libs/mbedtls/compile] Error 1
make[2]: Leaving directory '/home/we/lede'
make[1]: *** [package/Makefile:110: /home/we/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/we/lede'
make: *** [/home/we/lede/include/toplevel.mk:230: world] Error 2
```
改成2.16.12后 404

closed 2022-10-31 by aiamadeus