coolsnowwolf/lede #8931: 2月起编译的网件R7000固件无法使用

Suwipe, 2022-02-24:
2月起编译出的R7000固件无法使用，网口正常，能进web界面，但是无法拨号，系统日志报错kmodloader: no module folders for kernel version 5.4.175 found

尝试清除配置，重新配置选择网件R7000后保持默认配置编译问题依旧

详细日志
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.223388] bcma-host-soc 18000000.axi: bus0: Core 8 found: PCIe Gen 2 (manuf 0x4BF, id 0x501, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.233974] bcma-host-soc 18000000.axi: bus0: Core 9 found: PCIe Gen 2 (manuf 0x4BF, id 0x501, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.244573] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.244583] bcma-host-soc 18000000.axi: bus0: Core 10 found: ARM Cortex A9 core (ihost) (manuf 0x4BF, id 0x510, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.256656] bcma-host-soc 18000000.axi: bus0: Core 11 found: USB 2.0 (manuf 0x4BF, id 0x504, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.267074] bcma-host-soc 18000000.axi: bus0: Core 12 found: USB 3.0 (manuf 0x4BF, id 0x505, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.277497] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.277506] bcma-host-soc 18000000.axi: bus0: Core 13 found: SDIO3 (manuf 0x4BF, id 0x503, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.287749] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.287759] bcma-host-soc 18000000.axi: bus0: Core 14 found: I2S (manuf 0x4BF, id 0x834, rev 0x03, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.297836] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.297845] bcma-host-soc 18000000.axi: bus0: Core 15 found: ARM Cortex A9 JTAG (manuf 0x4BF, id 0x506, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.309221] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.309230] bcma-host-soc 18000000.axi: bus0: Core 16 found: Denali DDR2/DDR3 memory controller (manuf 0x4BF, id 0x507, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.322003] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.322012] bcma-host-soc 18000000.axi: bus0: Core 17 found: ROM (manuf 0x4BF, id 0x508, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.332117] bcma-host-soc 18000000.axi: bus0: Core 18 found: NAND flash controller (manuf 0x4BF, id 0x509, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.343762] bcma-host-soc 18000000.axi: bus0: bcma_of_get_irq() failed with rc=-22
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.343771] bcma-host-soc 18000000.axi: bus0: Core 19 found: SPI flash controller (manuf 0x4BF, id 0x50A, rev 0x01, class 0x0)
Thu Feb 24 20:32:37 2022 kern.err kernel: [    2.355215] bcma-host-soc 18000000.axi: bus0: Flash type not supported
Thu Feb 24 20:32:37 2022 kern.debug kernel: [    2.387922] bcma-host-soc 18000000.axi: bus0: Using SPROM revision 11 provided by platform.
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.388198] bgmac_bcma bcma0:3: Found PHY addr: 30 (NOREGS)
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.393809] bgmac_bcma bcma0:3: Support for Roboswitch not implemented
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.412364] b53_common: found switch: BCM53012, rev 3
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.418243] bgmac_bcma bcma0:4: Found PHY addr: 0
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.422988] bgmac_bcma bcma0:4: Support for Roboswitch not implemented
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.431116] bgmac_bcma bcma0:5: Found PHY addr: 0
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.435886] bgmac_bcma bcma0:5: Support for Roboswitch not implemented
Thu Feb 24 20:32:37 2022 kern.err kernel: [    2.442434] bgmac_bcma bcma0:5: Invalid MAC addr: 00:00:00:00:00:00
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.448736] bgmac_bcma bcma0:5: Using random MAC: 5a:d3:7b:88:c5:32
Thu Feb 24 20:32:37 2022 kern.err kernel: [    2.456617] bgmac_bcma bcma0:6: Unsupported core_unit 3
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.461890] bgmac_bcma: probe of bcma0:6 failed with error -524
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.582798] pcie_iproc_bcma bcma0:7: link: UP
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.587277] pcie_iproc_bcma bcma0:7: PCI host bridge to bus 0000:00
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.593580] pci_bus 0000:00: root bus resource [mem 0x08000000-0x0fffffff]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.600470] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.608435] pci 0000:00:00.0: [14e4:8012] type 01 class 0x060400
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.614477] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.624162] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.633857] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x1c may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.643632] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x1c may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.653414] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x3e may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.663208] pci 0000:00:00.0: PME# supported from D0 D3hot D3cold
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.669321] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4c may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.679921] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x3e may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.689709] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x4 may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.699392] pci_bus 0000:00: 1-byte config write to 0000:00:00.0 offset 0xc may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.709073] PCI: bus0: Fast back to back transfers disabled
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.714664] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
Thu Feb 24 20:32:37 2022 kern.warn kernel: [    2.722685] pci_bus 0000:00: 2-byte config write to 0000:00:00.0 offset 0x3e may corrupt adjacent RW1C bits
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.732568] pci 0000:01:00.0: [14e4:4360] type 00 class 0x028000
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.738641] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x00007fff 64bit]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.745555] pci 0000:01:00.0: supports D1 D2
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.750616] PCI: bus1: Fast back to back transfers disabled
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.756220] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.762864] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 01
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.769507] pci 0000:00:00.0: BAR 8: assigned [mem 0x08000000-0x080fffff]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.776320] pci 0000:01:00.0: BAR 0: assigned [mem 0x08000000-0x08007fff 64bit]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.783667] pci 0000:00:00.0: PCI bridge to [bus 01]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.788643] pci 0000:00:00.0:   bridge window [mem 0x08000000-0x080fffff]
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.912795] pcie_iproc_bcma bcma0:8: link: UP
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.917250] pcie_iproc_bcma bcma0:8: PCI host bridge to bus 0001:00
Thu Feb 24 20:32:37 2022 kern.info kernel: [    2.923554] pci_bus 0001:00: root bus resource [mem 0x40000000-0x47ffffff]
[... cut ...]
Thu Feb 24 20:32:45 2022 daemon.notice netifd: wan (2918): executing the following command as root:
Thu Feb 24 20:32:45 2022 daemon.notice netifd: wan (2918): 	mknod /dev/ppp c 108 0
Thu Feb 24 20:32:45 2022 daemon.notice netifd: wan (2918):
Thu Feb 24 20:32:45 2022 daemon.notice netifd: Interface 'wan' is now down
Thu Feb 24 20:32:45 2022 daemon.notice netifd: Interface 'wan' is disabled
Thu Feb 24 20:32:45 2022 daemon.notice netifd: Interface 'wan' is enabled
Thu Feb 24 20:32:45 2022 daemon.notice netifd: Interface 'wan' is setting up now

WYC-2020, 2022-02-28:
建议重新编译 不保留配置刷机看看

coolsnowwolf, 2026-09-05:
R7000 日志显示固件内核与模块目录版本不一致，属于旧构建产物混用/未清理问题；当前目标仍有 R7000 定义，但缺少现行干净构建复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf