coolsnowwolf/lede #8932: 编译错误，高手来看看，谢谢

jxph1123, 2022-02-24:
重新的拉的u源码，MT7621编译的kernel-5.10.100，出现如下错误，高手看看谢谢。

rm -f /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/localversion
/home/etek/lede/scripts/kconfig.pl  + /home/etek/lede/target/linux/generic/config-5.10 /home/etek/lede/target/linux/ramips/mt7621/config-5.10 > /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/etek/lede/.config >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target
/home/etek/lede/scripts/package-metadata.pl kconfig /home/etek/lede/tmp/.packageinfo /home/etek/lede/.config 5.10 > /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.override
/home/etek/lede/scripts/kconfig.pl 'm+' '+' /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.target /dev/null /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.override > /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set
mv /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.old
grep -v INITRAMFS /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.old > /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set
rm -rf /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/modules
cmp -s /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.prev || { cp /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config; cp /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.prev; }
export MAKEFLAGS= ; [ -d /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/user_headers ] || make -C /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100 KCFLAGS="-fmacro-prefix-map=/home/etek/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/etek/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 09:55:56 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/etek/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/etek/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.100 CC="mipsel-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/user_headers headers_install
grep '=[ym]' /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.config.set | LC_ALL=C sort | /home/etek/lede/staging_dir/host/bin/mkhash md5 > /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.vermagic
touch /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/.configured
rm -f /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/vmlinux /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100/System.map
make -C /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100 KCFLAGS="-fmacro-prefix-map=/home/etek/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/etek/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 09:55:56 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/etek/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/etek/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.100 CC="mipsel-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100'
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:480:warning: multi-line strings not supported
#
# No change to .config
#
make[5]: Leaving directory '/home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100'
make -C /home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100 KCFLAGS="-fmacro-prefix-map=/home/etek/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/etek/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 09:55:56 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/etek/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/etek/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.100 CC="mipsel-openwrt-linux-musl-gcc" vmlinux vmlinuz modules
make[5]: Entering directory '/home/etek/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/net/ethernet/ralink/mtk_eth_soc.o
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
drivers/net/ethernet/ralink/mtk_eth_soc.h:533:15: warning: 'struct flow_offload_hw_path' declared inside parameter list will not be visible outside of this definition or declaration
        struct flow_offload_hw_path *src,
               ^~~~~~~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1519:10: warning: 'struct flow_offload_hw_path' declared inside parameter list will not be visible outside of this definition or declaration
   struct flow_offload_hw_path *src,
          ^~~~~~~~~~~~~~~~~~~~
drivers/net/ethernet/ralink/mtk_eth_soc.c: In function 'fe_flow_offload':
drivers/net/ethernet/ralink/mtk_eth_soc.c:1524:9: error: dereferencing pointer to incomplete type 'struct flow_offload_hw_path'
  if (src->dev != dest->dev)
         ^~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1529:44: error: passing argument 4 of 'mtk_flow_offload' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtk_flow_offload(priv, type, flow, src, dest);
                                            ^~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
drivers/net/ethernet/ralink/mtk_eth_soc.h:533:37: note: expected 'struct flow_offload_hw_path *' but argument is of type 'struct flow_offload_hw_path *'
        struct flow_offload_hw_path *src,
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
drivers/net/ethernet/ralink/mtk_eth_soc.c:1529:49: error: passing argument 5 of 'mtk_flow_offload' from incompatible pointer type [-Werror=incompatible-pointer-types]
  return mtk_flow_offload(priv, type, flow, src, dest);
                                                 ^~~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:41:
[... cut ...]
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 1
make[2]: Leaving directory '/home/etek/lede'
target/Makefile:18: recipe for target '/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile' failed
make[1]: *** [/home/etek/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/etek/lede'
/home/etek/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

WYC-2020, 2022-02-24:
明天有时间看下

jxph1123, 2022-02-25:
重新编译了好几次，都是这里编译不过，期待大佬修复。

WYC-2020, 2022-02-25:
[image]
这样改下，因为 5.10 根本就没有打 640和641 补丁 是不能开启这个OFFLOAD的，这个和这个提交有关https://github.com/coolsnowwolf/lede/commit/29d4ee71ed4ce51fceeb0e5b3b85555f7f81ede0 ，不知道是不是漏提交patches，你这样改暂时能编译通过，功能测试一下  @jxph1123

WYC-2020, 2022-02-25:
https://github.com/WYC-2020/lede/commit/ba7cd1d4a1454da89a7347262900af5fae41a638 @jxph1123

jxph1123, 2022-02-25:
感谢大佬

pangfeng161, 2022-02-25:
也碰到类似问题  x86 5.15.25编译出现错误make[1]: *** [/home/pangzhao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pangzhao/lede'
Build failed - please re-run with -j1 to see the real error message
/home/pangzhao/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 1

closed 2022-03-01 by Beginner-Go