coolsnowwolf/lede #8933: Rockchip R4s 5.15编译报错

GreatMichaelLee, 2022-02-25:
shortcut-fe好像和新内核不适配？ ps: 已rm -rf ./tmp && make clean后编译的。

```
make[3]: Entering directory '/home/builder/lede_r4s/package/lean/shortcut-fe/shortcut-fe'
rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/.built
touch /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/.built_check
make -j1 -C "/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25" KCFLAGS="-fmacro-prefix-map=/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/builder/lede_r4s/staging_dir/host/include -I/home/builder/lede_r4s/staging_dir/hostpkg/include -I/home/builder/lede_r4s/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 17:18:43 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/builder/lede_r4s/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/builder/lede_r4s/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers" KERNELRELEASE=5.15.25  M="/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y  modules
make[4]: Entering directory '/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25'
  CC [M]  /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.o
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_udp':
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.c:1477:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.c: In function 'sfe_ipv4_recv_tcp':
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.c:2050:5: error: 'struct sk_buff' has no member named 'fast_forwarded'
  skb->fast_forwarded = 1;
     ^~
make[5]: *** [scripts/Makefile.build:277: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/sfe_ipv4.o] Error 1
make[4]: *** [Makefile:1868: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe] Error 2
make[4]: Leaving directory '/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25'
make[3]: *** [Makefile:92: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_r4s/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#1.04#0.24#1.25
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_r4s'
make[1]: *** [package/Makefile:110: /home/builder/lede_r4s/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_r4s'
make: *** [/home/builder/lede_r4s/include/toplevel.mk:230：world] 错误 2
```

WYC-2020, 2022-02-25:
5.15 l大佬的是不支持的 没有修复，如果需要 我哪里是修复了的

GreatMichaelLee, 2022-02-25:
> 5.15 l大佬的是不支持的 没有修复，如果需要 我哪里是修复了的

你的修复branch/commit id是哪个啊....

closed 2022-02-25 by coolsnowwolf