coolsnowwolf/lede #8934: R2S R4S 5.15内核编译报错

ohbaby30, 2022-02-25:
touch /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/.image
make -C image compile install TARGET_BUILD=
make[5]: Entering directory '/home/ohbaby/r4s15/lede/target/linux/rockchip/image'
cp -fpR /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/COPYING /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/COPYING.linux
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts -I/home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/include -I/home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/include/ -undef -D__DTS__  -o /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts
aarch64-openwrt-linux-musl-cpp: error: /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
make[5]: *** [Makefile:78: /home/ohbaby/r4s15/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb] Error 1
make[5]: Leaving directory '/home/ohbaby/r4s15/lede/target/linux/rockchip/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/ohbaby/r4s15/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/ohbaby/r4s15/lede/target/linux'
time: target/linux/install#7.49#3.08#12.52
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/ohbaby/r4s15/lede'
make[1]: *** [target/Makefile:19: /home/ohbaby/r4s15/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/ohbaby/r4s15/lede'
make: *** [/home/ohbaby/r4s15/lede/include/toplevel.mk:230：world] 错误 2

重新拉取过了

MagnyCopper, 2022-02-25:
rockchip-linux/rkbin 404了?

WYC-2020, 2022-02-25:
linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts: No such file or directory 这里不是提示了

ohbaby30, 2022-02-25:
> linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts: No such file or directory 这里不是提示了

大佬 我应该怎么做呢？复制一份r2s的dts然后改名？

WYC-2020, 2022-02-25:
等会看下

ohbaby30, 2022-02-25:
> 等会看下

好的谢谢大佬，而且我编译r4s也是一样的错误，照道理r4s用不到rk3328的dts啊

WYC-2020, 2022-02-25:
https://github.com/WYC-2020/lede/commit/bc9c930484816954a0951018ae765001a83e7c53
@ohbaby30    还有很多patches漏掉了的 ，建议时不要追新用5.10，我这个提交能保证编译通过，还有rk3399 这些patch是没有的，这个等他们更新吧，懒得折腾

WYC-2020, 2022-02-25:
https://github.com/WYC-2020/lede/commit/78ac94189fa5906afe688a62c868b8ea27777290
@ohbaby30  这个也更新一下，原来的源作者跑路了

ohbaby30, 2022-02-25:
> [WYC-2020/lede@78ac941](https://github.com/WYC-2020/lede/commit/78ac94189fa5906afe688a62c868b8ea27777290) @ohbaby30 这个也更新一下，原来的源作者跑路了

好的 谢谢大佬

closed 2022-02-25 by AmadeusGhost