coolsnowwolf/lede #8935: 如何进入failsafe模式？

devioa, 2022-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如何进入failsafe模式？
今天合并了[ae7f2dbf407b0b58252b0a2b4cfb784e57634fdb](https://github.com/coolsnowwolf/lede/commit/ae7f2dbf407b0b58252b0a2b4cfb784e57634fdb)的源码，因为注释掉了[zzz-default-settings#L34](https://github.com/coolsnowwolf/lede/blob/eb2644d09af43969cb611620b38b36c214714286/package/lean/default-settings/files/zzz-default-settings#L34)导致root密码为空，但是空密码也提示拒绝访问。
尝试了网上说的openwrt failsafe进入方法，但是各种花式戳reset键也没反应：本机ip改为192.168.1.2网关192.168.1.1，在system灯快闪后按下reset按钮，路由器处于system灯更快的闪烁状态，但是网关ping不通，ssh也连不上。
请问有没有什么方法解决？

### (2) 路由器型号和固件版本

A：AX3600

### (3) 详细日志

A：None

WYC-2020, 2022-02-28:
找进入方法，还不如来一个重刷快，现在都安装了不死鸟的嘛 随便刷

devioa, 2022-02-28:
> 找进入方法，还不如来一个重刷快，现在都安装了不死鸟的嘛 随便刷

因为当初是按恩山的教程刷的，没有不死，而且我也没找到AX3600不死

WYC-2020, 2022-03-01:
这...，那可能只有拆机ttl了 @devioa

devioa, 2022-03-02:
> 这...，那可能只有拆机ttl了 @devioa

确实，已经拆机TTL重新刷好了。而且我顺便又试了下进failsafe，通过192.168.1.1依然不行，但是ttl能进，主要是电脑根本ping不通，找不到192.168.1.1的主机，所以就进不去

closed 2022-03-02 by devioa