coolsnowwolf/lede #8936: Actions编译到package/kernel/ksmbd报错

lon91ong, 2022-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue 

最最相似的 #8642 但是平台不同, 具体错误类型也不一样

## 2. 详细叙述

### (1) 具体问题

A：Actions编译到package/kernel/ksmbd报错退出, 最近几天的更新引起的, 之前编译过5.4.175固件成功过.

### (2) 路由器型号和固件版本

A：硬改B70 SPI闪存版, 固件版本5.4.175

### (3) 详细日志

A：
```
make[3]: Entering directory '/workdir/openwrt/package/kernel/ksmbd'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.ksmbd-3.4.3.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "ksmbd-3.4.3.tar.gz" "a910c55d9e6924775e00504eddd00b49788603af29d0772cb9fb6722c189f628" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.4.3?"    '
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/.built_check
make -C /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.175 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 17:18:43 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.175 M="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.175'
make[5]: *** No rule to make target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/ksmbd_spnego_negtokeninit.asn1.c', needed by '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/ksmbd_spnego_negtokeninit.asn1.o'.  Stop.
Makefile:1734: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.175'
make[4]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3] Error 2
Makefile:73: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/.built' failed
make[3]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/ksmbd-3.4.3/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#1.12#0.39#1.36
make[2]: *** [package/kernel/ksmbd/compile] Error 1
    ERROR: package/kernel/ksmbd failed to build.
package/Makefile:114: recipe for target 'package/kernel/ksmbd/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
```

LondyGhost, 2022-02-25:
我编译D7800也遇到了  问题是我编译B70并不会出现ksmbd  是在毫无头绪。都是选上了 cifsd

LondyGhost, 2022-02-25:
`SHELL= flock /workdir/openwrt/tmp/.glib-2.70.0.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "glib-2.70.0.tar.xz" "200d7df811c5ba634afbf109f14bb40ba7fde670e89389885da14e27c0840742" "" "@GNOME/glib/2.70"    '
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.21#0.10#0.27
make[3]: Entering directory '/workdir/openwrt/package/kernel/ksmbd'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.ksmbd-3.4.3.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "ksmbd-3.4.3.tar.gz" "a910c55d9e6924775e00504eddd00b49788603af29d0772cb9fb6722c189f628" "" "https://codeload.github.com/cifsd-team/cifsd/tar.gz/3.4.3?"    '
rm -f /workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ksmbd-3.4.3/.built
touch /workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ksmbd-3.4.3/.built_check
make -C /workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi=target-arm_cortex-a15+neon-vfpv4_musl_eabi" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Feb 24 17:18:43 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.175 M="/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ksmbd-3.4.3" EXTRA_CFLAGS="-DCONFIG_SMB_INSECURE_SERVER=1" CONFIG_SMB_INSECURE_SERVER=y CONFIG_SMB_SERVER=m modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175'
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "sprint_oid" [/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ksmbd-3.4.3/ksmbd.ko] undefined!
make[5]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175'
make[4]: *** [Makefile:1647: modules] Error 2
make[3]: *** [Makefile:76: /workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ksmbd-3.4.3/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.89#0.32#1.19
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.`

LondyGhost, 2022-02-26:
> > 我编译D7800也遇到了 问题是我编译B70并不会出现ksmbd错误 ，毫无头绪。都是选上了 cifsd。 我是本地编译和云编译的 Ubuntu 20.04和21.10都会 应该不是编译环境问题。
> 
> 我的B70是硬改了SPI闪存的，难道是闪存规格的问题

不太可能吧 D7800也是NAND的啊

lon91ong, 2022-02-27:
> 我编译D7800也遇到了 问题是我编译B70并不会出现ksmbd错误 ，毫无头绪。都是选上了 cifsd。 我是本地编译和云编译的 Ubuntu 20.04和21.10都会 应该不是编译环境问题。

cifsd是哪个？是ksmbd的依赖吗？我的config好像没有

lon91ong, 2022-02-27:
今天编译正常了, 没有报错了！

closed 2022-02-27 by lon91ong

LondyGhost, 2022-02-27:
> 今天编译正常了, 没有报错了！

怎么解决的？