coolsnowwolf/lede #8938: 在线编译进行至protobuf时出现x509证书错误

Cptmacmillan2022007, 2022-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

2022-02-25T06:23:53.0106878Z Building targets
2022-02-25T06:23:53.0256787Z go: downloading github.com/golang/protobuf v1.5.2
2022-02-25T06:23:53.0318062Z go: downloading google.golang.org/protobuf v1.27.1
2022-02-25T06:23:53.8427311Z go: downloading github.com/pires/go-proxyproto v0.6.1
2022-02-25T06:23:53.8446486Z go: downloading golang.org/x/sys v0.0.0-20220128215802-99c3d69c2c27
2022-02-25T06:23:54.6176160Z go: downloading google.golang.org/grpc v1.44.0
2022-02-25T06:23:55.4279224Z go: downloading github.com/lucas-clemente/quic-go v0.25.0
2022-02-25T06:23:55.4287312Z go: downloading golang.org/x/net v0.0.0-20220127200216-cd36cc0744dd
2022-02-25T06:23:56.2333122Z go: downloading go.starlark.net v0.0.0-20211203141949-70c0e40ae128
2022-02-25T06:23:56.9824784Z go: downloading golang.org/x/crypto v0.0.0-20220128200615-198e4374d7ed
2022-02-25T06:23:57.7428483Z go: downloading github.com/xtls/go v0.0.0-20210920065950-d4af136d3672
2022-02-25T06:23:57.7445779Z go: downloading github.com/refraction-networking/utls v1.0.0
2022-02-25T06:23:58.5334847Z go: downloading github.com/gorilla/websocket v1.4.2
2022-02-25T06:23:58.5397394Z go: downloading github.com/ghodss/yaml v1.0.1-0.20190212211648-25d852aebe32
2022-02-25T06:23:59.3201809Z go: downloading github.com/pelletier/go-toml v1.9.4
2022-02-25T06:23:59.3305177Z go: downloading github.com/seiflotfy/cuckoofilter v0.0.0-20201222105146-bc6005554a0c
2022-02-25T06:24:00.1241189Z go: downloading github.com/v2fly/ss-bloomring v0.0.0-20210312155135-28617310f63e
2022-02-25T06:24:18.2753751Z ../../infra/conf/serial/loader.go:8:2: github.com/ghodss/yaml@v1.0.1-0.20190212211648-25d852aebe32: Get "https://goproxy.baidu.com/github.com/ghodss/yaml/@v/v1.0.1-0.20190212211648-25d852aebe32.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:59Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2754833Z ../../common/serial/typed_message.go:7:2: github.com/golang/protobuf@v1.5.2: Get "https://goproxy.baidu.com/github.com/golang/protobuf/@v/v1.5.2.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:53Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2755753Z ../../transport/internet/websocket/connection.go:8:2: github.com/gorilla/websocket@v1.4.2: Get "https://goproxy.baidu.com/github.com/gorilla/websocket/@v/v1.4.2.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:59Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2756677Z ../../app/dns/nameserver_quic.go:10:2: github.com/lucas-clemente/quic-go@v0.25.0: Get "https://goproxy.baidu.com/github.com/lucas-clemente/quic-go/@v/v0.25.0.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2757912Z ../../infra/conf/serial/loader.go:9:2: github.com/pelletier/go-toml@v1.9.4: Get "https://goproxy.baidu.com/github.com/pelletier/go-toml/@v/v1.9.4.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:24:00Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2758838Z ../../transport/internet/system_listener.go:8:2: github.com/pires/go-proxyproto@v0.6.1: Get "https://goproxy.baidu.com/github.com/pires/go-proxyproto/@v/v0.6.1.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:54Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2759936Z ../../transport/internet/tls/tls.go:6:2: github.com/refraction-networking/utls@v1.0.0: Get "https://goproxy.baidu.com/github.com/refraction-networking/utls/@v/v1.0.0.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:58Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2760975Z ../../common/antireplay/replayfilter.go:7:2: github.com/seiflotfy/cuckoofilter@v0.0.0-20201222105146-bc6005554a0c: Get "https://goproxy.baidu.com/github.com/seiflotfy/cuckoofilter/@v/v0.0.0-20201222105146-bc6005554a0c.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:24:00Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2761992Z ../../common/antireplay/bloomring.go:6:2: github.com/v2fly/ss-bloomring@v0.0.0-20210312155135-28617310f63e: Get "https://goproxy.baidu.com/github.com/v2fly/ss-bloomring/@v/v0.0.0-20210312155135-28617310f63e.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:24:00Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2763248Z ../../transport/internet/tcp/hub.go:9:2: github.com/xtls/go@v0.0.0-20210920065950-d4af136d3672: Get "https://goproxy.baidu.com/github.com/xtls/go/@v/v0.0.0-20210920065950-d4af136d3672.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:58Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2764371Z ../../app/router/condition.go:6:2: go.starlark.net@v0.0.0-20211203141949-70c0e40ae128: Get "https://goproxy.baidu.com/go.starlark.net/@v/v0.0.0-20211203141949-70c0e40ae128.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2765433Z ../../app/router/condition.go:7:2: go.starlark.net@v0.0.0-20211203141949-70c0e40ae128: Get "https://goproxy.baidu.com/go.starlark.net/@v/v0.0.0-20211203141949-70c0e40ae128.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2766585Z ../../proxy/shadowsocks/config.go:11:2: golang.org/x/crypto@v0.0.0-20220128200615-198e4374d7ed: Get "https://goproxy.baidu.com/golang.org/x/crypto/@v/v0.0.0-20220128200615-198e4374d7ed.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:57Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2767592Z ../../proxy/shadowsocks/config.go:12:2: golang.org/x/crypto@v0.0.0-20220128200615-198e4374d7ed: Get "https://goproxy.baidu.com/golang.org/x/crypto/@v/v0.0.0-20220128200615-198e4374d7ed.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:57Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2768781Z ../../common/ocsp/ocsp.go:11:2: golang.org/x/crypto@v0.0.0-20220128200615-198e4374d7ed: Get "https://goproxy.baidu.com/golang.org/x/crypto/@v/v0.0.0-20220128200615-198e4374d7ed.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:57Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2769785Z ../../proxy/vmess/encoding/auth.go:8:2: golang.org/x/crypto@v0.0.0-20220128200615-198e4374d7ed: Get "https://goproxy.baidu.com/golang.org/x/crypto/@v/v0.0.0-20220128200615-198e4374d7ed.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:57Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2770782Z ../../common/protocol/dns/io.go:7:2: golang.org/x/net@v0.0.0-20220127200216-cd36cc0744dd: Get "https://goproxy.baidu.com/golang.org/x/net/@v/v0.0.0-20220127200216-cd36cc0744dd.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2771872Z ../../app/dns/nameserver_quic.go:12:2: golang.org/x/net@v0.0.0-20220127200216-cd36cc0744dd: Get "https://goproxy.baidu.com/golang.org/x/net/@v/v0.0.0-20220127200216-cd36cc0744dd.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2772892Z ../../transport/internet/http/hub.go:11:2: golang.org/x/net@v0.0.0-20220127200216-cd36cc0744dd: Get "https://goproxy.baidu.com/golang.org/x/net/@v/v0.0.0-20220127200216-cd36cc0744dd.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:56Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2774497Z ../../transport/internet/filelocker_other.go:9:2: golang.org/x/sys@v0.0.0-20220128215802-99c3d69c2c27: Get "https://goproxy.baidu.com/golang.org/x/sys/@v/v0.0.0-20220128215802-99c3d69c2c27.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:54Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2775399Z ../../app/commander/commander.go:10:2: google.golang.org/grpc@v1.44.0: Get "https://goproxy.baidu.com/google.golang.org/grpc/@v/v1.44.0.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:55Z is after 2021-12-09T23:59:5 [... cut ...]
2022-02-25T06:24:18.2785236Z ../../common/serial/typed_message.pb.go:10:2: google.golang.org/protobuf@v1.27.1: Get "https://goproxy.baidu.com/google.golang.org/protobuf/@v/v1.27.1.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:53Z is after 2021-12-09T23:59:59Z
2022-02-25T06:24:18.2786138Z ../../common/serial/typed_message.pb.go:11:2: google.golang.org/protobuf@v1.27.1: Get "https://goproxy.baidu.com/google.golang.org/protobuf/@v/v1.27.1.zip": x509: certificate has expired or is not yet valid: current time 2022-02-25T06:23:53Z is after 2021-12-09T23:59:59Z

Cathgao, 2022-02-26:
[image]
证书过期多久了都

Cptmacmillan2022007, 2022-02-26:
> [image] 证书过期多久了都

对就是这个啊

closed 2022-02-27 by Cptmacmillan2022007