coolsnowwolf/lede #8939: 最新源码编译出错cannot find dependency antfs-mount for automount

lqhorochi, 2022-02-25:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency antfs-mount for automount
 * pkg_hash_fetch_best_installation_candidate: Packages for automount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/kenneth/lede'
make[1]: *** [package/Makefile:111: /home/kenneth/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/kenneth/lede'
make: *** [/home/kenneth/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-02-25:
几百年的老问题了说过怎么解决的

GreatMichaelLee, 2022-02-26:
> Collected errors:
>
> * pkg_hash_check_unresolved: cannot find dependency antfs-mount for automount
> * pkg_hash_fetch_best_installation_candidate: Packages for automount found, but incompatible with the [...]

make menuconfig里吧antfs取消掉。

GreatMichaelLee, 2022-02-26:
1. Ultilities->Filesystem->antfs-mount 取消勾选
2. Kernel Modules-> Filesystems->kmod-fs-antfs, 取消勾选

以上按顺序执行，希望以上能切实帮到楼主，分享传播自己的exp, 而不是一句“几百年老问题”打发你，哈哈哈，

WYC-2020, 2022-02-26:
@GreatMichaelLee 这难道不是几百年的老问题，相同issue 不知道搜索 ，你知道你说你的 请不要讽刺别人

jt8823, 2022-02-26:
> 1. Ultilities->Filesystem->antfs-mount 取消勾选
> 2. Kernel Modules-> Filesystems->kmod-fs-antfs, 取消勾选
> 
> 以上按顺序执行，希望以上能切实帮到楼主，分享传播自己的exp, 而不是一句“几百年老问题”打发你，哈哈哈，

你好   我也遇到了相同的问题, 同样报错,   
我找到了antfs-mount 和kmod-fs-antfs 选项,默认就没有勾选,  
但还是编译不成功啊

jt8823, 2022-02-26:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency antfs-mount for automount
 * pkg_hash_fetch_best_installation_candidate: Packages for automount found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/jt8823/lede'
make[1]: *** [package/Makefile:111: /home/jt8823/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/jt8823/lede'
make: *** [/home/jt8823/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-02-26:
删除 build 目录里面的automount-1文件夹 @jt8823

lqhorochi, 2022-02-26:
> 1. Ultilities->Filesystem->antfs-mount 取消勾选
> 2. Kernel Modules-> Filesystems->kmod-fs-antfs, 取消勾选
> 
> 以上按顺序执行，希望以上能切实帮到楼主，分享传播自己的exp, 而不是一句“几百年老问题”打发你，哈哈哈，

谢谢大佬!!!

XinSSS, 2022-02-28:
@WYC-2020 感谢...确实删掉这个就好了

是不是和之前的commit有关? 我记得前几天我看了下config里面这个配置是yes, 今天再看又没有勾选了

closed 2022-02-28 by AmadeusGhost