coolsnowwolf/lede #8940: 2022/2/26 shortcut-fe of IPQ806x 编译失败

rshi0212, 2022-02-26:
time: package/feeds/luci/luci-app-ttyd/compile#0.30#0.02#0.28
make[3]: Entering directory '/home/rshi0212/rt4230w-lede/feeds/packages/net/pdnsd-alt'
mkdir -p /home/rshi0212/rt4230w-lede/dl
SHELL= flock /home/rshi0212/rt4230w-lede/tmp/.pdnsd-1.2.9b-par.tar.xz.flock -c '        /home/rshi0212/rt4230w-lede/scripts/download.pl "/home/rshi0212/rt4230w-lede/dl" "pdnsd-1.2.9b-par.tar.xz" "e3e9c56cf91b12d8db73def2c247be2f726a052bed012f7a1e48946375f8e478" "" || (  /home/rshi0212/rt4230w-lede/scripts/dl_github_archive.py --dl-dir="/home/rshi0212/rt4230w-lede/dl" --url="https://github.com/shadowsocks/pdnsd.git" --version="a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d" --subdir="pdnsd-1.2.9b-par" --source="pdnsd-1.2.9b-par.tar.xz" --hash="e3e9c56cf91b12d8db73def2c247be2f726a052bed012f7a1e48946375f8e478" || (     echo "Checking out files from the git repository..."; mkdir -p /home/rshi0212/rt4230w-lede/tmp/dl && cd /home/rshi0212/rt4230w-lede/tmp/dl && rm -rf pdnsd-1.2.9b-par && [ \! -d pdnsd-1.2.9b-par ] && git clone  https://github.com/shadowsocks/pdnsd.git pdnsd-1.2.9b-par && (cd pdnsd-1.2.9b-par && git checkout a8e46ccba7b0fa2230d6c42ab6dcd92926f6c21d && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd pdnsd-1.2.9b-par && git log -1 --format='\''@%ct'\''` && rm -rf pdnsd-1.2.9b-par/.git &&        tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c pdnsd-1.2.9b-par |  xz -zc -7e > /home/rshi0212/rt4230w-lede/tmp/dl/pdnsd-1.2.9b-par.tar.xz && mv /home/rshi0212/rt4230w-lede/tmp/dl/pdnsd-1.2.9b-par.tar.xz /home/rshi0212/rt4230w-lede/dl/ && rm -rf pdnsd-1.2.9b-par; );  )    '
mkdir -p /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/rshi0212/rt4230w-lede/tmp/.root-copy.flock -c 'cp -fpR /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pdnsd-1.2.9b-par/.pkgdir/pdnsd-alt/. /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.pdnsd-alt_installed
echo "pdnsd-alt" >> /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/pdnsd-alt.default.install
make[3]: Leaving directory '/home/rshi0212/rt4230w-lede/feeds/packages/net/pdnsd-alt'
time: package/feeds/packages/pdnsd-alt/compile#0.12#0.02#0.12
make[3]: Entering directory '/home/rshi0212/rt4230w-lede/package/lean/shortcut-fe/shortcut-fe'
rm -f /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/.built
touch /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/.built_check
make -j1 -C "/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175" KCFLAGS="-fmacro-prefix-map=/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi=target-arm_cortex-a15+neon-vfpv4_musl_eabi" HOSTCFLAGS="-O2 -I/home/rshi0212/rt4230w-lede/staging_dir/host/include -I/home/rshi0212/rt4230w-lede/staging_dir/hostpkg/include -I/home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Feb 25 15:17:10 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/rshi0212/rt4230w-lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/rshi0212/rt4230w-lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/cryptodev-linux.symvers /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/gpio-button-hotplug.symvers /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/qca-mcs.symvers /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/qca-nss-crypto.symvers /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/qca-nss-gmac.symvers /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/symvers/qca-nss-drv.symvers" KERNELRELEASE=5.4.175  M="/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y EXTRA_CFLAGS+="-DCONFIG_SFE_ECM" CONFIG_SFE_ECM=y modules
make[4]: Entering directory '/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175'
  CC [M]  /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/sfe_cm.o
/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/sfe_cm.c: In function 'sfe_cm_post_routing':
/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/sfe_cm.c:314:14: error: unused variable 'net' [-Werror=unused-variable]
  struct net *net=NULL;
              ^~~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:262: /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/sfe_cm.o] Error 1
make[4]: *** [Makefile:1734: /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe] Error 2
make[4]: Leaving directory '/home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/linux-5.4.175'
make[3]: *** [Makefile:92: /home/rshi0212/rt4230w-lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/rshi0212/rt4230w-lede/package/lean/shortcut-fe/shortcut-fe'
time: package/lean/shortcut-fe/shortcut-fe/compile#0.81#0.11#0.88
    ERROR: package/lean/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/rshi0212/rt4230w-lede'
make[1]: *** [package/Makefile:110: /home/rshi0212/rt4230w-lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rshi0212/rt4230w-lede'
make: *** [/home/rshi0212/rt4230w-lede/include/toplevel.mk:230: world] Error 2
rshi0212@GPDWIN3-RSHI:~/rt4230w-lede$

Dawneng, 2022-02-26:
x86 平台也是这里报错

tianiue, 2022-02-26:
> x86 平台也是这里报错

同样报错
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/init.o
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/init.c: In function 'ath9k_of_init':
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/init.c:650:8: error: too few arguments to function 'of_get_mac_address'
  mac = of_get_mac_address(np);
        ^~~~~~~~~~~~~~~~~~
In file included from /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/of_net.h:3,
                 from /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/init.c:24:
./include/linux/of_net.h:25:19: note: declared here
 static inline int of_get_mac_address(struct device_node *np, u8 *mac)
                   ^~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:277: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/init.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1868: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/mac80211'
make[3]: *** [Makefile:584: /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
time: package/kernel/mac80211/regular/compile#1.49#0.41#1.83
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-26:
@rshi0212 那是l大拉取我的提交的时候 有一个提交没有拉去，现在应该可以，你更新一下

WYC-2020, 2022-02-26:
@tianiue  瞎搞你这个是自己驱动出错 和sfe 没关系

tianiue, 2022-02-26:
> @tianiue 瞎搞你这个是自己驱动出错 和sfe 没关系

找到问题了，内核改成5.15，不支持很多驱动，我修改了脚本，可以了

liaowubinf, 2022-03-30:
@WYC-2020  too few arguments to function 'of_get_mac_address'  大佬，  这个问题最近才出现，之前编译没问题， 要怎么弄呀
drivers/net/ethernet/ralink/mtk_eth_soc.c: In function 'fe_init':
drivers/net/ethernet/ralink/mtk_eth_soc.c:1405:13: error: too few arguments to function 'of_get_mac_address'
  mac_addr = of_get_mac_address(priv->dev->of_node);
             ^~~~~~~~~~~~~~~~~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:26:
./include/linux/of_net.h:14:12: note: declared here
 extern int of_get_mac_address(struct device_node *np, u8 *mac);
            ^~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:262: drivers/net/ethernet/ralink/mtk_eth_soc.o] Error 1
make[8]: *** [scripts/Makefile.build:[49](https://github.com/liaowubinf/Phicomm-K2P-Openwrt-Autobuild/runs/5750665029?check_suite_focus=true#step:11:49)7: drivers/net/ethernet/ralink] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/ethernet] Error 2
make[5]: Leaving directory '/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
make[4]: Leaving directory '/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/target/linux/ramips'
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[3]: Leaving directory '/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/target/linux'
time: target/linux/compile#6.26#2.70#8.[52](https://github.com/liaowubinf/Phicomm-K2P-Openwrt-Autobuild/runs/5750665029?check_suite_focus=true#step:11:52)
make[5]: *** [Makefile:1734: drivers] Error 2
make[4]: *** [Makefile:25: /home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7[62](https://github.com/liaowubinf/Phicomm-K2P-Openwrt-Autobuild/runs/5750665029?check_suite_focus=true#step:11:62)1/linux-5.4.187/.modules] Error 2
make[3]: *** [Makefile:11: compile] Error 2
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt'
make[1]: *** [target/Makefile:18: /home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt'
make: *** [/home/runner/work/Phicomm-K2P-Openwrt-Autobuild/Phicomm-K2P-Openwrt-Autobuild/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

closed 2023-12-22 by aiamadeus