coolsnowwolf/lede #8941: R4s build with k5.15 encounter doornet2.dts missing issue

GreatMichaelLee, 2022-02-26:
: Warning (reg_format): /usb@ff600000/dwc3@ff600000/usb-eth@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:415.12-420.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #address-cells value
/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:415.12-420.4: Warning (avoid_default_addr_size): /usb@ff600000/dwc3@ff600000/usb-eth@2: Relying on default #size-cells value
rm -f /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3328-doornet1.dtb.tmp
aarch64-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/include -I/home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/include/ -undef -D__DTS__  -o /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb.tmp /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts
aarch64-openwrt-linux-musl-cpp: error: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts: No such file or directory
aarch64-openwrt-linux-musl-cpp: warning: '-x assembler-with-cpp' after last input file has no effect
aarch64-openwrt-linux-musl-cpp: fatal error: no input files
compilation terminated.
make[5]: *** [Makefile:78: /home/builder/lede_r4s/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/image-rk3399-doornet2.dtb] Error 1
make[5]: Leaving directory '/home/builder/lede_r4s/target/linux/rockchip/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/builder/lede_r4s/target/linux/rockchip'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/builder/lede_r4s/target/linux'
time: target/linux/install#8.33#6.42#19.71
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/builder/lede_r4s'
make[1]: *** [target/Makefile:19: /home/builder/lede_r4s/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/builder/lede_r4s'
make: *** [/home/builder/lede_r4s/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-02-26:
去掉不要编译，这个升级5.15 后很多patch 都没更新过来，导致很多都有错误，或者自己去掉编译这个

GreatMichaelLee, 2022-02-26:
不知道这个x86跟arm的是不是通用，我记得之前也遇到过，当时好像就是考一个过来就完事了。

WYC-2020, 2022-02-26:
https://github.com/WYC-2020/lede/commit/f8761b6b8e41f693148343b95cb894573895fd0c 
@GreatMichaelLee   自己测试功能，不保证功能哈

closed 2022-02-28 by AmadeusGhost