coolsnowwolf/lede #8944: package/kernel/rtl8821cu failed to build

EnnawYang, 2022-02-26:
target: x86_64  
build env: github actions ubuntu-20.04  
last commit: https://github.com/coolsnowwolf/lede/commit/71d7ad88c55a7a1cc0293a021c3bf6b598f68288  
logs:  
```

Applying ./patches/001-use-kernel-byteorder.patch using plaintext: 
patching file include/drv_types.h

Applying ./patches/010-remove-extra-cflags.patch using plaintext: 
patching file Makefile

Applying ./patches/020-remove-repeat-flies.patch using plaintext: 
patching file include/drv_conf.h
patching file include/linux/old_unused_rtl_wireless.h (renamed from include/linux/wireless.h)
patching file include/rtl_autoconf.h (renamed from include/autoconf.h)

Applying ./patches/030-change-value-of-vht-enable.patch using plaintext: 
patching file os_dep/linux/os_intfs.c

Applying ./patches/040-wireless-5.8.patch using plaintext: 
patching file os_dep/linux/ioctl_cfg80211.c
patching file os_dep/linux/os_intfs.c
make[4]: Entering directory '/home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25'
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_cmd.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_security.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_debug.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_io.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_ioctl_query.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_ioctl_set.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_ieee80211.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mlme.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mlme_ext.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_mi.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_wlan_util.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_vht.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_pwrctrl.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_rf.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_recv.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_sta_mgt.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_ap.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/mesh/rtw_mesh.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_xmit.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_p2p.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_rson.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_tdls.o
  CC [M]  /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_br_ext.o
/home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_br_ext.c:20:11: fatal error: net/ipx.h: No such file or directory
  #include <net/ipx.h>
           ^~~~~~~~~~~
compilation terminated.
make[5]: *** [scripts/Makefile.build:277: /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/core/rtw_br_ext.o] Error 1
make[4]: *** [Makefile:1868: /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820] Error 2
make[4]: Leaving directory '/home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25'
make[3]: *** [Makefile:89: /home/runner/work/xxx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-2020-12-19-428a0820/.built] Error 2
time: package/kernel/rtl8821cu/compile#27.87#3.88#32.24

```

WYC-2020, 2022-02-26:
看我仓库
@EnnawYang

EnnawYang, 2022-02-26:
> 看我仓库 @EnnawYang

好我试试

EnnawYang, 2022-02-26:
编译过了，但我手头没空闲的设备测试，先关了哈

closed 2022-02-26 by EnnawYang