coolsnowwolf/lede #8948: kmod-crypto-misc missing: glue_helper.ko

helmiau, 2022-02-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：kmod-crypto-misc error build due to glue_helper.ko is missing

### (2) 路由器型号和固件版本

A：x86, ASUS N46VZ, Lede source before this https://github.com/coolsnowwolf/lede/commit/3ea7ae24c60e1c6f29959961cebb28d6858a73df commit

### (3) 详细日志

A：

```
make[2]: Entering directory '/openwrt/package/kernel/linux'
ERROR: module '/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.15.25/arch/x86/crypto/glue_helper.ko' is missing.
make[2]: *** [/openwrt/bin/targets/x86/generic/packages/kmod-crypto-misc_5.15.25-1_i386_pentium4.ipk] Error 1
modules/crypto.mk:752: recipe for target '/openwrt/bin/targets/x86/generic/packages/kmod-crypto-misc_5.15.25-1_i386_pentium4.ipk' failed
make[2]: Leaving directory '/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.53#0.31#2.84
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[1]: Leaving directory '/openwrt'
make[1]: *** [package/kernel/linux/compile] Error 1
make: *** [package/compile] Error 2
```

WYC-2020, 2022-02-27:
@helmiau 
https://github.com/WYC-2020/lede/commit/205d384f392ee6307fc73e083c67064ef6eaac65

WYC-2020, 2022-02-27:
@helmiau I'll look at it later

WYC-2020, 2022-02-27:
@helmiau 
https://github.com/WYC-2020/lede/commit/71e0b9d1334a8dc735231da4ff81e60c3006410d

helmiau, 2022-02-27:
@WYC-2020 i will test it later bro, thanks for your help

helmiau, 2022-02-27:
@WYC-2020 here is current error logs after applying your patch

```

2022-02-27T10:42:45.7121958Z #
2022-02-27T10:42:45.7122227Z # configuration written to .config
2022-02-27T10:42:45.7122514Z #
2022-02-27T10:42:45.7128899Z make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-02-27T10:42:45.7130191Z rm -rf /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/modules
2022-02-27T10:42:45.7177786Z make  -C "/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1" KCFLAGS="-fmacro-prefix-map=/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/host/include -I/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/hostpkg/include -I/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="helmiau" KBUILD_BUILD_HOST="helmiau.com" KBUILD_BUILD_TIMESTAMP="Sun Feb 27 05:31:25 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/netatop.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/trelay.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/nat46.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/gpio-nct5104d.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/fast-classifier.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/rtpengine.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/gpio-button-hotplug.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/dmx_usb_module.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/dahdi-linux.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/ksmbd.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/ovpn-dco.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/siit.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/gl-mifi-mcu.symvers /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/jool.symvers" KERNELRELEASE=5.15.25 EXTRA_CFLAGS="-I/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include -fmacro-prefix-map=/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1=backports-5.10.42-1  -I/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/staging_dir/target-x86_64_musl/usr/include/qca-nss-drv" KLIB_BUILD="/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25" MODPROBE=true KLIB=/lib/modules/5.15.25 KERNEL_SUBLEVEL=15 KBUILD_LDFLAGS_MODULE_PREREQ= modules
2022-02-27T10:42:45.7198204Z make[3]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-02-27T10:42:45.7319787Z make[4]: 'Kconfig.versions' is up to date.
2022-02-27T10:42:45.7487924Z make[6]: 'Kconfig.versions' is up to date.
2022-02-27T10:42:45.7595192Z make[7]: 'conf' is up to date.
2022-02-27T10:42:45.7739387Z boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
2022-02-27T10:42:45.7739995Z #
2022-02-27T10:42:45.7740243Z # configuration written to .config
2022-02-27T10:42:45.7740470Z #
2022-02-27T10:42:45.7908045Z Building backport-include/backport/autoconf.h ... done.
2022-02-27T10:42:46.5786598Z   CC [M]  /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/intel/iwlwifi/mvm/tt.o
[... cut ...]
2022-02-27T10:42:47.7663036Z package/Makefile:114: recipe for target 'package/kernel/mac80211/compile' failed
2022-02-27T10:42:47.7695019Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-02-27T10:42:47.7713249Z make: *** [package/compile] Error 2
2022-02-27T10:42:47.7713807Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2022-02-27T10:42:47.7789366Z ##[error]Process completed with exit code 2.
```

WYC-2020, 2022-02-27:
@helmiau https://github.com/WYC-2020/lede/commit/0bdae446b37ca151de2c17902635df59770c5c25

WYC-2020, 2022-02-27:
I fixed this problem before

helmiau, 2022-02-27:
okay, i hope this will fix everything.. until you can pull your commits to lede main branch again. Thanks for your support bro!.

helmiau, 2022-02-27:
bro @WYC-2020 , now this error occurs on my build, any advice?

```
2022-02-27T17:20:14.4390374Z make[3]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-02-27T17:20:14.4536876Z make[4]: 'Kconfig.versions' is up to date.
2022-02-27T17:20:14.4743117Z make[6]: 'Kconfig.versions' is up to date.
2022-02-27T17:20:14.4873272Z make[7]: 'conf' is up to date.
2022-02-27T17:20:14.5046744Z boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
2022-02-27T17:20:14.5047079Z #
2022-02-27T17:20:14.5047333Z # configuration written to .config
2022-02-27T17:20:14.5047588Z #
2022-02-27T17:20:14.5256010Z Building backport-include/backport/autoconf.h ... done.
2022-02-27T17:20:15.1514104Z   CC [M]  /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o
2022-02-27T17:20:16.2054930Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c: In function 'rt2x00lib_set_mac_address':
2022-02-27T17:20:16.2056618Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:1000:13: error: too few arguments to function 'of_get_mac_address'
2022-02-27T17:20:16.2057702Z   mac_addr = of_get_mac_address(rt2x00dev->dev->of_node);
2022-02-27T17:20:16.2058233Z              ^~~~~~~~~~~~~~~~~~
2022-02-27T17:20:16.2061181Z In file included from /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/of_net.h:3,
2022-02-27T17:20:16.2062347Z                  from /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:19:
2022-02-27T17:20:16.2112896Z ./include/linux/of_net.h:25:19: note: declared here
2022-02-27T17:20:16.2113285Z  static inline int of_get_mac_address(struct device_node *np, u8 *mac)
2022-02-27T17:20:16.2113608Z                    ^~~~~~~~~~~~~~~~~~
2022-02-27T17:20:16.2400613Z make[10]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
2022-02-27T17:20:16.2401670Z scripts/Makefile.build:277: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o' failed
2022-02-27T17:20:16.2403146Z make[9]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00] Error 2
2022-02-27T17:20:16.2406086Z make[8]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink] Error 2
2022-02-27T17:20:16.2407136Z make[7]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
2022-02-27T17:20:16.2408068Z make[6]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
2022-02-27T17:20:16.2412451Z make[5]: *** [modules] Error 2
2022-02-27T17:20:16.2413348Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00' failed
2022-02-27T17:20:16.2414034Z make[4]: *** [modules] Error 2
2022-02-27T17:20:16.2414611Z make[3]: *** [modules] Error 2
2022-02-27T17:20:16.2416878Z make[2]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
2022-02-27T17:20:16.2425500Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink' failed
2022-02-27T17:20:16.2426473Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless' failed
2022-02-27T17:20:16.2427355Z Makefile:1868: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1' failed
2022-02-27T17:20:16.2427921Z Makefile.build:13: recipe for target 'modules' failed
2022-02-27T17:20:16.2428367Z Makefile.real:93: recipe for target 'modules' failed
2022-02-27T17:20:16.2430975Z Makefile:120: recipe for target 'modules' failed
2022-02-27T17:20:16.2431876Z make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-02-27T17:20:16.2432724Z Makefile:578: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built' failed
2022-02-27T17:20:16.2433452Z make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/mac80211'
2022-02-27T17:20:16.2437424Z time: package/kernel/mac80211/regular/compile#1.47#0.59#2.42
[... cut ...]
2022-02-27T17:20:16.2441000Z package/Makefile:114: recipe for target 'package/kernel/mac80211/compile' failed
2022-02-27T17:20:16.2470737Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-02-27T17:20:16.2494507Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2022-02-27T17:20:16.2499138Z make: *** [package/compile] Error 2
2022-02-27T17:20:16.2561395Z ##[error]Process completed with exit code 2.
```

WYC-2020, 2022-02-27:
You have compiled too many third-party drivers, they are not compatible with 5.15, I will take a look later

helmiau, 2022-02-28:
@WYC-2020 i use 5.4 kernel only, but 5.10 and 5.15 kernel patches is following build, can these patches excluded from build?

WYC-2020, 2022-02-28:
https://github.com/WYC-2020/lede/commit/e2c22d3587249b51a3358f4d1b61fa3a587d37f6
@helmiau

helmiau, 2022-02-28:
how about these logs @WYC-2020 ?

```
2022-02-28T19:32:37.2778980Z #
2022-02-28T19:32:37.2779276Z # configuration written to .config
2022-02-28T19:32:37.2779588Z #
2022-02-28T19:32:37.2979063Z Building backport-include/backport/autoconf.h ... done.
2022-02-28T19:32:37.8935572Z   CC [M]  /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek/adm8211.o
2022-02-28T19:32:38.9356413Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek/adm8211.c:31:25: error: expected declaration specifiers or '...' before string constant
2022-02-28T19:32:38.9357659Z  MODULE_SUPPORTED_DEVICE("ADM8211");
2022-02-28T19:32:38.9358185Z                          ^~~~~~~~~
2022-02-28T19:32:38.9866698Z make[9]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek/adm8211.o] Error 1
2022-02-28T19:32:38.9868523Z scripts/Makefile.build:277: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek/adm8211.o' failed
2022-02-28T19:32:38.9870645Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek' failed
2022-02-28T19:32:38.9877699Z make[8]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/admtek] Error 2
2022-02-28T19:32:38.9879346Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless' failed
2022-02-28T19:32:38.9881091Z Makefile:1868: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1' failed
2022-02-28T19:32:38.9886587Z make[7]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
2022-02-28T19:32:38.9893296Z Makefile.build:13: recipe for target 'modules' failed
2022-02-28T19:32:38.9894965Z make[6]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
2022-02-28T19:32:38.9895687Z make[5]: *** [modules] Error 2
2022-02-28T19:32:38.9896147Z make[4]: *** [modules] Error 2
2022-02-28T19:32:38.9896598Z make[3]: *** [modules] Error 2
2022-02-28T19:32:38.9897672Z make[2]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
2022-02-28T19:32:38.9898649Z Makefile.real:93: recipe for target 'modules' failed
2022-02-28T19:32:38.9902266Z Makefile:120: recipe for target 'modules' failed
2022-02-28T19:32:38.9903432Z make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-02-28T19:32:38.9904933Z Makefile:578: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built' failed
2022-02-28T19:32:38.9907030Z make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/kernel/mac80211'
2022-02-28T19:32:38.9907817Z time: package/kernel/mac80211/regular/compile#1.53#0.50#2.37
2022-02-28T19:32:38.9909049Z     ERROR: package/kernel/mac80211 failed to build (build variant: regular).
```

WYC-2020, 2022-02-28:
I fixed this, and the lede was merged, the result was canceled by this 
https://github.com/coolsnowwolf/lede/commit/da34b3a2fb2da075f4e3c7058ab98adbb6eed514, just join

https://github.com/WYC-2020/lede/commit/9ea4c2e7634d73645bd3274f2d5fd8437580ea77
@helmiau

helmiau, 2022-03-01:
and now, how about this @WYC-2020 ?

```
2022-03-01T08:22:23.7111980Z make[3]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
2022-03-01T08:22:23.7237562Z make[4]: 'Kconfig.versions' is up to date.
2022-03-01T08:22:23.7404078Z make[6]: 'Kconfig.versions' is up to date.
2022-03-01T08:22:23.7508912Z make[7]: 'conf' is up to date.
2022-03-01T08:22:23.7641537Z boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
2022-03-01T08:22:23.7642133Z #
2022-03-01T08:22:23.7642564Z # configuration written to .config
2022-03-01T08:22:23.7642964Z #
2022-03-01T08:22:23.7809866Z Building backport-include/backport/autoconf.h ... done.
2022-03-01T08:22:24.3134774Z   CC [M]  /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.o
2022-03-01T08:22:25.3071027Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c: In function 'wlcore_probe':
2022-03-01T08:22:25.3072787Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:6787:46: error: 'FW_ACTION_HOTPLUG' undeclared (first use in this function); did you mean 'FLOW_ACTION_SAMPLE'?
2022-03-01T08:22:25.3074029Z    ret = request_firmware_nowait(THIS_MODULE, FW_ACTION_HOTPLUG,
2022-03-01T08:22:25.3075112Z                                               ^~~~~~~~~~~~~~~~~
2022-03-01T08:22:25.3075575Z                                               FLOW_ACTION_SAMPLE
2022-03-01T08:22:25.3076645Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:6787:46: note: each undeclared identifier is reported only once for each function it appears in
2022-03-01T08:22:25.3393528Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c: In function 'wl12xx_init_vif_data':
2022-03-01T08:22:25.3395359Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:2229:14: warning: this statement may fall through [-Wimplicit-fallthrough=]
2022-03-01T08:22:25.3486841Z    wlvif->p2p = 1;
2022-03-01T08:22:25.3487069Z    ~~~~~~~~~~~^~~
2022-03-01T08:22:25.3487655Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:2231:2: note: here
2022-03-01T08:22:25.3488053Z   case NL80211_IFTYPE_STATION:
2022-03-01T08:22:25.3488247Z   ^~~~
2022-03-01T08:22:25.3488923Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:2239:14: warning: this statement may fall through [-Wimplicit-fallthrough=]
2022-03-01T08:22:25.3489437Z    wlvif->p2p = 1;
2022-03-01T08:22:25.3489624Z    ~~~~~~~~~~~^~~
2022-03-01T08:22:25.3490186Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.c:2241:2: note: here
2022-03-01T08:22:25.3490571Z   case NL80211_IFTYPE_AP:
2022-03-01T08:22:25.3490778Z   ^~~~
2022-03-01T08:22:25.3639194Z make[10]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.o] Error 1
2022-03-01T08:22:25.3639998Z make[9]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore] Error 2
2022-03-01T08:22:25.3640822Z scripts/Makefile.build:277: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore/main.o' failed
2022-03-01T08:22:25.3644851Z make[8]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti] Error 2
2022-03-01T08:22:25.3646131Z make[7]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
2022-03-01T08:22:25.3646788Z make[6]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
2022-03-01T08:22:25.3647541Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti/wlcore' failed
2022-03-01T08:22:25.3648344Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ti' failed
2022-03-01T08:22:25.3649353Z scripts/Makefile.build:540: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless' failed
2022-03-01T08:22:25.3650077Z Makefile:1868: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1' failed
[... cut ...]
2022-03-01T08:22:25.3669324Z package/Makefile:114: recipe for target 'package/kernel/mac80211/compile' failed
2022-03-01T08:22:25.3697253Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-03-01T08:22:25.3710729Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2022-03-01T08:22:25.3714737Z make: *** [package/compile] Error 2
2022-03-01T08:22:25.3793124Z ##[error]Process completed with exit code 2.
```

WYC-2020, 2022-03-01:
Well, you should send the config directly.

helmiau, 2022-03-02:
here's  my .config file contents
```
# Target
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_generic=y
CONFIG_TARGET_x86_generic_DEVICE_generic=y

# Output Images
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_TARGET_ROOTFS_EXT4FS=n
CONFIG_TARGET_ROOTFS_SQUASHFS=y

# Output x86 Images
CONFIG_GRUB_TITLE="HelmiWrt eXtendZed by helmiau"
CONFIG_GRUB_IMAGES=y
CONFIG_GRUB_EFI_IMAGES=y
CONFIG_VDI_IMAGES=n
CONFIG_VMDK_IMAGES=n

# Extra Packages
CONFIG_PACKAGE_autocore-x86=y

# Network Devices
CONFIG_PACKAGE_kmod-3c59x=y
CONFIG_PACKAGE_kmod-8139cp=y
CONFIG_PACKAGE_kmod-8139too=y
CONFIG_PACKAGE_kmod-alx=y
CONFIG_PACKAGE_kmod-atl1=y
CONFIG_PACKAGE_kmod-atl1c=y
CONFIG_PACKAGE_kmod-atl1e=y
CONFIG_PACKAGE_kmod-atl2=y
CONFIG_PACKAGE_kmod-b44=y
CONFIG_PACKAGE_kmod-be2net=y
CONFIG_PACKAGE_kmod-bnx2=y
CONFIG_PACKAGE_kmod-bnx2x=y
CONFIG_PACKAGE_kmod-dm9000=y
CONFIG_PACKAGE_kmod-dummy=y
CONFIG_PACKAGE_kmod-e100=y
CONFIG_PACKAGE_kmod-e1000=y
CONFIG_PACKAGE_kmod-et131x=y
CONFIG_PACKAGE_kmod-ethoc=y
CONFIG_PACKAGE_kmod-forcedeth=y
CONFIG_PACKAGE_kmod-gigaset=y
CONFIG_PACKAGE_kmod-hfcmulti=y
CONFIG_PACKAGE_kmod-hfcpci=y
CONFIG_PACKAGE_kmod-i40e=y
CONFIG_PACKAGE_kmod-i40evf=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-igb=y
CONFIG_PACKAGE_kmod-igbvf=y
CONFIG_PACKAGE_kmod-ixgbe=y
CONFIG_PACKAGE_kmod-ixgbevf=y
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-macvlan=y
CONFIG_PACKAGE_kmod-mii=y
CONFIG_PACKAGE_kmod-natsemi=y
CONFIG_PACKAGE_kmod-ne2k-pci=y
CONFIG_PACKAGE_kmod-niu=y
CONFIG_PACKAGE_kmod-of-mdio=y
CONFIG_PACKAGE_kmod-pcnet32=y
CONFIG_PACKAGE_kmod-phy-broadcom=y
CONFIG_PACKAGE_kmod-phy-realtek=y
CONFIG_PACKAGE_kmod-phylib-broadcom=y
CONFIG_PACKAGE_kmod-r6040=y
CONFIG_PACKAGE_kmod-r8125=y
CONFIG_PACKAGE_kmod-r8169=y
CONFIG_PACKAGE_kmod-siit=y
CONFIG_PACKAGE_kmod-sis190=y
CONFIG_PACKAGE_kmod-sis900=y
CONFIG_PACKAGE_kmod-skge=y
[... 771 lines cut ...]
CONFIG_I2C_MUX_GPMUX=y
CONFIG_I2C_MUX_LTC4306=m
CONFIG_I2C_MUX_PCA9541=m
CONFIG_I2C_MUX_PCA954x=m
CONFIG_I2C_MUX_PINCTRL=y
CONFIG_I2C_MUX_REG=y
CONFIG_I2C_DEMUX_PINCTRL=y
CONFIG_I2C_MUX_MLXCPLD=m
CONFIG_I2C_HELPER_AUTO=y
CONFIG_I2C_ALGOBIT=y
CONFIG_I2C_DESIGNWARE_CORE=y
CONFIG_I2C_DESIGNWARE_SLAVE=y
CONFIG_I2C_DESIGNWARE_PLATFORM=y
CONFIG_I2C_DESIGNWARE_PCI=m
CONFIG_I2C_GPIO=y
CONFIG_I2C_GPIO_FAULT_INJECTOR=y
CONFIG_I2C_MESON=y
CONFIG_I2C_RK3X=y
CONFIG_I2C_SLAVE=y
CONFIG_I2C_SLAVE_EEPROM=m
CONFIG_I2C_SLAVE_TESTUNIT=m
CONFIG_SPI=y
CONFIG_SPI_MASTER=y
CONFIG_SPI_MEM=y
```

helmiau, 2022-03-02:
i tried to rebuild again, then these errors shown @WYC-2020 :
```
2022-03-02T08:28:07.3204437Z make[3]: Entering directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.15.25'
2022-03-02T08:28:07.9071901Z   CC [M]  /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/core/rtw_br_ext.o
2022-03-02T08:28:08.6266940Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/core/rtw_br_ext.c:20:11: fatal error: net/ipx.h: No such file or directory
2022-03-02T08:28:08.6269644Z   #include <net/ipx.h>
2022-03-02T08:28:08.6271687Z            ^~~~~~~~~~~
2022-03-02T08:28:08.6273739Z compilation terminated.
2022-03-02T08:28:08.6311376Z scripts/Makefile.build:277: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/core/rtw_br_ext.o' failed
2022-03-02T08:28:08.6314252Z make[4]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/core/rtw_br_ext.o] Error 1
2022-03-02T08:28:08.6316826Z make[3]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3] Error 2
2022-03-02T08:28:08.6318691Z make[2]: *** [/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/.built] Error 2
2022-03-02T08:28:08.6321594Z Makefile:1868: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3' failed
2022-03-02T08:28:08.6322548Z make[3]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/linux-5.15.25'
2022-03-02T08:28:08.6325506Z Makefile:74: recipe for target '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/build_dir/target-i386_pentium4_musl/linux-x86_generic/rtl8188eu-2021-02-06-1e7145f3/.built' failed
2022-03-02T08:28:08.6326297Z make[2]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/package/community/rtl8188eu'
2022-03-02T08:28:08.6326760Z time: package/community/rtl8188eu/compile#1.02#0.37#1.58
2022-03-02T08:28:08.6327188Z     ERROR: package/community/rtl8188eu failed to build.
2022-03-02T08:28:08.6332526Z package/Makefile:114: recipe for target 'package/community/rtl8188eu/compile' failed
2022-03-02T08:28:08.6363678Z make[1]: Leaving directory '/home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt'
2022-03-02T08:28:08.6370688Z make[1]: *** [package/community/rtl8188eu/compile] Error 1
2022-03-02T08:28:08.6379013Z /home/runner/work/HelmiWrt-OS/HelmiWrt-OS/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2022-03-02T08:28:08.6383370Z make: *** [package/compile] Error 2
```

WYC-2020, 2022-03-03:
@helmiau I use the configuration file you gave to compile x86-64 here. I removed all luci apps to reduce the compilation time, and the compilation is no problem. I don’t have the rtl8188eu driver here.

helmiau, 2022-03-03:
@WYC-2020 i forgot, i use this source https://github.com/immortalwrt/immortalwrt/tree/openwrt-18.06-k5.4/package/kernel/rtl8188eu as rtl8188eu driver

can you recompile it? thanks bro.

WYC-2020, 2022-03-04:
@helmiau  Do not compile any driver about mac80211 that comes with lede, only compile your new ones, they conflict

WYC-2020, 2022-03-04:
@helmiau 
Your compilation option is to compile all packages, there may be many conflicts, it is recommended to compile according to your actual needs

helmiau, 2022-03-05:
thanks bro @WYC-2020 , i will make some changes later

closed 2022-03-05 by helmiau

thomaswcy, 2022-03-08:
I've just got a similar problem while compiling the firmware for NanoPi R2S. you can see the log below:
`make[2]: Entering directory '/home/runner/work/R2S/R2S/openwrt/package/kernel/linux'
Package kmod-drm is missing dependencies for the following libraries:
fb.ko
make[2]: Leaving directory '/home/runner/work/R2S/R2S/openwrt/package/kernel/linux'
make[2]: *** [modules/video.mk:263: /home/runner/work/R2S/R2S/openwrt/bin/targets/rockchip/armv8/packages/kmod-drm_5.15.26-1_aarch64_generic.ipk] Error 1
time: package/kernel/linux/compile#3.80#0.46#4.24
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/R2S/R2S/openwrt'
make: *** [/home/runner/work/R2S/R2S/openwrt/include/toplevel.mk:230: package/compile] Error 2`
Do you know what is the problem? Are you able to fix it? Thank you!