coolsnowwolf/lede #8949: 云编译出错希望能帮助分析一下

Licolnlee, 2022-02-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译出错希望能帮助分析一下

### (2) 路由器型号和固件版本

A：x86/64

### (3) 详细日志

A：

cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:277: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock/xt_pknock.o] Error 1
make[8]: *** [scripts/Makefile.build:540: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions/pknock] Error 2
make[7]: *** [Makefile:1868: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions] Error 2
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25'
make[6]: *** [Makefile:461: modules] Error 2
make[5]: *** [Makefile:495: all-recursive] Error 1
make[4]: *** [Makefile:380: all] Error 2
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/extensions'
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13'
make[3]: *** [Makefile:177: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#1.19#0.22#1.39
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-27:
什么鬼包依赖这个库吗，我自己搜索我所有的库，就没看到那个依赖这个

pangfeng161, 2022-02-27:
这几天怎么都出现编译出错

Licolnlee, 2022-02-27:
> 什么鬼包依赖这个库吗，我自己搜索我所有的库，就没看到那个依赖这个

好像是我选择的python3和python3-pip里面的某个依赖出错了，取消了之后现在编译正常了，但是也无法通过opkg安装，也显示依赖有问题。

breakings, 2022-02-27:
iptables 3.19修好了

WYC-2020, 2022-02-28:
https://github.com/WYC-2020/packages/commit/fed3c86b5fbf0a851885fdcbff14428aab685f91
@Licolnlee

Licolnlee, 2022-02-28:
> [WYC-2020/packages@fed3c86](https://github.com/WYC-2020/packages/commit/fed3c86b5fbf0a851885fdcbff14428aab685f91) @Licolnlee

收到，感谢！辛苦了！

closed 2022-02-28 by Licolnlee