coolsnowwolf/lede #8951: 刚源码库更新后X86 编译失败

baodp, 2022-02-26:
WARNING: Makefile 'package/feeds/packages/bmx7-dnsupdate/Makefile' has a dependency on 'bmx7', which does not exist
WARNING: Makefile 'package/feeds/packages/bmx7-dnsupdate/Makefile' has a dependency on 'bmx7-json', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ahcp/Makefile' has a dependency on 'ahcpd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-easymesh/Makefile' has a dependency on 'batctl-default', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-easymesh/Makefile' has a dependency on 'kmod-batman-adv', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd-mod-jsoninfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd-mod-txtinfo', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6-json', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7-json', which does not exist
make[1]: Entering directory '/home/bdp/lede'
make[2]: Entering directory '/home/bdp/lede'
make[2]: Entering directory '/home/bdp/lede'
rm -rf /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86
+ mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl
+ mkdir -p /home/bdp/lede/staging_dir/host
+ cd /home/bdp/lede/staging_dir/target-x86_64_musl
+ mkdir -p bin lib stamp usr/include usr/lib
+ cd /home/bdp/lede/staging_dir/host
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /home/bdp/lede/build_dir/target-x86_64_musl/stamp
mkdir -p /home/bdp/lede/build_dir/host/stamp /home/bdp/lede/staging_dir/host/include/sys
touch /home/bdp/lede/staging_dir/target-x86_64_musl/.prepared
install -m0644 /home/bdp/lede/tools/include/*.h /home/bdp/lede/staging_dir/host/include/
install -m0644 /home/bdp/lede/tools/include/sys/*.h /home/bdp/lede/staging_dir/host/include/sys/
ln -snf lib /home/bdp/lede/staging_dir/host/lib64
touch /home/bdp/lede/staging_dir/host/.prepared
make[3]: Entering directory '/home/bdp/lede/tools/flock'
make[3]: Leaving directory '/home/bdp/lede/tools/flock'
time: tools/flock/compile#0.01#0.00#0.02
make[3]: Entering directory '/home/bdp/lede/tools/xz'
make[2]: Leaving directory '/home/bdp/lede'
make[3]: Leaving directory '/home/bdp/lede/tools/xz'
time: tools/xz/compile#0.02#0.00#0.04
make[3]: Entering directory '/home/bdp/lede/tools/sed'
make[3]: Leaving directory '/home/bdp/lede/tools/sed'
time: tools/sed/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/bdp/lede/tools/tar'
make[3]: Entering directory '/home/bdp/lede/tools/patch'
make[3]: Leaving directory '/home/bdp/lede/tools/patch'
time: tools/patch/compile#0.02#0.00#0.03
make[3]: Leaving directory '/home/bdp/lede/tools/tar'
time: tools/tar/compile#0.02#0.00#0.03
make[3]: Entering directory '/home/bdp/lede/tools/pkgconf'
make[3]: Entering directory '/home/bdp/lede/tools/m4'
make[3]: Entering directory '/home/bdp/lede/tools/ninja'
make[3]: Entering directory '/home/bdp/lede/tools/autoconf-archive'
make[3]: Entering directory '/home/bdp/lede/tools/cpio'
make[3]: Entering directory '/home/bdp/lede/tools/mtools'
make[3]: Entering directory '/home/bdp/lede/tools/lzma'
make[3]: Entering directory '/home/bdp/lede/tools/patch-image'
make[3]: Entering directory '/home/bdp/lede/tools/sstrip'
make[3]: Entering directory '/home/bdp/lede/tools/expat'
make[3]: Entering directory '/home/bdp/lede/tools/xxd'
make[3]: Entering directory '/home/bdp/lede/tools/ucl'
make[3]: Entering directory '/home/bdp/lede/tools/zip'
make[3]: Leaving directory '/home/bdp/lede/tools/patch-image'
time: tools/patch-image/compile#0.05#0.00#0.06
make[3]: Leaving directory '/home/bdp/lede/tools/ninja'
time: tools/ninja/compile#0.06#0.00#0.07
make[3]: Leaving directory '/home/bdp/lede/tools/pkgconf'
time: tools/pkgconf/compile#0.05#0.01#0.07
make[3]: Leaving directory '/home/bdp/lede/tools/xxd'
time: tools/xxd/compile#0.05#0.00#0.06
make[3]: Leaving directory '/home/bdp/lede/tools/mtools'
[... 119 lines cut ...]
patching file bootstrap
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 1423 with fuzz 1.
touch /home/bdp/lede/build_dir/host/cmake-3.22.1/.prepared8c8bb8f34847dc64d9c68969110d069f_6664517399ebbbc92a37c5bb081b5c53
(cd /home/bdp/lede/build_dir/host/cmake-3.22.1/; if [ -x configure ]; then cp -fpR /home/bdp/lede/scripts/config.{guess,sub} /home/bdp/lede/build_dir/host/cmake-3.22.1// && CC="gcc" CFLAGS="-O2 -I/home/bdp/lede/staging_dir/host/include " CXX="g++" CPPFLAGS="-I/home/bdp/lede/staging_dir/host/include " LDFLAGS="-L/home/bdp/lede/staging_dir/host/lib " CONFIG_SHELL="/usr/bin/env bash" CC="gcc" CXX="g++" MAKEFLAGS="--jobserver-auth=3,4 " CXXFLAGS="-O2 -I/home/bdp/lede/staging_dir/host/include " MAKE="/home/bdp/lede/staging_dir/host/bin/ninja"  bash ./configure --parallel="--jobserver-auth=3,4" --prefix=/home/bdp/lede/staging_dir/host ; fi )
---------------------------------------------
CMake 3.22.1, Copyright 2000-2021 Kitware, Inc. and Contributors
C compiler on this system is: gcc -O2 -I/home/bdp/lede/staging_dir/host/include
C++ compiler on this system is: g++ -O2 -I/home/bdp/lede/staging_dir/host/include
---------------------------------------------
Error when bootstrapping CMake:
Cannot find appropriate Makefile processor on this system.
Please specify one using environment variable MAKE.
---------------------------------------------
Log of errors: /home/bdp/lede/build_dir/host/cmake-3.22.1/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:40: /home/bdp/lede/build_dir/host/cmake-3.22.1/.configured] Error 8
make[3]: Leaving directory '/home/bdp/lede/tools/cmake'
time: tools/cmake/compile#0.71#0.72#1.73
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:160: tools/cmake/compile] Error 1
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [tools/Makefile:156: /home/bdp/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230: world] Error 2

Beginner-Go, 2022-02-26:
别急 我看看

baodp, 2022-02-26:
> 别急 我看看

不急，辛苦了。

breakings, 2022-02-26:
cmake翻车了

Beginner-Go, 2022-02-26:
现在试一试，我在本地测试过了 应该正常了

baodp, 2022-02-26:
> 现在试一试，我在本地测试过了 应该正常了

谢谢，编译通过了。

closed 2022-02-26 by Beginner-Go