coolsnowwolf/lede #8957: 编译出错

luohuawei168, 2022-02-27:
麻烦帮看看这是什么问题，没添加其它package，直接下载直接编译，总是提示空间不足，VM虚拟机环境
cat: write error: No space left on device
sed: couldn't write 43 items to stdout: No space left on device
configure: creating ./config.status
cat: write error: No space left on device
cat: write error: No space left on device
./configure: 26293: echo: echo: I/O error
./configure: 26297: echo: echo: I/O error
./configure: 26305: echo: echo: I/O error
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
uniq: write error: No space left on device
./configure: 27215: echo: echo: I/O error
./configure: 27230: echo: echo: I/O error
cat: write error: No space left on device
cat: write error: No space left on device
cat: write error: No space left on device
make[4]: Entering directory '/home/l/lede/build_dir/target-x86_64_musl/ncurses-6.2'
make[4]: *** No rule to make target 'preinstall'.  Stop.
make[4]: Leaving directory '/home/l/lede/build_dir/target-x86_64_musl/ncurses-6.2'
sed: couldn't write 24 items to stdout: No space left on device
sed: couldn't flush stdout: No space left on device
./configure: 32: echo: echo: I/O error
./configure: 32: echo: echo: I/O error
./configure: 35: echo: echo: I/O error
make[3]: *** [Makefile:190: /home/l/lede/build_dir/target-x86_64_musl/ncurses-6.2/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 2
make[3]: Leaving directory '/home/l/lede/package/libs/ncurses'
time: package/libs/ncurses/compile#2.57#1.75#7.60
    ERROR: package/libs/ncurses failed to build.
make[2]: *** [package/Makefile:116: package/libs/ncurses/compile] Error 1
make[2]: Leaving directory '/home/l/lede'
make[1]: *** [package/Makefile:110: /home/l/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/l/lede'
make: *** [/home/l/lede/include/toplevel.mk:230: world] Error 2

zopargr, 2022-02-27:
硬盘空间不够了呀

yjs507, 2022-02-27:
最少给40g，多多益善

luohuawei168, 2022-02-27:
50G 还是不够，改了120GB了。谢谢各位回复

closed 2022-02-27 by luohuawei168