coolsnowwolf/lede #8961: 5.15内核在线编译amule报错

Cptmacmillan2022007, 2022-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

2022-02-28T05:29:10.4478386Z config.status: creating src/utils/plasmamule/Makefile
2022-02-28T05:29:10.4733081Z config.status: creating src/utils/wxCas/Makefile
2022-02-28T05:29:10.4940438Z config.status: creating src/utils/wxCas/docs/Makefile
2022-02-28T05:29:10.5125912Z config.status: creating src/utils/wxCas/src/Makefile
2022-02-28T05:29:10.5318070Z config.status: creating src/utils/xas/Makefile
2022-02-28T05:29:10.5522604Z config.status: creating src/utils/xas/autostart-xas
2022-02-28T05:29:10.5706654Z config.status: creating src/utils/xas/docs/Makefile
2022-02-28T05:29:10.5941828Z config.status: creating src/skins/Makefile
2022-02-28T05:29:10.6122528Z config.status: creating src/webserver/Makefile
2022-02-28T05:29:10.6310183Z config.status: creating src/webserver/src/Makefile
2022-02-28T05:29:10.6508247Z config.status: creating src/webserver/default/Makefile
2022-02-28T05:29:10.6708866Z config.status: creating unittests/Makefile
2022-02-28T05:29:10.6892396Z config.status: creating unittests/muleunit/Makefile
2022-02-28T05:29:10.7086349Z config.status: creating unittests/tests/Makefile
2022-02-28T05:29:10.7285452Z config.status: creating config.h
2022-02-28T05:29:10.7412603Z config.status: executing depfiles commands
2022-02-28T05:29:11.7961981Z config.status: executing po-directories commands
2022-02-28T05:29:11.8058399Z config.status: creating po/POTFILES
2022-02-28T05:29:11.8160414Z config.status: creating po/Makefile
2022-02-28T05:29:11.8218434Z config.status: executing denoiser commands
2022-02-28T05:29:12.2032840Z configure: WARNING: unrecognized options: --without-x
2022-02-28T05:29:12.2033077Z 
2022-02-28T05:29:12.2033568Z 
2022-02-28T05:29:12.2034294Z   Configure script has finished system check.
2022-02-28T05:29:12.2034625Z 
2022-02-28T05:29:12.2035049Z   Configured aMule SVN for 'x86_64-openwrt-linux-gnu'.
2022-02-28T05:29:12.2035557Z 
2022-02-28T05:29:12.2036764Z   aMule enabled options:
2022-02-28T05:29:12.2037040Z 
2022-02-28T05:29:12.2037211Z   **** aMule Core ****
2022-02-28T05:29:12.2037590Z   Prefix where aMule should be installed?                    /usr
2022-02-28T05:29:12.2038052Z   Should aMule be compiled with i18n support?                no
2022-02-28T05:29:12.2038594Z   Should aMule be compiled in debug mode?                    no
2022-02-28T05:29:12.2039022Z   Should aMule be compiled with profiling?                   no
2022-02-28T05:29:12.2039447Z   Should aMule be compiled with optimizations?               no
2022-02-28T05:29:12.2039874Z   Should aMule be compiled with UPnP support?                yes
2022-02-28T05:29:12.2040325Z   Should aMule be compiled with IP2country support?          no
2022-02-28T05:29:12.2040754Z   Should aMule monolithic application be built?              no
2022-02-28T05:29:12.2041300Z   Should aMule daemon version be built?                      yes
2022-02-28T05:29:12.2041710Z   Should aMule remote gui be built?                          no
2022-02-28T05:29:12.2041973Z 
2022-02-28T05:29:12.2042159Z   **** aMule TextClient ****
2022-02-28T05:29:12.2042548Z   Should aMule Command Line Client be built?                 yes
2022-02-28T05:29:12.2042819Z 
2022-02-28T05:29:12.2043000Z   **** aMule WebServer ****
2022-02-28T05:29:12.2043376Z   Should aMule WebServer be built?                           yes
2022-02-28T05:29:12.2043633Z 
2022-02-28T05:29:12.2044115Z   **** aMule ED2K Links Handler ****
2022-02-28T05:29:12.2044469Z   Should aMule ED2K Links Handler be built?                  no
[... 153 lines cut ...]
2022-02-28T05:30:09.2301384Z Linking amulecmd
2022-02-28T05:30:09.3169380Z Compiling amuled.cpp
2022-02-28T05:30:10.9400505Z Compiling amule.cpp
2022-02-28T05:30:14.8867712Z Compiling BaseClient.cpp
2022-02-28T05:30:18.9517702Z Compiling ClientList.cpp
2022-02-28T05:30:21.9749518Z Compiling ClientCreditsList.cpp
2022-02-28T05:30:23.1926399Z ClientCreditsList.cpp: In member function 'void CClientCreditsList::InitalizeCrypting()':
2022-02-28T05:30:23.1928288Z ClientCreditsList.cpp:315:10: error: 'CryptoPP::RSASSA_PKCS1v15_SHA_Verifier' {aka 'class CryptoPP::PK_FinalTemplate<CryptoPP::TF_VerifierImpl<CryptoPP::TF_SignatureSchemeOptions<CryptoPP::TF_SS<CryptoPP::RSA, CryptoPP::PKCS1v15, CryptoPP::SHA1, int>, CryptoPP::RSA, CryptoPP::PKCS1v15_SignatureMessageEncodingMethod, CryptoPP::SHA1> > >'} has no member named 'DEREncode'
2022-02-28T05:30:23.1977293Z    pubkey.DEREncode(asink);
2022-02-28T05:30:23.1977560Z           ^~~~~~~~~
2022-02-28T05:30:23.4832366Z make[7]: *** [Makefile:2620: amuled-ClientCreditsList.o] Error 1
2022-02-28T05:30:23.4840193Z make[6]: *** [Makefile:3411: all-recursive] Error 1
2022-02-28T05:30:23.4847965Z make[5]: *** [Makefile:588: all-recursive] Error 1
2022-02-28T05:30:23.4848587Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/aMule-2.3.2'
2022-02-28T05:30:23.4858955Z make[4]: *** [Makefile:495: all] Error 2
2022-02-28T05:30:23.4859697Z make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/amule'
2022-02-28T05:30:23.4860128Z time: package/feeds/packages/amule/compile#87.96#10.84#95.96
2022-02-28T05:30:23.4861146Z make[3]: *** [Makefile:114: /workdir/openwrt/build_dir/target-x86_64_musl/aMule-2.3.2/.built] Error 2
2022-02-28T05:30:23.4862924Z     ERROR: package/feeds/packages/amule failed to build.
2022-02-28T05:30:23.4867871Z make[2]: *** [package/Makefile:116: package/feeds/packages/amule/compile] Error 1
2022-02-28T05:30:23.4878402Z make[2]: Leaving directory '/workdir/openwrt'
2022-02-28T05:30:23.4897662Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-02-28T05:30:23.4898593Z make[1]: Leaving directory '/workdir/openwrt'
2022-02-28T05:30:23.4902892Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-02-28T05:30:23.4945175Z ##[error]Process completed with exit code 2.

WYC-2020, 2022-02-28:
https://github.com/WYC-2020/packages/commit/03cad6353755b7354418a15f98a285537fad87df
@Cptmacmillan2022007

WYC-2020, 2022-02-28:
自己多去各种库看，不要一遇到就来问嘛。别人早就修复了的

Beginner-Go, 2022-02-28:
feeds后试试？

netlovehf, 2022-03-01:
> 自己多去各种库看，不要一遇到就来问嘛。别人早就修复了的

git pull后编唐译还是同样错误，请问这是怎么回事，怎么解决？

WYC-2020, 2022-03-01:
什么错误，自己多检查，反正我没问题，最好直接用我那个luci和package仓库编译看，还有错误我才知道怎么修，其他杂交无能为力了，哈哈

coolsnowwolf, 2022-03-07:
fixed

closed 2022-03-07 by coolsnowwolf