coolsnowwolf/lede #8962: keepalived编译报错

dihao1127, 2022-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
keepalived编译报错
Q：是否搜索了issue (使用 "x" 选择)
[*  ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：2022-2-27  到2022-2-28日两天  无论在线编译离线编译均提示一样的错误，不编译keepalived就能成！

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

In file included from ../../keepalived/include/layer4.h:28,
                 from layer4.c:38:
/Lede_source/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/net/if.h:116:8: error: redefinition of 'struct ifconf'
 struct ifconf {
        ^~~~~~
In file included from /Lede_source/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/icmp.h:23,
                 from layer4.c:30:
/Lede_source/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/linux/if.h:284:8: note: originally defined here
 struct ifconf  {
        ^~~~~~
make[6]: *** [Makefile:410: layer4.o] Error 1
make[6]: Leaving directory '/Lede_source/openwrt/build_dir/target-x86_64_musl/keepalived-2.2.2/keepalived/core'
make[5]: *** [Makefile:558: all-recursive] Error 1
make[5]: Leaving directory '/Lede_source/openwrt/build_dir/target-x86_64_musl/keepalived-2.2.2/keepalived'
make[4]: *** [Makefile:440: all-recursive] Error 1
make[3]: *** [Makefile:275: /Lede_source/openwrt/build_dir/target-x86_64_musl/keepalived-2.2.2/.built] Error 2
make[4]: Leaving directory '/Lede_source/openwrt/build_dir/target-x86_64_musl/keepalived-2.2.2'
make[3]: Leaving directory '/Lede_source/openwrt/feeds/packages/net/keepalived'
time: package/feeds/packages/keepalived/compile#0.37#0.08#0.42
    ERROR: package/feeds/packages/keepalived failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/keepalived/compile] Error 1
make[2]: Leaving directory '/Lede_source/openwrt'
make[1]: *** [package/Makefile:110: /Lede_source/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Lede_source/openwrt'
make: *** [/Lede_source/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-28:
没遇到，我是gcc11

WYC-2020, 2022-02-28:
明天我试下编译keepalived看下

closed 2022-02-28 by dihao1127

dihao1127, 2022-02-28:
最后成功 不知道是因为内核用的5.12还是  只勾选keepalived 不勾二级菜单的原因成功了，希望不是偶然！
只编译了：
passwall
keepalived
Turbo ACC
filetransfer
luci-theme-argon