coolsnowwolf/lede #8963: Your BIOS Boot Partition is under 1 MiB, please increase its size..

Dawneng, 2022-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

提示该分区小于1M， 在Image Options 没有找相关调整选项，请问如何调整大小？

### (2) 路由器型号和固件版本

x86_64

### (3) 详细日志

openwrt/staging_dir/host/bin/grub-bios-setup: warning: Your BIOS Boot Partition is under 1 MiB, please increase its size..

WYC-2020, 2022-02-28:
这个问题我很早就发现了，只是找不到那个地方该哈哈

WYC-2020, 2022-02-28:
有大佬知道的提示下文件名即可

WYC-2020, 2022-03-01:
我终于修复了，看来还是得自己动手

Dawneng, 2022-03-01:
> 我终于修复了，看来还是得自己动手
求教

WYC-2020, 2022-03-01:
自己看仓库
@Dawneng

Dawneng, 2022-03-01:
> 自己看仓库 @Dawneng

好勒，谢谢大佬！

closed 2022-03-01 by Dawneng

YinYu86, 2022-03-24:
这个问题的答案是什么？在哪里修改？

chengjiaking, 2022-04-05:
在哪里看

leepentium, 2022-05-10:
同问，在哪里看？

xeyesen, 2022-06-29:
修改文件，/target/linux/x86/image/Makefile

define Build/combined
	$(CP) $(KDIR)/$(KERNEL_NAME) $@.boot/boot/vmlinuz
	-$(CP) $(STAGING_DIR_ROOT)/boot/. $@.boot/boot/
	$(CP) $(STAGING_DIR_IMAGE)/grub2/boot.img $@.boot/boot/grub/
	$(CP) $(STAGING_DIR_IMAGE)/grub2/$(if $(filter $(1),efi),gpt,$(GRUB2_VARIANT))-core.img \
		$@.boot/boot/grub/core.img
	$(if $(filter $(1),efi),
		$(INSTALL_DIR) $@.boot/efi/boot
		$(CP) $(STAGING_DIR_IMAGE)/grub2/boot$(if $(CONFIG_x86_64),x64,ia32).efi $@.boot/efi/boot/
	)
	PADDING="1" SIGNATURE="$(IMG_PART_SIGNATURE)" \
		$(if $(filter $(1),efi),GUID="$(IMG_PART_DISKGUID)") $(SCRIPT_DIR)/gen_image_generic.sh \
		$@ \
		$(CONFIG_TARGET_KERNEL_PARTSIZE) $@.boot \
		$(CONFIG_TARGET_ROOTFS_PARTSIZE) $(IMAGE_ROOTFS) \
		256
endef

找到这行，把256改成1024保存

@zmhu @leepentium @chengjiaking