coolsnowwolf/lede #8964: 编译报错，提示eip93-ring.c文件有错误

wang92332, 2022-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错，提示eip93模块有一个代码错误

### (2) 路由器型号和固件版本

A：xiaomi-R4A-千兆版

### (3) 详细日志

A：

```
make[3]: Entering directory '/home/why/openwrt/package/lean/mtk-eip93'
mkdir -p /home/why/openwrt/dl
SHELL= flock /home/why/openwrt/tmp/.mtk-eip93-2020-12-10-bb5fd1c8.tar.xz.flock -c '  	/home/why/openwrt/scripts/download.pl "/home/why/openwrt/dl" "mtk-eip93-2020-12-10-bb5fd1c8.tar.xz" "d2a4c5da7fb695e575de4b4335b73dcbfea195ef76791133d36287065a66386b" "" || (  /home/why/openwrt/scripts/dl_github_archive.py --dl-dir="/home/why/openwrt/dl" --url="https://github.com/vschagen/mtk-eip93.git" --version="bb5fd1c88a8a74de06b70c281bceeb14d0d960ef" --subdir="mtk-eip93-2020-12-10-bb5fd1c8" --source="mtk-eip93-2020-12-10-bb5fd1c8.tar.xz" --hash="d2a4c5da7fb695e575de4b4335b73dcbfea195ef76791133d36287065a66386b" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/why/openwrt/tmp/dl && cd /home/why/openwrt/tmp/dl && rm -rf mtk-eip93-2020-12-10-bb5fd1c8 && [ \! -d mtk-eip93-2020-12-10-bb5fd1c8 ] && git clone  https://github.com/vschagen/mtk-eip93.git mtk-eip93-2020-12-10-bb5fd1c8 && (cd mtk-eip93-2020-12-10-bb5fd1c8 && git checkout bb5fd1c88a8a74de06b70c281bceeb14d0d960ef && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd mtk-eip93-2020-12-10-bb5fd1c8 && git log -1 --format='\''@%ct'\''` && rm -rf mtk-eip93-2020-12-10-bb5fd1c8/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c mtk-eip93-2020-12-10-bb5fd1c8 | 	xz -zc -7e > /home/why/openwrt/tmp/dl/mtk-eip93-2020-12-10-bb5fd1c8.tar.xz && mv /home/why/openwrt/tmp/dl/mtk-eip93-2020-12-10-bb5fd1c8.tar.xz /home/why/openwrt/dl/ && rm -rf mtk-eip93-2020-12-10-bb5fd1c8; );  )    '
rm -f /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/.built
touch /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/.built_check
make -C "/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100" KCFLAGS="-fmacro-prefix-map=/home/why/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/why/openwrt/staging_dir/host/include -I/home/why/openwrt/staging_dir/hostpkg/include -I/home/why/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Feb 28 04:39:50 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/why/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/why/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mt76.symvers" KERNELRELEASE=5.10.100 M="/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src" modules
make[4]: Entering directory '/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100'
  CC [M]  /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-ring.o
/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-ring.c: In function 'mtk_get_descriptor':
/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-ring.c:77:3: error: implicit declaration of function 'dev_err'; did you mean 'xa_err'? [-Werror=implicit-function-declaration]
   dev_err(mtk->dev, "Cant get Cdesc");
   ^~~~~~~
   xa_err
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:280: /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src/eip93-ring.o] Error 1
make[4]: *** [Makefile:1822: /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/src] Error 2
make[4]: Leaving directory '/home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.100'
make[3]: *** [Makefile:61: /home/why/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mtk-eip93-2020-12-10-bb5fd1c8/.built] Error 2
make[3]: Leaving directory '/home/why/openwrt/package/lean/mtk-eip93'
time: package/lean/mtk-eip93/compile#0.94#0.25#1.06
    ERROR: package/lean/mtk-eip93 failed to build.
make[2]: *** [package/Makefile:116: package/lean/mtk-eip93/compile] Error 1
make[2]: Leaving directory '/home/why/openwrt'
make[1]: *** [package/Makefile:110: /home/why/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/why/openwrt'
make: *** [/home/why/openwrt/include/toplevel.mk:230: world] Error 2

```

WYC-2020, 2022-02-28:
@wang92332 
https://github.com/WYC-2020/lede/commit/55e500945580ca8558b0f30763a91c185b1d7971

wang92332, 2022-02-28:
> @wang92332 [WYC-2020/lede@55e5009](https://github.com/WYC-2020/lede/commit/55e500945580ca8558b0f30763a91c185b1d7971)

感谢大佬，成功了

closed 2022-02-28 by Beginner-Go