coolsnowwolf/lede #8965: 编译失败 mt76 failed to build

Nonosword, 2022-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
更换5.15编译后使用正常，但发现NetGear的A6210（MT7612U）插入没有无线设置选项，检查config后添加 CONFIG_PACKAGE_kmod-mt76x2u=y重新编译，两次尝试均触发该问题：ERROR: package/kernel/mt76 failed to build

### (2) 路由器型号和固件版本

A：Rocketchip-r4s-armv8-内核5.15

### (3) 详细日志

A：
 make[3] -C package/kernel/mt76 compile
    ERROR: package/kernel/mt76 failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 1
...
...
make[4]: Entering directory '/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25'
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/eeprom.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/mac.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/init.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/phy.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/mcu.o
  LD [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/mt76x2-common.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb_init.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb_main.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb_mac.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb_mcu.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/usb_phy.o
  LD [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mt76x2/mt76x2u.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mmio.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/util.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/trace.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/dma.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/mac80211.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/debugfs.o
  CC [M]  /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/eeprom.o
/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/eeprom.c: In function 'mt76_eeprom_override':
/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/eeprom.c:99:9: error: too few arguments to function 'of_get_mac_address'
   mac = of_get_mac_address(np);
         ^~~~~~~~~~~~~~~~~~
In file included from /home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/usr/include/mac80211-backport/linux/of_net.h:3,
                 from /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/eeprom.c:6:
./include/linux/of_net.h:16:12: note: declared here
 extern int of_get_mac_address(struct device_node *np, u8 *mac);
            ^~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:277: /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/eeprom.o] Error 1
make[4]: *** [Makefile:1868: /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d] Error 2
make[3]: *** [Makefile:427: /home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/mt76-2021-07-15-bbebea7d/.built] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.25'
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#11.31#1.83#13.16
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-02-28:
参考这个提交自己修复就是了，原理都差不多，懒得搞了https://github.com/coolsnowwolf/lede/commit/1c96c4ace0e84370c1a11ba1d4d4763c68a2c95d
@Nonosword

WYC-2020, 2022-02-28:
这很多都没编译过直接就切了，这个问题还会出现很多的

Nonosword, 2022-02-28:
> 参考这个提交自己修复就是了，原理都差不多，懒得搞了[1c96c4a](https://github.com/coolsnowwolf/lede/commit/1c96c4ace0e84370c1a11ba1d4d4763c68a2c95d) @Nonosword

好的谢谢，我去看看

WYC-2020, 2022-02-28:
@Nonosword 弄好没嘛，没有等会有时间给你修下

WYC-2020, 2022-03-01:
https://github.com/WYC-2020/lede/commit/3cc62304a7829cb0dc95328cb6809cf57e3dba40
@Nonosword

Nonosword, 2022-03-01:
> @Nonosword 弄好没嘛，没有等会有时间给你修下

感谢感谢[合十]，确实需要帮助，而且也没抽出时间来处理

ricksuzade-maker, 2022-03-01:
大雕已经修复了，快去试试吧

closed 2022-03-01 by Beginner-Go