coolsnowwolf/lede #8971: 春节后Newifi3新编译固件管理界面打不开  报 500 错误

leirace, 2022-03-01:
春节后新编译固件刷机后管理界面打不开  报 500 错误

* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：这段时间应该是春节后,同步代码后编译的固件就一直500错误, 一开始以为是配置没清空或者luci包原因, 但恢复默认设置, luci-app也全部用默认, 前后编译了很多次后还是一样的错误.

### (2) 路由器型号和固件版本

A：Newifi-D2

### (3) 详细日志

<html>
<body>

<!DOCTYPE html>
--
  | <html lang="zh-cn">
  | <head>
  | <meta charset="utf-8">
  | <title>Status: 500 Internal Server Error
  | /usr/lib/lua/luci/dispatcher.lua:474: Failed to execute function dispatcher target for entry '/'.
  | The called action terminated with an exception:
  | /usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'.
  | A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
  | A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/material/header'.
  | A runtime error occured: [string "/usr/lib/lua/luci/view/themes/material/head..."]:162: attempt to index local 'boardinfo' (a nil value)
  | stack traceback:
  | [C]: in function 'assert'
  | /usr/lib/lua/luci/dispatcher.lua:474: in function 'dispatch'
  | /usr/lib/lua/luci/dispatcher.lua:126: in function </usr/lib/lua/luci/dispatcher.lua:125>

</body>
</html>

yonggedebaqi, 2022-03-01:
我的编译正常

leirace, 2022-03-03:
> 

应该编译环境问题,清理了编译目录,全部重新编译后解决了, 谢谢!

closed 2022-03-03 by leirace