coolsnowwolf/lede #8972: IPQ40XX AC58U Target Images 选项很少

benathan, 2022-03-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

DigitalOcean部署全新Ubuntu 20.04 LTS x64，make menuconfig后选择IPQ40XX及AC58U时，Target Images 选项较少，无法指定如root filesystem partition size。反覆勾选squashfs/ext4/ramdisk没有效果(改变)，导致插件无法编译进固件。
[image]
### (2) 路由器型号和固件版本

IPQ40XX AC58U, 使用了71263b9

### (3) 详细日志

WYC-2020, 2022-03-01:
那就是别人不让你指定呗 哈哈

benathan, 2022-03-01:
> 那就是别人不让你指定呗 哈哈

还请指点迷津。我把要的插件都打*了，可最后还是编译出纯净版OpenWrt，只有一堆ipk。。

WYC-2020, 2022-03-01:
都给你说了 别人不让你自定义分区大小 你看他的打包文件嘛也知道嘛
[image]

WYC-2020, 2022-03-01:
你以为这种硬路由能像软路由哪有随便选插件撒，能有多大的存储空间嘛。明显你自己选插件多了超出大小了 肯定生不成镜像了
@benathan

benathan, 2022-03-01:
@WYC-2020 
原来如此，谢谢你

closed 2022-03-01 by benathan