coolsnowwolf/lede #8974: xiaomi R3 &xiaomi MINI编译报错

wokilp, 2022-03-02:
make[3]: Leaving directory '/home/test/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#0.44#0.12#0.54
mkdir -p /home/test/lede/dl
SHELL= flock /home/test/lede/tmp/.ppp-2.4.9.git-2021-01-04.tar.xz.flock -c '    /home/test/lede/scripts/download.pl "/home/test/lede/dl" "ppp-2.4.9.git-2021-01-04.tar.xz" "429cb5fcff36e1d8698766130711d4764347f08b83233dfb4831bea21616efef" "" || (  /home/test/lede/scripts/dl_github_archive.py --dl-dir="/home/test/lede/dl" --url="https://github.com/paulusmack/ppp" --version="4fb319056f168bb8379865b91b4fd3e1ada73f1e" --subdir="ppp-2.4.9.git-2021-01-04" --source="ppp-2.4.9.git-2021-01-04.tar.xz" --hash="429cb5fcff36e1d8698766130711d4764347f08b83233dfb4831bea21616efef" || (  echo "Checking out files from the git repository..."; mkdir -p /home/test/lede/tmp/dl && cd /home/test/lede/tmp/dl && rm -rf ppp-2.4.9.git-2021-01-04 && [ \! -d ppp-2.4.9.git-2021-01-04 ] && git clone  https://github.com/paulusmack/ppp ppp-2.4.9.git-2021-01-04 && (cd ppp-2.4.9.git-2021-01-04 && git checkout 4fb319056f168bb8379865b91b4fd3e1ada73f1e && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd ppp-2.4.9.git-2021-01-04 && git log -1 --format='\''@%ct'\''` && rm -rf ppp-2.4.9.git-2021-01-04/.git &&    tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c ppp-2.4.9.git-2021-01-04 |        xz -zc -7e > /home/test/lede/tmp/dl/ppp-2.4.9.git-2021-01-04.tar.xz && mv /home/test/lede/tmp/dl/ppp-2.4.9.git-2021-01-04.tar.xz /home/test/lede/dl/ && rm -rf ppp-2.4.9.git-2021-01-04; );  )    '
mkdir -p /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/test/lede/tmp/.root-copy.flock -c 'cp -fpR /home/test/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/ppp-default/ppp-2.4.9.git-2021-01-04/.pkgdir/ppp/. /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.ppp_installed
echo "ppp" >> /home/test/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/ppp.default.install
mkdir -p /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/test/lede/tmp/.root-copy.flock -c 'cp -fpR /home/test/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/ppp-default/ppp-2.4.9.git-2021-01-04/.pkgdir/ppp-mod-pppoe/. /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.ppp-mod-pppoe_installed
echo "ppp-mod-pppoe" >> /home/test/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/ppp.default.install
make[3]: Leaving directory '/home/test/lede/package/network/services/ppp'
time: package/network/services/ppp/default/compile#0.30#0.08#0.34
mkdir -p /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/test/lede/tmp/.root-copy.flock -c 'cp -fpR /home/test/lede/build_dir/target-mipsel_24kc_musl/luci-app-nlbwmon/.pkgdir/luci-app-nlbwmon/. /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-app-nlbwmon_installed
echo "luci-app-nlbwmon" >> /home/test/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-nlbwmon.default.install
mkdir -p /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/test/lede/tmp/.root-copy.flock -c 'cp -fpR /home/test/lede/build_dir/target-mipsel_24kc_musl/luci-app-nlbwmon/.pkgdir/luci-i18n-nlbwmon-zh-cn/. /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/test/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.luci-i18n-nlbwmon-zh-cn_installed
echo "luci-i18n-nlbwmon-zh-cn" >> /home/test/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/luci-app-nlbwmon.default.install
make[3]: Leaving directory '/home/test/lede/feeds/luci/applications/luci-app-nlbwmon'
time: package/feeds/luci/luci-app-nlbwmon/compile#0.44#0.12#0.50
make[2]: Leaving directory '/home/test/lede'
make[1]: *** [package/Makefile:110: /home/test/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/test/lede'
make: *** [/home/test/lede/include/toplevel.mk:230: world] Error 2
r3编译完成

wokilp, 2022-03-02:
拉的是3月2号的最新源文件

WYC-2020, 2022-03-02:
单线程日志

wokilp, 2022-03-02:
ok，我重新编译一下

wokilp, 2022-03-02:
> 单线程日志

感谢老哥，单线程编译过了

closed 2022-03-02 by wokilp

WYC-2020, 2022-03-02:
这个op 再第一次多线程编译的时候做的不是很好 容易出问题，不过我基本都是多线程编译，都会过，不过有时候会抽风

wokilp, 2022-03-02:
> 这个op 再第一次多线程编译的时候做的不是很好 容易出问题，不过我基本都是多线程编译，都会过，不过有时候会抽风

没事，我写了一个脚本，第一遍让他单线程先编译，没问题再批量编译。