coolsnowwolf/lede #8975: 3.2号版本编译报错tmp/.config-package.in:88488:error: recursive dependency detected!

wokilp, 2022-03-02:
Collecting package info: done
Collecting target info: done
tmp/.config-package.in:88488:error: recursive dependency detected!
tmp/.config-package.in:88488:   symbol PACKAGE_amule is selected by PACKAGE_antileech
tmp/.config-package.in:88516:   symbol PACKAGE_antileech is selected by PACKAGE_amule
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

这个amule插件，.config配置文件里面没有配置，为什么还会提示缺少依赖呢
menu "P2P"
        config PACKAGE_amule
                tristate "amule............................ A multi-platform eMule-like ed2k client"
                default y if DEFAULT_amule
                default m if ALL
                select PACKAGE_libreadline
                select PACKAGE_libupnp
                select PACKAGE_librt if USE_GLIBC
                select PACKAGE_antileech
                select PACKAGE_libintl-full if BUILD_NLS
                select PACKAGE_libcryptopp if !AMULE_CRYPTOPP_STATIC_LINKING
                select PACKAGE_libncurses
                select PACKAGE_libwxbase
                select PACKAGE_libiconv-full if BUILD_NLS
                select PACKAGE_libpthread
                select PACKAGE_libpthread if USE_GLIBC
                select PACKAGE_libc
                select PACKAGE_libpng
                select PACKAGE_libbfd
                help
                 A multi-platform eMule-like ed2k client
                 http://www.amule.org/

        config AMULE_CRYPTOPP_STATIC_LINKING
                bool "Link libcryptopp statically"
                default n

        config PACKAGE_antileech
                tristate "antileech............................................ AntiLeech for amule"
                default y if DEFAULT_antileech
                default m if ALL
                select PACKAGE_amule
                select PACKAGE_libwxbase
                select PACKAGE_libpthread if USE_GLIBC
                select PACKAGE_libc
                select PACKAGE_librt if USE_GLIBC
                help
                 AntiLeech for amule
                 http://www.amule.org/

WYC-2020, 2022-03-02:
你确定你这个是lede 的amule

wokilp, 2022-03-02:
> 你确定你这个是lede 的amule
git clone https://github.com/coolsnowwolf/lede
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
make -j8 download
make -j$(($(nproc) + 1)) V=s
里面的东西我没有改过，.config配置文件是之前创建好的，make defconfig执行了一下

WYC-2020, 2022-03-02:
编译电驴不需要选择编译antileech
[image]
看下这里是不是选择了
[image]

wokilp, 2022-03-02:
> 并且即使选择了也不会报错的，packages 下面有，你怕不是用的l的packages吧

这个是3月1号拉取的，之前的packages一点都没有留

WYC-2020, 2022-03-02:
那 rm -rf  feed

wokilp, 2022-03-02:
> 那 rm -rf feed

ok

WYC-2020, 2022-03-02:
或者直接万能大法，make distclean  然后重新 update 和install

wokilp, 2022-03-02:
> 或者直接万能大法，make distclean 然后重新 update 和install

老哥稳，已经ok了
代码拉取完成
Collecting package info: done
Collecting target info: done
#
# configuration written to .config
#
time: target/linux/prereq#0.36#0.04#0.39
 make[1] tools/flock/compile
 make[2] -C tools/flock compile
 make[1] tools/download
 make[2] -C tools/cpio download
 make[2] -C tools/cmake download
 make[2] -C tools/autoconf download
 make[2] -C tools/bc download
 make[2] -C tools/automake download
 make[2] -C tools/autoconf-archive download
 make[2] -C tools/bison download
 make[2] -C tools/dosfstools download

WYC-2020, 2022-03-02:
那就关闭issue

closed 2022-03-02 by wokilp