coolsnowwolf/lede #8978: X86 编译失败

skaterjuice, 2022-03-02:
make[5]: *** [scripts/Makefile.build:277: /home/masterbuilder/lede/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.o] Error 1
make[4]: *** [Makefile:1868: /home/masterbuilder/lede/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529] Error 2
make[3]: *** [Makefile:47: /home/masterbuilder/lede/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/.built] Error 2
make[2]: *** [package/Makefile:116: package/feeds/packages/antfs/compile] Error 1
make[1]: *** [package/Makefile:110: /home/masterbuilder/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/home/masterbuilder/lede/include/toplevel.mk:230: world] Error 2

wi11iamzhao, 2022-03-02:
same problem

shjohnnyyu, 2022-03-02:
都不先查issues的吗？ 同样的问题不是有解决方案。

skaterjuice, 2022-03-03:
> 

感谢您的专家建议。

closed 2022-03-03 by Beginner-Go