coolsnowwolf/lede #8981: 不管怎样编译都不成功，原版源码就成功就是什么问题

HAN5201, 2022-03-02:
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/hwdata-0.347/.pkgdir/pciids/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.pciids_installed
echo "pciids" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/hwdata.default.install
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/hwdata-0.347/.pkgdir/usbids/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.usbids_installed
echo "usbids" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/hwdata.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/hwdata'
time: package/feeds/packages/hwdata/compile#0.24#0.06#0.28
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/iputils'
Makefile:28: /workdir/openwrt/include/meson.mk: No such file or directory
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/iputils'
make[3]: *** No rule to make target '/workdir/openwrt/include/meson.mk'.  Stop.
time: package/feeds/packages/iputils/compile#0.15#0.03#0.17
    ERROR: package/feeds/packages/iputils failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/feeds/packages/iputils/compile] Error 1
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-03-02:
Makefile:28: /workdir/openwrt/include/meson.mk: No such file or directory
Makefile引用路径看下对了没

HAN5201, 2022-03-02:
这是云编译

AmadeusGhost, 2022-03-03:
显然你这个是自己杂交的源码。

closed 2022-03-03 by AmadeusGhost

GC-Lai, 2023-07-07:
new package put into old openwrt, it can't find meson.mk

-include $(INCLUDE_DIR)/meson.mk       to ignore