coolsnowwolf/lede #8987: antfs在线编译报错

Cptmacmillan2022007, 2022-03-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

2022-03-02T07:29:10.3169273Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/kernel/antfs'
2022-03-02T07:29:10.4941493Z mkdir -p /workdir/openwrt/dl
2022-03-02T07:29:10.4972763Z SHELL= flock /workdir/openwrt/tmp/.antfs-2020-02-10-b41ba529.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "antfs-2020-02-10-b41ba529.tar.xz" "dae039c0fe5bf1a2c8c1cca4211d607a4d6f56fc41b38444e2234b40d710d9db" "" || (  /workdir/openwrt/scripts/dl_github_archive.py --dl-dir="/workdir/openwrt/dl" --url="https://github.com/klukonin/antfs.git" --version="b41ba529f6b64b429527e09a06ce0326d5456c05" --subdir="antfs-2020-02-10-b41ba529" --source="antfs-2020-02-10-b41ba529.tar.xz" --hash="dae039c0fe5bf1a2c8c1cca4211d607a4d6f56fc41b38444e2234b40d710d9db" || ( 	echo "Checking out files from the git repository..."; mkdir -p /workdir/openwrt/tmp/dl && cd /workdir/openwrt/tmp/dl && rm -rf antfs-2020-02-10-b41ba529 && [ \! -d antfs-2020-02-10-b41ba529 ] && git clone  https://github.com/klukonin/antfs.git antfs-2020-02-10-b41ba529 && (cd antfs-2020-02-10-b41ba529 && git checkout b41ba529f6b64b429527e09a06ce0326d5456c05 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd antfs-2020-02-10-b41ba529 && git log -1 --format='\''@%ct'\''` && rm -rf antfs-2020-02-10-b41ba529/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c antfs-2020-02-10-b41ba529 | 	xz -zc -7e > /workdir/openwrt/tmp/dl/antfs-2020-02-10-b41ba529.tar.xz && mv /workdir/openwrt/tmp/dl/antfs-2020-02-10-b41ba529.tar.xz /workdir/openwrt/dl/ && rm -rf antfs-2020-02-10-b41ba529; );  )    '
2022-03-02T07:29:10.5487506Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/.built
2022-03-02T07:29:10.5516429Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/.built_check
2022-03-02T07:29:10.5549821Z make -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529" CONFIG_ANTFS_FS=m CONFIG_ANTFS_SYMLINKS=y ANTFS_VERSION=07.19-b41ba529 modules
2022-03-02T07:29:10.5604246Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25'
2022-03-02T07:29:11.0289880Z   CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.o
2022-03-02T07:29:11.5515167Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c: In function 'antfs_setattr':
2022-03-02T07:29:11.5516557Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1106:15: error: passing argument 1 of 'setattr_copy' from incompatible pointer type [-Werror=incompatible-pointer-types]
2022-03-02T07:29:11.5517217Z   setattr_copy(inode, attr);
2022-03-02T07:29:11.5517546Z                ^~~~~
2022-03-02T07:29:11.5518218Z In file included from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/antfs.h:171,
2022-03-02T07:29:11.5518982Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:21:
2022-03-02T07:29:11.5519712Z ./include/linux/fs.h:3448:19: note: expected 'struct user_namespace *' but argument is of type 'struct inode *'
2022-03-02T07:29:11.5520236Z  void setattr_copy(struct user_namespace *, struct inode *inode,
2022-03-02T07:29:11.5520627Z                    ^~~~~~~~~~~~~~~~~~~~~~~
2022-03-02T07:29:11.5521745Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1106:22: error: passing argument 2 of 'setattr_copy' from incompatible pointer type [-Werror=incompatible-pointer-types]
2022-03-02T07:29:11.5523817Z   setattr_copy(inode, attr);
2022-03-02T07:29:11.5524107Z                       ^~~~
2022-03-02T07:29:11.5524764Z In file included from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/antfs.h:171,
2022-03-02T07:29:11.5525505Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:21:
2022-03-02T07:29:11.5526215Z ./include/linux/fs.h:3448:58: note: expected 'struct inode *' but argument is of type 'struct iattr *'
2022-03-02T07:29:11.5526697Z  void setattr_copy(struct user_namespace *, struct inode *inode,
2022-03-02T07:29:11.5527078Z                                             ~~~~~~~~~~~~~~^~~~~
2022-03-02T07:29:11.5527789Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1106:2: error: too few arguments to function 'setattr_copy'
2022-03-02T07:29:11.5528283Z   setattr_copy(inode, attr);
2022-03-02T07:29:11.5528559Z   ^~~~~~~~~~~~
2022-03-02T07:29:11.5529174Z In file included from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/antfs.h:171,
2022-03-02T07:29:11.5529914Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:21:
2022-03-02T07:29:11.5530366Z ./include/linux/fs.h:3448:6: note: declared here
2022-03-02T07:29:11.5530765Z  void setattr_copy(struct user_namespace *, struct inode *inode,
2022-03-02T07:29:11.5531091Z       ^~~~~~~~~~~~
2022-03-02T07:29:11.5533418Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c: In function 'antfs_getattr':
2022-03-02T07:29:11.5536638Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1224:19: error: passing argument 1 of 'generic_fillattr' from incompatible pointer type [-Werror=incompatible-pointer-types]
2022-03-02T07:29:11.5537276Z   generic_fillattr(inode, stat);
2022-03-02T07:29:11.5537558Z                    ^~~~~
2022-03-02T07:29:11.5540050Z In file included from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/antfs.h:171,
2022-03-02T07:29:11.5540819Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:21:
2022-03-02T07:29:11.5541545Z ./include/linux/fs.h:3328:23: note: expected 'struct user_namespace *' but argument is of type 'struct inode *'
2022-03-02T07:29:11.5554864Z  void generic_fillattr(struct user_namespace *, struct inode *, struct kstat *);
2022-03-02T07:29:11.5556078Z                        ^~~~~~~~~~~~~~~~~~~~~~~
2022-03-02T07:29:11.5557060Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1224:26: error: passing argument 2 of 'generic_fillattr' from incompatible pointer type [-Werror=incompatible-pointer-types]
2022-03-02T07:29:11.5557637Z   generic_fillattr(inode, stat);
2022-03-02T07:29:11.5557912Z                           ^~~~
2022-03-02T07:29:11.5558467Z In file included from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/antfs.h:171,
2022-03-02T07:29:11.5559147Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:21:
2022-03-02T07:29:11.5559775Z ./include/linux/fs.h:3328:48: note: expected 'struct inode *' but argument is of type 'struct kstat *'
2022-03-02T07:29:11.5560233Z  void generic_fillattr(struct user_namespace *, struct inode *, struct kstat *);
[... 27 lines cut ...]
2022-03-02T07:29:11.5590177Z   .getattr = antfs_getattr,
2022-03-02T07:29:11.5590431Z              ^~~~~~~~~~~~~
2022-03-02T07:29:11.5591105Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1425:13: note: (near initialization for 'antfs_common_inode_operations.getattr')
2022-03-02T07:29:11.5592499Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/antfs-2020-02-10-b41ba529/dir.c:1441:13: error: initialization of 'int (*)(struct user_namespace *, struct dentry *, struct iattr *)' from incompatible pointer type 'int (*)(struct dentry *, struct iattr *)' [-Werror=incompatible-pointer-types]
2022-03-02T07:29:11.5593218Z   .setattr = antfs_setattr,
2022-03-02T07:29:11.5593470Z              ^~~~~~~~~~~~~
[... cut ...]
2022-03-02T07:29:11.5791641Z make[2]: *** [package/Makefile:116: package/feeds/packages/antfs/compile] Error 1
2022-03-02T07:29:11.5798981Z make[2]: Leaving directory '/workdir/openwrt'
2022-03-02T07:29:11.5804256Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2022-03-02T07:29:11.5816588Z make[1]: Leaving directory '/workdir/openwrt'
2022-03-02T07:29:11.5823675Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2022-03-02T07:29:11.5914258Z ##[error]Process completed with exit code 2.

WYC-2020, 2022-03-03:
说了几百万遍了 linux 5.15 不要搭配antfs

Cptmacmillan2022007, 2022-03-03:
> 说了几百万遍了 linux 5.15 不要搭配antfs

可以在5.15上用哪个ntfs模块啊？

closed 2022-03-03 by Cptmacmillan2022007