coolsnowwolf/lede #8990: 最新源码报错，x64编译通不过

Yofo-me, 2022-03-03:
### 日志如下
```shell
make[4]: Leaving directory '/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
rm -rf /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/modules
make  -C "/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1" KCFLAGS="-fmacro-prefix-map=/home/yoki/openwrt-lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/yoki/openwrt-lede/staging_dir/host/include -I/home/yoki/openwrt-lede/staging_dir/hostpkg/include -I/home/yoki/openwrt-lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Mar  1 18:21:12 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/yoki/openwrt-lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/yoki/openwrt-lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/ksmbd.symvers" KERNELRELEASE=5.15.25 EXTRA_CFLAGS="-I/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include -fmacro-prefix-map=/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1=backports-5.10.42-1  -I/home/yoki/openwrt-lede/staging_dir/target-x86_64_musl/usr/include/qca-nss-drv" KLIB_BUILD="/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.25" MODPROBE=true KLIB=/lib/modules/5.15.25 KERNEL_SUBLEVEL=15 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[4]: Entering directory '/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/compat/main.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/compat/drivers-base-devcoredump.o
  LD [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/compat/compat.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/caps.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/initvals.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/eeprom.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/gpio.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/desc.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/dma.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/qcu.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/pcu.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/phy.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/reset.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/attach.o
  CC [M]  /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.o
/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.c:93:25: error: expected declaration specifiers or '...' before string constant
 MODULE_SUPPORTED_DEVICE("Atheros 5xxx WLAN cards");
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:277: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k/base.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath5k] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1868: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: *** [Makefile:584: /home/yoki/openwrt-lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[3]: Leaving directory '/home/yoki/openwrt-lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#22.43#16.80#38.83
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/yoki/openwrt-lede'
make[1]: *** [package/Makefile:110: /home/yoki/openwrt-lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/openwrt-lede'
make: *** [/home/yoki/openwrt-lede/include/toplevel.mk:230: world] Error 2
```

make clean && make -j1 V=s还是不行，

WYC-2020, 2022-03-03:
补丁被人revert了

WYC-2020, 2022-03-03:
https://github.com/WYC-2020/lede/commit/9ea4c2e7634d73645bd3274f2d5fd8437580ea77

breakings, 2022-03-03:
> [WYC-2020/lede@9ea4c2e](https://github.com/WYC-2020/lede/commit/9ea4c2e7634d73645bd3274f2d5fd8437580ea77)

bison: error while loading shared libraries: libtextstyle.so.0: cannot open shared object file: No such file or directory 编译工具链和内核这个文件也说找不到了，不知道哪出了问题

wi11iamzhao, 2022-03-03:
same problem with kernel version 5.15 on x86_64

lxl6125, 2022-03-03:
> > [WYC-2020/lede@9ea4c2e](https://github.com/WYC-2020/lede/commit/9ea4c2e7634d73645bd3274f2d5fd8437580ea77)
> 
> bison: error while loading shared libraries: libtextstyle.so.0: cannot open shared object file: No such file or directory 编译工具链和内核这个文件也说找不到了，不知道哪出了问题

一样报错
bison: error while loading shared libraries: libtextstyle.so.0: cannot open shared object file: No such file or directory

WYC-2020, 2022-03-03:
反正我没遇到哈哈

breakings, 2022-03-03:
> 一样报错 bison: error while loading shared libraries: libtextstyle.so.0: cannot open shared object file: No such file or directory

bison 升级到3.8.2就好了

closed 2022-03-03 by Beginner-Go

WYC-2020, 2022-03-04:
这个错误bison没任何关系

WYC-2020, 2022-03-04:
主题问题和bios没任何关系，不要乱搞，导致issue被误关闭了

Ray-Eldath, 2022-04-04:
> bison 升级到3.8.2就好了

你好，我的bison是3.8.2，编译内核的时候还是有这个问题，请问你是怎么修复的呢？