coolsnowwolf/lede #8999: 编译错误，请教大佬！

sjkhsl, 2022-03-04:
错误情况：
mkdir -p /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi
sed 's#msdos1#gpt1#g' ./files/grub-early.cfg >/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg
/home/sjkhsl/openwrt/lede/staging_dir/host/bin/grub-mkimage -d /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-early.cfg -o /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/image/grub2/bootx64.efi at_keyboard boot chain configfile fat linux ls part_gpt reboot serial efi_gop efi_uga
/home/sjkhsl/openwrt/lede/staging_dir/host/bin/grub-mkimage -d /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/grub-core -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_keyboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi.installed
mkdir -p /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/sjkhsl/openwrt/lede/tmp/.root-copy.flock -c 'cp -fpR /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.06/.pkgdir/grub2-efi/. /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
echo "grub2-efi" >> /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/sjkhsl/openwrt/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#72.75#8.67#78.65
make[2]: Leaving directory '/home/sjkhsl/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/openwrt/lede'
make: *** [/home/sjkhsl/openwrt/lede/include/toplevel.mk:230：world] 错误 2

自己搜索过错误问题，找不到解决办法啊！求教大佬～

WYC-2020, 2022-03-04:
单线程日志

sjkhsl, 2022-03-04:
> 单线程日志

什么意思？

LPDDR6-10000MHz, 2022-03-04:
> > 单线程日志
> 
> 什么意思？

执行 make -j1 V=s 再上日志

sjkhsl, 2022-03-04:
Applying ./patches/brcm/998-survey.patch using plaintext: 
patching file drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c
patching file drivers/net/wireless/broadcom/brcm80211/brcmfmac/core.c
patching file drivers/net/wireless/broadcom/brcm80211/brcmfmac/core.h

Applying ./patches/rtl/002-v5.13-rtlwifi-implement-set_tim-by-update-beacon-content.patch using plaintext: 
patching file drivers/net/wireless/realtek/rtlwifi/core.c
patching file drivers/net/wireless/realtek/rtlwifi/core.h
patching file drivers/net/wireless/realtek/rtlwifi/usb.c
patching file drivers/net/wireless/realtek/rtlwifi/wifi.h
tar -C /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1 -xzf /home/sjkhsl/openwrt/lede/dl/ipw2100-fw-1.3.tgz
tar -C /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1 -xzf /home/sjkhsl/openwrt/lede/dl/ipw2200-fw-3.1.tgz
tar -C /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1 -xjf /home/sjkhsl/openwrt/lede/dl/zd1211-firmware-1.4.tar.bz2
rm -rf /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/ssb /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/bcma /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/net/bluetooth
rm -f /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/cordic.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/crc8.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/eeprom_93cx6.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/wl12xx.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/spi/libertas_spi.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/net/ieee80211.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/bcm47xx_nvram.h
echo 'compat-wireless-5.10.42-1-3-r4313-4a8985e5e' > /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/compat_version
touch /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.prepared_586b16e8464c5776711f03a138532741_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.configured_*
rm -f /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.mac80211.regular_installed
cmp /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/ath9k_platform.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.26/include/linux/ath9k_platform.h
cmp /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/ath5k_platform.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.26/include/linux/ath5k_platform.h
cmp /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include/linux/rt2x00_platform.h /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.26/include/linux/rt2x00_platform.h
touch /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.configured_de6a90e803509f4116af53bce7bd2371
rm -f /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built
touch /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built_check
. /home/sjkhsl/openwrt/lede/include/shell.sh; var2file "V_mac80211_config" /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.config
make -C "/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1" KCFLAGS="-fmacro-prefix-map=/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/sjkhsl/openwrt/lede/staging_dir/host/include -I/home/sjkhsl/openwrt/lede/staging_dir/hostpkg/include -I/home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Mar  4 05:11:22 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/sjkhsl/openwrt/lede/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/sjkhsl/openwrt/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/ksmbd.symvers /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers" KERNELRELEASE=5.15.26 EXTRA_CFLAGS="-I/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/include -fma [... cut ...]
time: package/kernel/mac80211/regular/compile#7.44#0.86#8.13
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/openwrt/lede'
make: *** [/home/sjkhsl/openwrt/lede/include/toplevel.mk:230：world] 错误 2

sjkhsl, 2022-03-04:
> 单线程日志

麻烦大佬在看一下，我删除后重新拉库编译，还是这个错误。

sjkhsl, 2022-03-04:
> > > 单线程日志
> > 
> > 
> > 什么意思？
> 
> 执行 make -j1 V=s 再上日志

哦哦，我上传了，大佬帮忙看看！

WYC-2020, 2022-03-04:
这个问题我说过，revert把有一个补丁去除了。自己加回来

WYC-2020, 2022-03-04:
居然issue还被关闭了，那个问题还没解决的，你看这个issue，我说了解决办法
https://github.com/coolsnowwolf/lede/issues/8990

sjkhsl, 2022-03-04:
> 居然issue还被关闭了，那个问题还没解决的，你看这个issue，我说了解决办法 #8990

目录里没有这个文件，自己添加一个吗？
target/linux/generic/backport-5.15/003-add-module_supported_device-macro.patch

WYC-2020, 2022-03-04:
> > 居然issue还被关闭了，那个问题还没解决的，你看这个issue，我说了解决办法 #8990
> 
> 目录里没有这个文件，自己添加一个吗？ target/linux/generic/backport-5.15/003-add-module_supported_device-macro.patch

你这问的，都被撤销了肯定没有了，自己按我给的那个加上即可

sjkhsl, 2022-03-04:
> > > 居然issue还被关闭了，那个问题还没解决的，你看这个issue，我说了解决办法 #8990
> > 
> > 
> > 目录里没有这个文件，自己添加一个吗？ target/linux/generic/backport-5.15/003-add-module_supported_device-macro.patch
> 
> 你这问的，都被撤销了肯定没有了，自己按我给的那个加上即可

文件手动添加了，那这是源码的问题？
多线程跑了一遍又报错了，单线程跑一遍试试！

sjkhsl, 2022-03-04:
> > > 居然issue还被关闭了，那个问题还没解决的，你看这个issue，我说了解决办法 #8990
> > 
> > 
> > 目录里没有这个文件，自己添加一个吗？ target/linux/generic/backport-5.15/003-add-module_supported_device-macro.patch
> 
> 你这问的，都被撤销了肯定没有了，自己按我给的那个加上即可

# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o
/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c: In function 'rt2x00lib_set_mac_address':
/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:1000:13: error: too few arguments to function 'of_get_mac_address'
  mac_addr = of_get_mac_address(rt2x00dev->dev->of_node);
             ^~~~~~~~~~~~~~~~~~
In file included from /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/of_net.h:3,
                 from /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:19:
./include/linux/of_net.h:25:19: note: declared here
 static inline int of_get_mac_address(struct device_node *np, u8 *mac)
                   ^~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:277: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink] Error 2
make[8]: *** [scripts/Makefile.build:540: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless] Error 2
make[7]: *** [Makefile:1868: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1'
make[3]: *** [Makefile:584: /home/sjkhsl/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/.built] Error 2
make[3]: Leaving directory '/home/sjkhsl/openwrt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#0.78#0.12#0.88
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/openwrt/lede'
make: *** [/home/sjkhsl/openwrt/lede/include/toplevel.mk:230：world] 错误 2

还是错误！

WYC-2020, 2022-03-04:
> # configuration written to .config
> Building backport-include/backport/autoconf.h ... done. CC [M] [...]

这是另外一个错误，这个错误有修复案列，如果急自己修复一下就可以了，不急等我有时间看

WYC-2020, 2022-03-04:
@sjkhsl https://github.com/WYC-2020/lede/commit/893ba3d9e6984f90560a0f93921f651ee3ae96cf

WYC-2020, 2022-03-04:
@sjkhsl 以前修复过了，刚看了一下哈哈

sjkhsl, 2022-03-05:
> @sjkhsl以前修复过了，刚看了哈哈

大佬，我直接下载你的lede库编译，还是保存～最近源码更新出问题了吧？

WYC-2020, 2022-03-05:
@sjkhsl 祖国的花朵你选的驱动有关，都是遇到错误修复，我不可能全修复，我这边编译我的需求正常我就没管了，你遇到错误贴出来就行，我看到我有时间会修复

closed 2022-03-07 by coolsnowwolf