coolsnowwolf/lede #9001: 编译问题make menuconfig

dsyo2008, 2022-03-04:
make menuconfig时发现了passwall不在luci里，确实加feeds更新没问题。前几天一直正常，现在习惯清理并更新，然后发现这passwall不在luci里。那passwall关issues，没法反馈，只好在这里反馈。非常感谢

WYC-2020, 2022-03-04:
我这里一切正常，你怕是用的源有问题

dsyo2008, 2022-03-04:
> 我这里一切正常，你怕是用的源有问题

你是哪个源？我是用xiaorouji的passwall，清理并更新后没问题，但是不在luci真奇怪

WYC-2020, 2022-03-04:
> > 我这里一切正常，你怕是用的源有问题
> 
> 你是哪个源？我是用xiaorouji的passwall，清理并更新后没问题，但是不在luci真奇怪

小肉鸡的

dsyo2008, 2022-03-04:
> > > 我这里一切正常，你怕是用的源有问题
> > 
> > 
> > 你是哪个源？我是用xiaorouji的passwall，清理并更新后没问题，但是不在luci真奇怪
> 
> 小肉鸡的

= =一样啊，我为什么找不到luci里毛？

WYC-2020, 2022-03-04:
> > > > 我这里一切正常，你怕是用的源有问题
> > > 
> > > 
> > > 你是哪个源？我是用xiaorouji的passwall，清理并更新后没问题，但是不在luci真奇怪
> > 
> > 
> > 小肉鸡的
> 
> = =一样啊，我为什么找不到luci里毛？

你是不是没有install哦，不行就make distclean后全新搞

dsyo2008, 2022-03-04:
> 

当然，其实是删除源，重新拉源，然后软件源更新并安装正常，没任何错误，结果luci没找到passwall真见鬼

WYC-2020, 2022-03-04:
> > 
> 
> 当然，其实是删除源，重新拉源，然后软件源更新并安装正常，没任何错误，结果luci没找到passwall真见鬼

不会吧，我一个小时前还用过勒，难道你姿势不对哈哈

dsyo2008, 2022-03-04:
> > > 
> > 
> > 
> > 当然，其实是删除源，重新拉源，然后软件源更新并安装正常，没任何错误，结果luci没找到passwall真见鬼
> 
> 不会吧，我一个小时前还用过勒，难道你姿势不对哈哈

绝对操作没问题
[image]

momonoshiori, 2022-03-04:
自己去项目地址看，Special circumstances, stop open.

dsyo2008, 2022-03-04:
> 

卧槽，才发现了。。。又失去这项目，难道继续用SSRP么

closed 2022-03-04 by Beginner-Go

lxl6125, 2022-03-04:
https://github.com/xiaorouji/openwrt-passwall/tree/3aff3af88536227d12fb7206992af64ff21cf4d2/
手动下载吧，2022-03-03 23:55最后更新的

WYC-2020, 2022-03-04:
还好我有fork仓库的习惯，我仓库应该是最新的

dsyo2008, 2022-03-05:
> 

我用了你的源OK了哈

yjs507, 2022-03-05:
> 

这个要怎么搞呀，刚学编译，不懂怎么操作

violetice, 2023-07-24:
> 你是不是没有install哦，不行就make distclean后全新搞

我今天也这样，就是没有install，笑死。