coolsnowwolf/lede #9004: ISP释放ip后PPPoE重连成功，但是没有互联网

Jermin, 2022-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

ISP释放ip后PPPoE重连成功，但是无法访问互联网

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：R2S OpenWrt R22.2.2 / LuCI Master (git-22.052.34668-380e994)

### (3) 详细日志

A：

```
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is now down
Fri Mar  4 21:20:19 2022 kern.info kernel: [553088.870734] r8152 4-1:1.0 eth1: carrier on
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is disabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is enabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is setting up now
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Mar  4 21:20:19 2022 kern.info kernel: [553088.918805] r8152 4-1:1.0 eth1: carrier on
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is now down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is disabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is enabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is setting up now
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Mar  4 21:20:19 2022 daemon.err insmod: module is already loaded - slhc
Fri Mar  4 21:20:19 2022 daemon.err insmod: module is already loaded - ppp_generic
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Mar  4 21:20:19 2022 kern.info kernel: [553088.967082] r8152 4-1:1.0 eth1: carrier on
Fri Mar  4 21:20:19 2022 daemon.err insmod: module is already loaded - pppox
Fri Mar  4 21:20:19 2022 daemon.err insmod: module is already loaded - pppoe
Fri Mar  4 21:20:19 2022 daemon.notice netifd: wan (18157): uci: Entry not found
Fri Mar  4 21:20:19 2022 daemon.notice netifd: wan (18157): sh: out of range
Fri Mar  4 21:20:19 2022 daemon.notice netifd: wan (18157): Command failed: Permission denied
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is now down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is disabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is enabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is setting up now
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Mar  4 21:20:19 2022 kern.info kernel: [553089.063048] r8152 4-1:1.0 eth1: carrier on
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is now down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is disabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is enabled
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' is setting up now
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is down
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Network device 'eth1' link is up
Fri Mar  4 21:20:19 2022 daemon.notice netifd: Interface 'wan' has link connectivity
[... 713 lines cut ...]
Fri Mar  4 21:21:09 2022 daemon.info pppd[20467]: Renamed interface ppp0 to pppoe-wan
Fri Mar  4 21:21:09 2022 daemon.info pppd[20467]: Using interface pppoe-wan
Fri Mar  4 21:21:09 2022 daemon.notice pppd[20467]: Connect: pppoe-wan <--> eth1
Fri Mar  4 21:21:12 2022 daemon.info pppd[20467]: syncppp not active
Fri Mar  4 21:21:12 2022 daemon.info pppd[20467]: Remote message: Authentication success,Welcome!
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: PAP authentication succeeded
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: peer from calling number A0:F4:79:7F:37:A9 authorized
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: local  LL address fe80::c4ca:8c3e:a6fe:ebfe
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: remote LL address fe80::a2f4:79ff:fe7f:37a9
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: local  IP address 100.72.47.206
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: remote IP address 100.72.0.1
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: primary   DNS address 58.22.96.66
Fri Mar  4 21:21:12 2022 daemon.notice pppd[20467]: secondary DNS address 218.104.128.106
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Interface 'wan' is now up
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Fri Mar  4 21:21:12 2022 daemon.notice netifd: Interface 'wan_6' is now up
Fri Mar  4 21:21:27 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Mar  4 21:21:28 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (pppoe-wan)
Fri Mar  4 21:21:28 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Fri Mar  4 21:21:44 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan_6 (pppoe-wan)
```

WYC-2020, 2022-03-04:
我这边也是拨号也会断网重连没你这个问题

violarulan, 2022-05-16:
同遇到类似问题，5.15内核，我是使用 ifup wan 重连，R4S

pppoe成功获取到了ipv4地址，但是路由器自身和下边设备无法上网

laaArmy, 2022-06-27:
把WAN6删除掉

violarulan, 2022-06-27:
更新：我是阉割版r4s的，目前已解决问题：手动修改一个网口的mac。原来两个网口的mac地址是一样的。

82kg, 2022-11-12:
ssh或TTYD进去
在/etc/ppp/options中加入下面两行
`maxfail 0`
`persist`

closed 2026-09-05 by coolsnowwolf