coolsnowwolf/lede #9006: 更新最新源码编译X86_64 minidlna出错？

mitsukileung, 2022-03-04:
X86_64

install -d -m0755 /home/koyko/lede/bin/packages/x86_64/packages
/home/koyko/lede/staging_dir/host/bin/fakeroot /home/koyko/lede/scripts/ipkg-build -m "" /home/koyko/lede/build_dir/target-x86_64_musl/opus-1.3.1/ipkg-x86_64/libopus /home/koyko/lede/bin/packages/x86_64/packages
Packaged contents of /home/koyko/lede/build_dir/target-x86_64_musl/opus-1.3.1/ipkg-x86_64/libopus into /home/koyko/lede/bin/packages/x86_64/packages/libopus_1.3.1-3_x86_64.ipk
echo "libopus" >> /home/koyko/lede/staging_dir/target-x86_64_musl/pkginfo/opus.default.install
make[3]: Leaving directory '/home/koyko/lede/feeds/packages/libs/opus'
time: package/feeds/packages/opus/compile#0.48#0.13#0.58
make[3]: Entering directory '/home/koyko/lede/feeds/packages/multimedia/ffmpeg'
mkdir -p /home/koyko/lede/dl
SHELL= flock /home/koyko/lede/tmp/.ffmpeg-5.0.tar.xz.flock -c '  	/home/koyko/lede/scripts/download.pl "/home/koyko/lede/dl" "ffmpeg-5.0.tar.xz" "51e919f7d205062c0fd4fae6243a84850391115104ccf1efc451733bc0ac7298" "" "https://ffmpeg.org/releases/"    '
mkdir -p /home/koyko/lede/bin/targets/x86/64/packages /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/CONTROL /home/koyko/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib
cp -fpR /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-install/usr/lib/lib{avcodec,avdevice,avfilter,avformat,avutil,swresample,swscale}.so.* /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/
find /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/koyko/lede/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/koyko/lede/scripts/strip-kmod.sh" PATCHELF="/home/koyko/lede/staging_dir/host/bin/patchelf" /home/koyko/lede/scripts/rstrip.sh /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libavdevice.so.59.4.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libavutil.so.57.17.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libavfilter.so.8.24.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libavcodec.so.59.18.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libswscale.so.6.4.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libavformat.so.59.16.100: shared object
rstrip.sh: /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/usr/lib/libswresample.so.4.3.100: shared object
(cd /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/koyko/lede/bin/packages/x86_64/packages
/home/koyko/lede/staging_dir/host/bin/fakeroot /home/koyko/lede/scripts/ipkg-build -m "" /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full /home/koyko/lede/bin/packages/x86_64/packages
Packaged contents of /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/ipkg-x86_64/libffmpeg-full into /home/koyko/lede/bin/packages/x86_64/packages/libffmpeg-full_5.0-1_x86_64.ipk
mkdir -p /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/koyko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koyko/lede/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-5.0/.pkgdir/libffmpeg-full/. /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libffmpeg-full_installed
echo "libffmpeg-full" >> /home/koyko/lede/staging_dir/target-x86_64_musl/pkginfo/ffmpeg.full.install
make[3]: Leaving directory '/home/koyko/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#1.74#0.48#2.10
make[3]: Entering directory '/home/koyko/lede/feeds/packages/multimedia/minidlna'
mkdir -p /home/koyko/lede/dl
SHELL= flock /home/koyko/lede/tmp/.minidlna-1.3.0.tar.gz.flock -c '  	/home/koyko/lede/scripts/download.pl "/home/koyko/lede/dl" "minidlna-1.3.0.tar.gz" "47d9b06b4c48801a4c1112ec23d24782728b5495e95ec2195bbe5c81bc2d3c63" "" "@SF/minidlna"    '
mkdir -p /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/koyko/lede/tmp/.root-copy.flock -c 'cp -fpR /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/.pkgdir/minidlna/. /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/koyko/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.minidlna_installed
mkdir -p /home/koyko/lede/bin/targets/x86/64/packages /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/CONTROL /home/koyko/lede/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/usr/sbin
install -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-install/usr/sbin/minidlnad /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/usr/sbin/minidlnad
install -d -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/init.d
install -m0755 ./files/minidlna.init /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/init.d/minidlna
install -d -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/config
install -m0600 ./files/minidlna.config /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/config/minidlna
install -d -m0755 /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/sysctl.d
install -m0600 ./files/minidlna.sysctl /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna/etc/sysctl.d/30-minidlna.conf
find /home/koyko/lede/build_dir/target-x86_64_musl/minidlna-1.3.0/ipkg-x86_64/minidlna -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package minidlna is missing dependencies for the following libraries:
libavcodec.so.58
libavformat.so.58
libavutil.so.56
make[3]: *** [Makefile:83: /home/koyko/lede/bin/packages/x86_64/packages/minidlna_1.3.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/koyko/lede/feeds/packages/multimedia/minidlna'
time: package/feeds/packages/minidlna/compile#0.34#0.14#0.44
    ERROR: package/feeds/packages/minidlna failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/minidlna/compile] Error 1
make[2]: Leaving directory '/home/koyko/lede'
make[1]: *** [package/Makefile:110: /home/koyko/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/koyko/lede'
make: *** [/home/koyko/lede/include/toplevel.mk:230：world] 错误 2

大佬们，这个错误要怎么解决？

Beginner-Go, 2022-03-04:
别急 我看看

mitsukileung, 2022-03-04:
> 别急 我看看

谢谢大佬

Beginner-Go, 2022-03-04:
@WYC-2020 大佬 您看看？

Beginner-Go, 2022-03-04:
说是缺少依赖库

baodp, 2022-03-04:
我默认配置，编译没出现问题哦。

WYC-2020, 2022-03-04:
有时间我看下，初步看应该和ffmpeg有关系

sjkhsl, 2022-03-05:
远程、本地都这样，感觉源码有问题～之前编译的时候都没碰到过这些问题～

WYC-2020, 2022-03-05:
这可能和最近更新的库有关

WYC-2020, 2022-03-05:
遇到错误吧日志贴出来就行了，大佬们看到会修复

WYC-2020, 2022-03-05:
刚编译通过没有任何问题，找下自己原因

mitsukileung, 2022-03-05:
> 刚编译通过没有任何问题，找下自己原因

好的，我重拉试试

Beginner-Go, 2022-03-05:
我这边也测试了 没问题

WYC-2020, 2022-03-05:
> 我这边也测试了 没问题

那可以关闭这个issue了

closed 2022-03-05 by Beginner-Go