coolsnowwolf/lede #9010: 编译报错，求教~

sjkhsl, 2022-03-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

编译报错

### (1) 具体问题

   copy_to_user(user_buf, from, bytes);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  LD [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ath9k.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9002_hw.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_hw.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/hw.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_phy.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9002_phy.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar5008_phy.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9002_calib.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_calib.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/calib.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/eeprom.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/eeprom_def.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/eeprom_4k.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/eeprom_9287.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ani.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/mac.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9002_mac.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_mac.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_eeprom.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_paprd.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ar9003_rtt.o
  LD [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ath9k_hw.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/common.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/common-init.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/common-beacon.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/common-debug.o
  LD [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ath9k_common.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_hst.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/hif_usb.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/wmi.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_txrx.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_main.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_beacon.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_init.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_gpio.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/htc_drv_debug.o
  LD [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath9k/ath9k_htc.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/mac.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/debug.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/core.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/htc.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/htt.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/htt_rx.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/htt_tx.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/txrx.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/wmi.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/wmi-tlv.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/bmi.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/hw.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/p2p.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/swap.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/thermal.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/leds.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/debugfs_sta.o
  CC [M]  /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ath/ath10k/wow.o
[... 17 lines cut ...]
In file included from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/backport-include/linux/of_net.h:3,
                 from /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c:19:
./include/linux/of_net.h:25:19: note: declared here
 static inline int of_get_mac_address(struct device_node *np, u8 *mac)
                   ^~~~~~~~~~~~~~~~~~
make[11]: *** [scripts/Makefile.build:277: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00/rt2x00dev.o] Error 1
make[10]: *** [scripts/Makefile.build:540: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink/rt2x00] Error 2
make[9]: *** [scripts/Makefile.build:540: /home/sjkhsl/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-5.10.42-1/drivers/net/wireless/ralink] Error 2
[... cut ...]
time: package/kernel/mac80211/regular/compile#46.81#4.28#50.73
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/sjkhsl/lede'
make[1]: *** [package/Makefile:110: /home/sjkhsl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sjkhsl/lede'
make: *** [/home/sjkhsl/lede/include/toplevel.mk:230: world] Error 2

编译报错，求教大佬解答一下~

wi11iamzhao, 2022-03-05:
这是因为[5.15内核的一个patch](https://github.com/coolsnowwolf/lede/blob/3ff761aa6db7979c63e9793e3c0fdde6f6a5ba30/target/linux/generic/backport-5.15/003-add-module_supported_device-macro.patch)[被revert了](https://github.com/coolsnowwolf/lede/commit/da34b3a2fb2da075f4e3c7058ab98adbb6eed514)
在有更优雅的解决方法前可以手动把这个patch放回去

WYC-2020, 2022-03-05:
这个我不是给你发了一个修改嘛，你把patch放好就可以的 https://github.com/WYC-2020/lede/commit/893ba3d9e6984f90560a0f93921f651ee3ae96cf

WYC-2020, 2022-03-05:
算了你肯定是不知道放那个位置，放这里[package/kernel/mac80211/patches/rt2x00/001-rt2x00-driver-compile-with-kernel-5.15.patch]

sjkhsl, 2022-03-05:
> 算了你肯定是不知道放那个位置，放这里[package/kernel/mac80211/patches/rt2x00/001-rt2x00-driver-compile-with-kernel-5.15.patch]

感谢大佬，我放错位置了～～～

WYC-2020, 2022-03-05:
问题解决了就关闭issue，看到很老火，都这么多issue了 @sjkhsl

molun, 2022-03-06:
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

这个也是同样的原因吗？HC5962编译时报错

WYC-2020, 2022-03-06:
> make[2]: Leaving directory '/workdir/openwrt' make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2 Error: Process completed with exit code 2.
> 
> 这个也是同样的原因吗？HC5962编译时报错

就这日志鬼知道怎么回事，编译错误贴单线程日志

closed 2022-03-07 by AmadeusGhost