coolsnowwolf/lede #9011: 编译报错，求教大佬

royalwei, 2022-03-05:
checking whether to enable JIS-mapped Japanese font support in GD... no
checking for gdlib >= 2.1.0... yes
checking for gdImageCreateFromPng in -lgd... no
checking for gdImageCreateFromAvif in -lgd... no
checking for gdImageCreateFromWebp in -lgd... no
checking for gdImageCreateFromJpeg in -lgd... no
checking for gdImageCreateFromXpm in -lgd... no
checking for gdImageCreateFromBmp in -lgd... no
checking for gdImageCreateFromTga in -lgd... no
checking for gdImageStringFT in -lgd... no
checking for gdVersionString in -lgd... no
checking for gdImageGetInterpolationMethod in -lgd... no
checking for gdImageCreate in -lgd... no
configure: error: GD build test failed. Please check the config.log for details.
make[3]: *** [Makefile:619: /home/openwrt/lede/build_dir/target-x86_64_musl/php-8.1.3/.configured_cd25afd9635084cf3ac88aa61e05bffb] Error 1
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/lang/php8'
time: package/feeds/packages/php8/compile#7.35#3.91#9.64
    ERROR: package/feeds/packages/php8 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/php8/compile] Error 1
make[2]: Leaving directory '/home/openwrt/lede'
make[1]: *** [package/Makefile:110: /home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
make: *** [/home/openwrt/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2022-03-05:
你们一天哪来那么多编译问题哦，贴单线程日志

royalwei, 2022-03-05:
> checking whether to enable JIS-mapped Japanese font support in GD... no checking for gdlib >= 2.1.0... yes checking for gdImageCreateFromPng in -lgd... no checking for gdImageCreateFromAvif in [...]

closed 2022-03-05 by royalwei