coolsnowwolf/lede #9012: 升级FFMPEG到5.0后luci-app-minidlna无法启动。

wuzhuang123, 2022-03-05:
检查/usr/lib/目录下ffmpeg的这些库版本比较高。系统日志如下：
Sat Mar  5 19:41:14 2022 daemon.err minidlnad[4004]: Error loading shared library libavformat.so.58: No such file or directory (needed by /usr/sbin/minidlnad)
Sat Mar  5 19:41:14 2022 daemon.err minidlnad[4004]: Error loading shared library libavcodec.so.58: No such file or directory (needed by /usr/sbin/minidlnad)
Sat Mar  5 19:41:14 2022 daemon.err minidlnad[4004]: Error loading shared library libavutil.so.56: No such file or directory (needed by /usr/sbin/minidlnad)
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: av_strerror: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: av_log_set_level: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: av_reduce: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: avformat_close_input: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: avformat_find_stream_info: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: av_dict_get: symbol not found
Sat Mar  5 19:41:15 2022 daemon.err minidlnad[4004]: Error relocating /usr/sbin/minidlnad: avformat_open_input: symbol not found

WYC-2020, 2022-03-05:
这是动态链接库里面有几个符号没找到，有时间看下

wuzhuang123, 2022-03-05:
> 这是动态链接库里面有几个符号没找到，有时间看下

几天前FFMPEG在4.X版本时minidlna是可以用的。此外，貌似FFMPEG无法下载m3u8视频。

WYC-2020, 2022-03-05:
> > 这是动态链接库里面有几个符号没找到，有时间看下
> 
> 几天前FFMPEG在4.X版本时minidlna是可以用的。此外，貌似FFMPEG无法下载m3u8视频。

应该不可能，ffmpeg必须可以处理m3u8 作为音视频库开源的大佬不可能不支持m3u8

wuzhuang123, 2022-03-05:
> > > 这是动态链接库里面有几个符号没找到，有时间看下
> > 
> > 
> > 几天前FFMPEG在4.X版本时minidlna是可以用的。此外，貌似FFMPEG无法下载m3u8视频。
> 
> 应该不可能，ffmpeg必须可以处理m3u8 作为音视频库开源的大佬不可能不支持m3u8

好的，我再试试。

WYC-2020, 2022-03-05:
[image]
[image]
@wuzhuang123  我这边没有任何问题，自己排查一下

Beginner-Go, 2022-03-05:
实在不行 我回退ffmpg4.x ?

wuzhuang123, 2022-03-05:
> 实在不行 我回退ffmpg4.x ?

我再确认一下，不选择ffmpeg试试。感谢。

WYC-2020, 2022-03-05:
> 实在不行 我回退ffmpg4.x ?

不用回退 我这边没任何问题 5.0

wuzhuang123, 2022-03-05:
> > 实在不行 我回退ffmpg4.x ?
> 
> 不用回退 我这边没任何问题 5.0

好的，我继续排查一下，感谢。

WYC-2020, 2022-03-05:
我看你的lib 根本就没对什么时候来一个56  你怕不是自己编译的吧 用的别人编译好的直接安装的
[image]

WYC-2020, 2022-03-05:
@wuzhuang123 这个插件才是写的有点问题，在保存一次后在此保存并不会进行重新启动，即使修改了监听端口也是一样，我简单粗暴修复了一下，有更好修复的可以优化
https://github.com/WYC-2020/luci/commit/8adedc449bb7aa71d6d115a8e321cbfd8507819d

wuzhuang123, 2022-03-05:
> 我看你的lib 根本就没对什么时候来一个56 你怕不是自己编译的吧 用的别人编译好的直接安装的 [image]

没有，ffmpeg和luci-app-minidlna都是自己编译的，当前最新的。我再重新清理编译一下。
[image]

WYC-2020, 2022-03-05:
> > 我看你的lib 根本就没对什么时候来一个56 你怕不是自己编译的吧 用的别人编译好的直接安装的 [image]
> 
> 没有，ffmpeg和luci-app-minidlna都是自己编译的，当前最新的。我再重新清理编译一下。 [image]

那你重新编译一下吧，清理一下肯定是你哪里有问题，因为我这边一切正常，我也是拉取合并l大的 虽然我自己定制了一些，但是整体和l大的还是没多大差别，你自己在排查一下

wuzhuang123, 2022-03-05:
> > > 我看你的lib 根本就没对什么时候来一个56 你怕不是自己编译的吧 用的别人编译好的直接安装的 [image]
> > 
> > 
> > 没有，ffmpeg和luci-app-minidlna都是自己编译的，当前最新的。我再重新清理编译一下。 [image]
> 
> 那你重新编译一下吧，清理一下肯定是你哪里有问题，因为我这边一切正常，我也是拉取合并l大的 虽然我自己定制了一些，但是整体和l大的还是没多大差别，你自己在排查一下

好的，感谢。

WYC-2020, 2022-03-05:
看你这个日志
[image]
明显就是在编译minidlna 的时候他链接的ffmpeg 不是链接的5.0的
@wuzhuang123

wuzhuang123, 2022-03-05:
不选择FFMPEG仍然不行，难道是我dl库有问题？

breakings, 2022-03-08:
ldd /usr/sbin/minidlnad `` 就知道动态链接库是否正常了 重新 make package/feeds/packages/minidlda/{clean,compile}

wuzhuang123, 2022-03-08:
非常感谢，创建软链接后服务可以运行，问题已解决。
[image]

breakings, 2022-03-08:
> 非常感谢，创建软链接后服务可以运行，问题已解决。 [image]

最好是重新编译一下 minidlda 再重装强制安装下ipk包就好了

wuzhuang123, 2022-03-08:
好的，正在全新编译镜像，感谢。

closed 2022-03-11 by wuzhuang123