coolsnowwolf/lede #9014: 大佬，ath79里的机型编译出来的固件USB无法使用

iotdoc58, 2022-03-06:
非常感谢大神提供的源码  使用编译WR703N 启动后 USB无电源输出，
采用旧内核版本USB 就正常。
不知道是怎么回事，单独给予供电也无法使用。是不是固件中没有 Vbus没有使能。？研究了很长时间无效，麻烦大神帮忙给予解决一下，谢谢了！

jkeegg, 2023-03-13:
同USB无法使用，板子上已经短路R113，还是检测不到新设备，旧版本正常

pxianghui, 2025-07-21:
我最这几天编译这个固件遇到一样的问题，不知道大佬还会不会关注了，毕竟芯片有点过时了，不过还是希望能解决一下，用别人编译好的固件USB一切正常

jkeegg, 2025-07-21:
> 我最这几天编译这个固件遇到一样的问题，不知道大佬还会不会关注了，毕竟芯片有点过时了，不过还是希望能解决一下，用别人编译好的固件USB一切正常

可以参考一下我之前的修改，很久之前改的，没在这个版本验证过
https://github.com/jkeegg/lede/commit/8d12b4b732d0c92dabed81d6b89437f92e7f5224
target/linux/ath79/dts/ar9330.dtsi这个文件

pxianghui, 2025-07-21:
没想到这么快就有人回复，感谢还有人关注，我研究一下

pxianghui, 2025-07-21:
用你这个dtsi文件编译了2编还是不行

coolsnowwolf, 2026-09-05:
已定位并修复源码问题。当前 ath79 6.6 的 AR933x USB PHY backport 驱动按 phy、host、suspend-override 获取 reset，但 ar9330.dtsi 仍使用旧的 usb-phy、usb-host、usb-suspend-override 名称，导致 WR703N 的 USB PHY/控制器复位资源匹配失败。已按上游 OpenWrt 修复同步调整，提交并推送：b7e3e43f2b6 ath79: fix AR933x USB reset names。使用 dtc 编译 ar9331_tplink_tl-wr703n.dts 成功生成 DTB；请用包含该提交的固件刷机后验证 USB 枚举。

closed 2026-09-05 by coolsnowwolf