coolsnowwolf/lede #9016: rblibtorrent is missing dependencies for the following libraries: libatomic.so.1

tqangxl, 2022-03-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

Package rblibtorrent is missing dependencies for the following libraries:
libatomic.so.1
check 
https://openwrt.org/packages/table/start?dataflt%5BName_pkg-dependencies*%7E%5D=libatomic

### (2) 路由器型号和固件版本

A：root@OpenWrt:~# cat /proc/cpuinfo
processor	: 0
cpu		: APM821XX
clock		: 1000.000010MHz
revision	: 28.131 (pvr 12c4 1c83)
bogomips	: 2000.00

timebase	: 1000000010
platform	: PowerPC 44x Platform
model		: Netgear WNDR4700/WNDR4720 Series
Memory		: 256 MB
查看当前路由器的架构
root@OpenWrt:~# opkg print-architecture | awk '{print $2}'
all
noarch
powerpc_464fp
git 最新的版本

### (3) 详细日志

A：install -d -m0755 /home/james/Desktop/lede/build_dir/target-powerpc_464fp_musl/rblibtorrent-1.2.15/ipkg-powerpc_464fp/rblibtorrent/usr/lib
cp -fpR /home/james/Desktop/lede/build_dir/target-powerpc_464fp_musl/rblibtorrent-1.2.15/ipkg-install/usr/lib/*.so* /home/james/Desktop/lede/build_dir/target-powerpc_464fp_musl/rblibtorrent-1.2.15/ipkg-powerpc_464fp/rblibtorrent/usr/lib
find /home/james/Desktop/lede/build_dir/target-powerpc_464fp_musl/rblibtorrent-1.2.15/ipkg-powerpc_464fp/rblibtorrent -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package rblibtorrent is missing dependencies for the following libraries:
libatomic.so.1
make[3]: *** [Makefile:71: /home/james/Desktop/lede/bin/packages/powerpc_464fp/packages/rblibtorrent_1.2.15-2_powerpc_464fp.ipk] Error 1
make[3]: Leaving directory '/home/james/Desktop/lede/feeds/packages/libs/rblibtorrent'
time: package/feeds/packages/rblibtorrent/compile#326.12#45.42#365.42
    ERROR: package/feeds/packages/rblibtorrent failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/rblibtorrent/compile] Error 1
make[2]: Leaving directory '/home/james/Desktop/lede'
make[1]: *** [package/Makefile:110: /home/james/Desktop/lede/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/james/Desktop/lede'
make: *** [/home/james/Desktop/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2022-03-07:
我这边查看并没你说的哪个名字变化
[image]

breakings, 2022-03-07:
好像是听说有些平台不能用libatomic

tqangxl, 2022-03-07:
> check
> https://openwrt.org/packages/table/start?dataflt%5BName_pkg-dependencies*%7E%5D=libatomic

tqangxl, 2022-03-08:
/home/james/Desktop/openwrt/build_dir/target-powerpc_464fp_musl/toolchain/ipkg-powerpc_464fp/libatomic

Package: libatomic1
Version: 8.4.0-3
Depends: libgcc1
Provides: libatomic
Source: package/libs/toolchain
SourceName: libatomic
License: GPL-3.0-with-GCC-exception
Section: libs
SourceDateEpoch: 1644640906
ABIVersion: 1
Status: unknown hold not-installed
Essential: yes
Maintainer: Felix Fietkau <nbd@nbd.name>
Architecture: powerpc_464fp
Installed-Size: 7289
Description:  Atomic support library

tqangxl, 2022-03-08:
@WYC-2020 @breakings @zx2c4 @ynezz  pls help

breakings, 2022-03-08:
> @WYC-2020 @breakings @zx2c4 @ynezz pls help

没编译powerpc这个平台，arm64和x86都是正常的

WYC-2020, 2022-03-08:
这个不清楚没看，因为我编译基本都是x86，这个后面有时间看 @tqangxl @breakings

WYC-2020, 2022-03-08:
[image]
[image]
[image]
并没有你说的哪个情况，我这边一切正常，多半自己的问题，如果你是组合的那就另当别论 @tqangxl

breakings, 2022-03-08:
> [image] [image] [image] 并没有你说的哪个情况，我这边一切正常，多半自己的问题，如果你是组合的那就另当别论 @tqangxl

所以觉得他那个报错好奇怪，难道是旧版GCC的问题？

WYC-2020, 2022-03-08:
> > [image] [image] [image] 并没有你说的哪个情况，我这边一切正常，多半自己的问题，如果你是组合的那就另当别论 @tqangxl
> 
> 所以觉得他那个报错好奇怪，难道是旧版GCC的问题？

谁知道哦，反正没问题就不管

tqangxl, 2022-03-08:
> > [image] [image] [image] 并没有你说的哪个情况，我这边一切正常，多半自己的问题，如果你是组合的那就另当别论 @tqangxl
> 
> 所以觉得他那个报错好奇怪，难道是旧版GCC的问题？

感谢感谢，GCC改成11版本就过了这个报错

tqangxl, 2022-03-08:
> [image] [image] [image] 并没有你说的哪个情况，我这边一切正常，多半自己的问题，如果你是组合的那就另当别论 @tqangxl

感谢您能帮忙验证！谢谢您！

closed 2022-03-08 by tqangxl