coolsnowwolf/lede #9017: amule compile issue with latest code

GreatMichaelLee, 2022-03-07:
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/amule'
mkdir -p /home/builder/lede_x86/dl
SHELL= flock /home/builder/lede_x86/tmp/.aMule-2.3.2-r4b87b20.tar.bz2.flock -c '        /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "aMule-2.3.2-r4b87b20.tar.bz2" "1575d31d68f9df881100c9e30f18fc7dec8493a8ae20787f41b6ef23604a78ff" "" || (  /home/builder/lede_x86/scripts/dl_github_archive.py --dl-dir="/home/builder/lede_x86/dl" --url="https://github.com/amule-project/amule.git" --version="4b87b20" --subdir="aMule-2.3.2" --source="aMule-2.3.2-r4b87b20.tar.bz2" --hash="1575d31d68f9df881100c9e30f18fc7dec8493a8ae20787f41b6ef23604a78ff" || (   echo "Checking out files from the git repository..."; mkdir -p /home/builder/lede_x86/tmp/dl && cd /home/builder/lede_x86/tmp/dl && rm -rf aMule-2.3.2 && [ \! -d aMule-2.3.2 ] && git clone  https://github.com/amule-project/amule.git aMule-2.3.2 && (cd aMule-2.3.2 && git checkout 4b87b20 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd aMule-2.3.2 && git log -1 --format='\''@%ct'\''` && rm -rf aMule-2.3.2/.git &&       tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c aMule-2.3.2 |       bzip2 -c > /home/builder/lede_x86/tmp/dl/aMule-2.3.2-r4b87b20.tar.bz2 && mv /home/builder/lede_x86/tmp/dl/aMule-2.3.2-r4b87b20.tar.bz2 /home/builder/lede_x86/dl/ && rm -rf aMule-2.3.2; );  )    '
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2/.built_check
true -i 's;^CRYPTOPP_LIBS.*;CRYPTOPP_LIBS \= "/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/lib/libcryptopp.a";g' /home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2/src/Makefile
make -C /home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2 HOSTCC="gcc" DESTDIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2/ipkg-install" all
make[4]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2'
Compiling ClientCreditsList.cpp
ClientCreditsList.cpp: In member function 'void CClientCreditsList::InitalizeCrypting()':
ClientCreditsList.cpp:315:10: error: 'CryptoPP::RSASSA_PKCS1v15_SHA_Verifier' {aka 'class CryptoPP::PK_FinalTemplate<CryptoPP::TF_VerifierImpl<CryptoPP::TF_SignatureSchemeOptions<CryptoPP::TF_SS<CryptoPP::RSA, CryptoPP::PKCS1v15, CryptoPP::SHA1, int>, CryptoPP::RSA, CryptoPP::PKCS1v15_SignatureMessageEncodingMethod, CryptoPP::SHA1> > >'} has no member named 'DEREncode'
   pubkey.DEREncode(asink);
          ^~~~~~~~~
make[7]: *** [Makefile:2620: amuled-ClientCreditsList.o] Error 1
make[6]: *** [Makefile:3411: all-recursive] Error 1
make[5]: *** [Makefile:588: all-recursive] Error 1
make[4]: *** [Makefile:495: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2'
make[3]: *** [Makefile:114: /home/builder/lede_x86/build_dir/target-x86_64_musl/aMule-2.3.2/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/amule'
time: package/feeds/packages/amule/compile#1.49#0.35#1.79
    ERROR: package/feeds/packages/amule failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/amule/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-03-07:
这玩意你稍微搜索一下相同的issue 就会找到我说过解决方法哈哈

coolsnowwolf, 2022-03-07:
fixed

closed 2022-03-07 by coolsnowwolf

GreatMichaelLee, 2022-03-07:
> 这玩意你稍微搜索一下相同的issue 就会找到我说过解决方法哈哈

貌似没有提交到这里？