coolsnowwolf/lede #9018: 缺个依赖宝 kmod-fs-ntfs3-oot  不知道哪里找

lmxslpc, 2022-03-07:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-fs-ntfs3-oot for ntfs3-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for ntfs3-mount found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for automount:
 * 	kmod-fs-ntfs3-oot
 * opkg_install_cmd: Cannot install package automount.
make[2]: *** [package/Makefile:70: package/install] Error 255

WYC-2020, 2022-03-07:
你怕是编译的5.15内核吧，这个要手动删除build文件夹里面的ntfs3-mount文件夹，再编译看下，这些都是老问题了，主要是更换内核后ntfs3-mount的依赖并不会更新导致

lmxslpc, 2022-03-07:
好的 大佬  我试试

WYC-2020, 2022-03-07:
这个问题要想根本解决就得让这个库每次都强制重新来一遍

lmxslpc, 2022-03-07:
大佬 还不是不行 能讲具体点么

WYC-2020, 2022-03-07:
首先确认config 里面有没选择编译ntfs3-oot 这个，如果没有就删除build 里面的 automount ntfs3-mount 再编译，如果有选择编译这个 就取消编译这个，再删除build 里面的automount ntfs3-mount  @lmxslpc

WYC-2020, 2022-03-07:
如果还不行 再确保config中没有编译ntfs3-oot后，执行make clean，再编译，如果还不行就万能大法，不过基本上经过前面的都能编译过了

lmxslpc, 2022-03-07:
ntfs3-mount 这个包是干什么的？默认选中我就没动 
还有什么是万能大法

LPDDR6-10000MHz, 2022-03-07:
巧了，我也刚遇到😂，删除源码重新下载再编译

WYC-2020, 2022-03-07:
> ntfs3-mount 这个包是干什么的？默认选中我就没动 还有什么是万能大法

自己看l大的提交，这就一个挂载ntfs分区的一个驱动而已，万能大法就是make distclean 然后全新编译

lmxslpc, 2022-03-07:
好的 谢谢大佬 我干脆删掉了

closed 2022-03-07 by AmadeusGhost