coolsnowwolf/lede #9019: scripts/feeds update -a 报错

Cptmacmillan2022007, 2022-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：克隆了最新源码之后出现的

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

Can't exec "make": No such file or directory at ./scripts/feeds line 22.
Use of uninitialized value in split at ./scripts/feeds line 22.
Use of uninitialized value $mkver[0] in pattern match (m//) at ./scripts/feeds line 24.
Use of uninitialized value $mkver[1] in pattern match (m//) at ./scripts/feeds line 25.
Use of uninitialized value $mkver[2] in split at ./scripts/feeds line 27.
Use of uninitialized value $mkv1 in numeric ge (>=) at ./scripts/feeds line 28.
Use of uninitialized value $mkv1 in numeric eq (==) at ./scripts/feeds line 28.
Unsupported version of make found: make

WYC-2020, 2022-03-07:
怕是自己环境都没搭建好，就在编译吧

GreatMichaelLee, 2022-03-07:
哈哈，这个太有趣了。

WYC-2020, 2022-03-07:
看你这个日志就感觉没有make,确定环境部署好了，有趣吧

closed 2022-03-07 by Cptmacmillan2022007